genodelabs/genode #3475: ssh: pipe into exec channel - raw mode

sid-com, 2019-08-19:


sid-com, 2019-08-19:
5fa1ce8 is a workaround for transferring files larger than 64k. Any suggestions on more elegant solutions are appreciated.

sid-com, 2019-10-07:
Dear genodians. 

I have a page fault that happens occasionally on the patches I provided in here. This happens also when I cherry-pick the commits to genodelabs/master. The page fault occurs every couple of hundred ssh connections to ssh_terminal. 

```
...
ssh_exec_channel.run iteration 122
test exec channel echo
spawn sshpass -p xuon ssh -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -l noux localhost -p 5555 echo test 122
[init -> ssh_terminal] 2000-01-01 00:02:29 login user noux (123) with password
[init -> ssh_terminal] 2000-01-01 00:02:29 request Terminal for user noux (0x165050)
[init -> exec_terminal] <request_terminal user="noux" command="echo test 122"/>
[init -> dynamic] Warning: noux assigned RAM (64M) exceeds available RAM (32510556)
[init -> dynamic -> noux] tar archive 'bash.tar' local at 0x15c000, size is 8081408
[init -> dynamic -> noux] tar archive 'coreutils-minimal.tar' local at 0x911000, size is 1413120
[init -> dynamic -> noux] tar archive 'vim-minimal.tar' local at 0xa6a000, size is 2142208
[init -> dynamic -> noux] Warning: stdin VFS path not defined, connecting to terminal session
[init -> dynamic -> noux] Warning: stdout VFS path not defined, connecting to terminal session
[init -> dynamic -> noux] Warning: stderr VFS path not defined, connecting to terminal session
[init -> dynamic -> noux] --- noux started ---
Warning: Permanently added '[localhost]:5555' (ED25519) to the list of known hosts.
test 122
[init -> dynamic -> noux] Warning: heap could not free memory block
[init -> ssh_terminal] 2000-01-01 00:02:30 logout user noux (123)
[init -> dynamic] child "noux" exited with exit value 0
ssh_exec_channel.run iteration 123
test exec channel echo
spawn sshpass -p xuon ssh -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -l noux localhost -p 5555 echo test 123
[init -> exec_terminal] <request_terminal user="noux" exit="now"/>
no RM attachment (READ pf_addr=0xc0 pf_ip=0x1401f5f from pager_object: pd='init -> ssh_terminal' thread='ep') 
page fault, pd='init -> ssh_terminal' thread='ep' cpu=0 ip=0x1401f5f address=0xc0 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1
Error: Test execution timed out
make: *** [Makefile:324: run/ssh_exec_channel] Error 254
```

In the runscript gems/run/ssh_exec_channel.run I set `ld_verbose="yes"` to the config node of `ssh_terminal`. The memory layout of `ssh_terminal` is as follows:

```
[init -> ssh_terminal]   0xa0000000 .. 0xafffffff: stack area
[init -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> acpi_drv] Found MCFG
[init -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPCRSDT', revision 1, creator 'BXPC' (1)
[init -> ssh_terminal]   0x30000 .. 0x15afff: ld.lib.so
[init -> ssh_terminal]   0xae35000 .. 0xaffffff: libc.lib.so
[init -> ssh_terminal]   0xadc2000 .. 0xae34fff: vfs.lib.so
[init -> ssh_terminal]   0x1034000 .. 0x103cfff: libc_pipe.lib.so
[init -> ssh_terminal]   0x103d000 .. 0x10dafff: libssh.lib.so
[init -> ssh_terminal]   0x10db000 .. 0x1347fff: libcrypto.lib.so
[init -> ssh_terminal]   0x1348000 .. 0x135efff: zlib.lib.so
[init -> platform_drv] ECAM/MMCONF range 00:00.0-ff:1f.7 - addr [00000000b0000000,00000000c0000000)
[init -> platform_drv] Root bridge: 00:00.0
[init -> ssh_terminal] using the pipe libc plugin
[init -> ssh_terminal]   0xadb6000 .. 0xadc1fff: vfs_jitterentropy.lib.so
[init -> acpi_drv] SMBIOS table (entry point: 0x165710 structures: 0xf5730)
[init -> ssh_terminal]   0xad9e000 .. 0xadb5fff: vfs_lxip.lib.so
```

The page fault occurs at address `0x1401f5f` which is in the range between `0x135efff` (zlib.lib.so) and `0xadb6000` (vfs_jitterentropy.lib.so). What's going on here?
It looks like the page fault happens before any of the callbacks get called I assume the issue to be in libssh. However, the address of the instruction pointer is no where near the range of libssh.lib.so...

@cnuke could you help me out here?

chelmuth, 2019-10-08:
This looks like a perfect candidate for `qemu -sS` debugging.

chelmuth, 2020-04-17:
This issue is stale since 4 months. May it be closed or are there updates?

sidhussmann, 2020-04-18:
@chelmuth it's really up to you. I'm willing to put effort in it again, if this feature is something y'all interested in.

chelmuth, 2020-04-20:
So, I close this issue now with the option to reopen on interest.

closed 2020-04-20 by chelmuth