genodelabs/genode #3476: dup2()ing to stderr twice results in crash

senier, 2019-08-19:
I have the following simple libc program:

```cpp
      int fd = open ("/tmp/stderr-1", O_CREAT|O_RDWR);                                                                  
      dup2(fd, 2);                                                                                                      
      close (fd);                                                                                                       
      fd = open ("/tmp/stderr-2", O_CREAT|O_RDWR);                                                                      
      dup2(fd, 2);
```

When `/tmp` is configured as a `<ram/>` node in VFS, this results in a crash (tested on nova and base-linux, both x86_64):

```sh
no RM attachment (READ pf_addr=0x0 pf_ip=0x11313bf from pager_object: pd='init -> test_dup_stderr' thread='ep')
```

When using a separate `ram_fs` server for `/tmp`, the error is slightly different:

```sh
[init -> ram_fs] Error: Invalid_handle
[init -> test_dup_stderr] Warning: ack for unknown File_system handle 1
```

The culprit is the second `dup2` in the last line. When commented out, everything is find. It seems like the crash is happening when closing the old FD 2 in the VFS plugin.

nfeske, 2019-11-19:
Should be fixed in master.

senier, 2019-12-13:
Works with 19.11.

closed 2019-12-13 by senier