genodelabs/genode #3478: libc: fork, wait4, getpid

nfeske, 2019-08-19:
Genode's C runtime lacks support for `fork`. For running Unix applications depending on the forking of processes on top of Genode, one had to use the Noux runtime. In the future, I'd like to consolidate the feature set of Noux into the regular C runtime to simplify the porting of software to Genode and to overcome today's performance limitations of Noux. Hence, the libc must be equipped with a locally implemented fork mechanism and related functions like `getpid`, `waitpid` (aka `wait4`).

nfeske, 2019-08-19:
The commits on my "fork" branch (https://github.com/nfeske/genode/commits/fork) implement the feature. Note that the branch contains a few small commits that are not tagged with this issue but should nevertheless also be reviewed and merged.

nfeske, 2019-08-21:
Marked as fixed now. Thanks to the additional commits https://github.com/nfeske/genode/commit/5170beb3b6887420ef24e1cf66c09bf354979f60 and https://github.com/nfeske/genode/commit/3af91ced7c3a4ed718e5b02fae3c3ca9570dd07b, the fork test runs on base-linux also. Thanks to @ssumpf for the support!

nfeske, 2019-08-22:
Commit https://github.com/genodelabs/genode/commit/8f163f6a63e81a06927d776b32174d5b0c8289fc adds the forwarding of the current working directory and the RTC configuration to the child.

cproc, 2019-08-26:
This is the backtrace of a page fault that occurs with Arora. It looks like a static constructor calls `malloc()` before `init_malloc()` has been called.

```
#0  Genode::cmpxchg (new_val=0, cmp_val=1, dest=0x2a8) at /.../repos/base/include/spec/x86/cpu/atomic.h:38
#1  spinlock_lock (lock_variable=0x2a8) at /.../repos/base/src/include/base/internal/spin_lock.h:37
#2  Genode::Cancelable_lock::lock (this=this@entry=0x2a8) at /.../repos/base/src/lib/base/lock.cc:67
#3  Genode::Lock::lock (this=0x2a8) at /.../repos/base/include/base/lock.h:33
#4  Genode::Lock_guard<Genode::Lock>::Lock_guard (lock=..., this=<synthetic pointer>) at /.../repos/base/include/base/lock_guard.h:40
#5  Libc::Malloc::alloc (size=32, this=0x0) at /.../repos/libports/src/lib/libc/malloc.cc:158
#6  malloc () at /.../repos/libports/src/lib/libc/malloc.cc:232
#7  operator new (sz=1074790400, sz@entry=32) at /.../contrib/stdcxx-79ad8821269a443aab21bfc1d0e9b8913332b6d1/src/lib/stdcxx/libsupc++/new_op.cc:47
#8  QMutex::QMutex(QMutex::RecursionMode) () at /.../contrib/stdcxx-79ad8821269a443aab21bfc1d0e9b8913332b6d1/include/stdcxx/bits/atomic_base.h:682
#9  (anonymous namespace)::Q_QGS_resourceMutex::Holder::Holder (this=0x3fcf880 <(anonymous namespace)::Q_QGS_resourceMutex::innerFunction()::holder>) at /.../contrib/qt5-a4aa73007396403d6a54a0a3a659420603cd7a8c/src/lib/qt5/qt5/qtbase/src/corelib/io/qresource.cpp:152
#10 (anonymous namespace)::Q_QGS_resourceMutex::innerFunction () at /.../contrib/qt5-a4aa73007396403d6a54a0a3a659420603cd7a8c/src/lib/qt5/qt5/qtbase/src/corelib/io/qresource.cpp:152
#11 QGlobalStatic<QMutex, (anonymous namespace)::Q_QGS_resourceMutex::innerFunction, (anonymous namespace)::Q_QGS_resourceMutex::guard>::operator() (this=<optimized out>) at /.../contrib/qt5-a4aa73007396403d6a54a0a3a659420603cd7a8c/include/QtCore/qglobalstatic.h:135
#12 qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) () at /.../contrib/qt5-a4aa73007396403d6a54a0a3a659420603cd7a8c/src/lib/qt5/qt5/qtbase/src/corelib/io/qresource.cpp:850
#13 qInitResources_mimetypes () at qrc_mimetypes.cpp:15799
#14 (anonymous namespace)::initializer::initializer (this=0x3fcf778 <(anonymous namespace)::dummy>) at qrc_mimetypes.cpp:15817
#15 __static_initialization_and_destruction_0 (__priority=65535, __initialize_p=1) at qrc_mimetypes.cpp:15816
#16 _GLOBAL__sub_I_qrc_mimetypes.cpp(void) () at qrc_mimetypes.cpp:15817
#17 _init () at /.../repos/base/src/lib/ldso/startup/startup.cc:36
#18 Linker::Dynamic::call_init_function (this=<optimized out>) at /.../repos/base/src/lib/ldso/include/dynamic.h:214
#19 Linker::Init::exec_static_constructors (this=0x501baef0 <Linker::Init::list()::_list>) at /.../repos/base/src/lib/ldso/include/init.h:127
#20 Linker::Init::initialize (stage=Linker::STAGE_SO, bind=Linker::BIND_LAZY, this=0x501baef0 <Linker::Init::list()::_list>) at /.../repos/base/src/lib/ldso/include/init.h:104
#21 Linker::Root_object::Root_object(Genode::Env&, Genode::Allocator&, char const*, Linker::Bind, Linker::Keep) () at /.../repos/base/src/lib/ldso/dependency.cc:114
#22 Genode::Shared_object::Shared_object(Genode::Env&, Genode::Allocator&, char const*, Genode::Shared_object::Bind, Genode::Shared_object::Keep) () at /.../repos/base/src/lib/ldso/shared_object.cc:66
#23 Vfs::Global_file_system_factory::_load_factory(Vfs::Env&, Genode::String<128ul> const&) () at /.../repos/base/include/util/string.h:712
#24 Vfs::Global_file_system_factory::_probe_external_factory(Vfs::Env&, Genode::Xml_node) () at /.../repos/base/include/util/string.h:712
#25 Vfs::Global_file_system_factory::create(Vfs::Env&, Genode::Xml_node) () at /.../repos/os/src/lib/vfs/file_system_factory.cc:194
#26 Vfs::Dir_file_system::Dir_file_system (this=<optimized out>, env=..., node=..., fs_factory=...) at /.../repos/os/include/vfs/dir_file_system.h:389
#27 Vfs::Dir_file_system::Dir_file_system (this=<optimized out>, env=..., node=..., fs_factory=...) at /.../repos/os/include/vfs/simple_env.h:49
#28 Vfs::Simple_env::Simple_env (config=..., alloc=..., env=..., this=0xaff95a8 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+8904>) at /.../repos/base/include/util/interface.h:20
#29 Libc::Env_implementation::Env_implementation (alloc=..., env=..., this=0xaff9440 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space+8544>) at /.../repos/libports/src/lib/libc/task.cc:96
#30 Libc::Kernel::Kernel (this=0xaff72e0 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space>, env=..., heap=...) at /.../repos/libports/src/lib/libc/task.cc:706
#31 Unmanaged_singleton_constructor::call<Libc::Kernel, Genode::Env&, Genode::Allocator&> (dst=0xaff72e0 <Libc::Kernel* unmanaged_singleton<Libc::Kernel, 8, Genode::Env&, Genode::Allocator&>(Genode::Env&, Genode::Allocator&)::object_space> "\020=\373\n") at /.../repos/base/src/include/base/internal/unmanaged_singleton.h:58
#32 unmanaged_singleton<Libc::Kernel> () at /.../repos/base/src/include/base/internal/unmanaged_singleton.h:85
#33 Component::construct (env=...) at /.../repos/libports/src/lib/libc/task.cc:1315
[... cut ...]
#44 Genode::Object_pool<Genode::Rpc_object_base>::apply<Genode::Rpc_entrypoint::entry()::<lambda(Genode::Rpc_object_base*)> > (func=..., capid=<optimized out>, this=0x500f3b50 <Genode::bootstrap_component()::startup+528>) at /.../repos/base/include/base/object_pool.h:146
#45 Genode::Rpc_entrypoint::entry() () at /.../repos/base/src/lib/base/rpc_dispatch_loop.cc:77
#46 Genode::Thread::_thread_start() () at /.../repos/base/include/base/output.h:54
#47 thread_start () at /.../repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

nfeske, 2019-08-27:
Thanks for the insightful backtrace. The dynamic linker happens to execute the static constructors (of all ELF objects) as a side effect of constructing the VFS, *before* the libc has called `Env::exec_static_constructors`. This is not expected and wrong. Previously, the `malloc` support got initialized a bit earlier, masking this issue. But this was only by chance.

The problem seems to be the dynamic linker's built-in policy to execute initializers at construction time of a `Shared_object`. I guess that the assumption was that any `Shared_object` is constructed by an application (via `ld_open`) at a point where the libc initialization is already done. However, such objects are also constructed when loading VFS plugins during the libc initialization.

I suspect the problem to be at https://github.com/nfeske/genode/blob/release_notes/repos/base/src/lib/ldso/include/init.h#L103-104. This condition should not be true before `Env::exec_static_constructors` has been called. (but to validate this claim, I have the reproduce the problem. Building Qt and Webkit now...)

nfeske, 2019-08-27:
> I suspect the problem to be at ...

That is indeed the case. The stage is passed as hard-wired argument here: https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/ldso/dependency.cc#L111  It should instead be a linker-global state variable.

commit a1d7a72a (2019-08-21): libc: fork, getpid, and wait4

This patch complements the C runtime with support for fork, getpid, and
wait4 (and its cousin 'waitpid').

Fixes #3478
commit bb5827b4 (2019-08-28): libc: fork, getpid, and wait4

This patch complements the C runtime with support for fork, getpid, and
wait4 (and its cousin 'waitpid').

Fixes #3478
closed 2019-08-28 by chelmuth