genodelabs/genode #3481: libc: execve

nfeske, 2019-08-21:
As the counterpart of fork (#3478), we desire an implementation of `execve` in Genode's libc to accommodate the usual Unix command line tools.

nfeske, 2019-08-22:
Commits https://github.com/genodelabs/genode/commit/8fba4e2b7205525c894438c89ce0c1dd30fde90a and https://github.com/genodelabs/genode/commit/50f0afd900069b81e18d89b6f6f4ba18837a90af equip the dynamic linker with a base mechanism suitable for implementing execve on top.

@ssumpf could you please have a look at the two commits?

nfeske, 2019-08-22:
Commit https://github.com/genodelabs/genode/commit/3eb465dacb6e40dd4b170162d983fd7ccc181134 implements the mechanism. I already merged this commit to staging to see how the autopilot fares. @chelmuth I'd appreciate you reviewing the commit.

Combined, the fork and execve mechanism already allow noux-like scenarios like the one introduced by commit https://github.com/genodelabs/genode/commit/c9e68b106090af2480ec0a194f77ceb936d3a410. However, there are still a few known limitations: First, shebangs ("!#") are not interpreted yet. Second, pipes don't exist. Third, the foreground/background handling is not solid. E.g., when pressing Control-D in a child process, both the child and parent will see it. Fourth, the executable binaries must all be stored in `/bin`. Finally, many corner cases are probably missing (looking at the list of potential error codes returned by execve).

That said, as of now, one can use an interactive shell (bash), coreutils, and vim. The mechanism even works on base-linux, which was never possible with noux.

chelmuth, 2019-08-23:
Wow, I just finished my review and am quite surprised how lean the implementation is (it just adds 400 lines of code in this commit). Great!

nfeske, 2019-08-23:
Thanks for the review and the compliment. I just added the fixup https://github.com/genodelabs/genode/commit/5fa74ac869084371525cc6db222826a9da85973c to address your remarks.

ssumpf, 2019-08-27:
@nfeske: ff4432b8 fixes `noux_tool_chain_auto`, somehow the `.` caused the problem, leading `stat` to not find the executable file.

nfeske, 2019-08-27:
@ssumpf Verrückt. Thanks for identifying the cause and for fixing it.

nfeske, 2019-08-27:
@ssumpf can you please give https://github.com/nfeske/genode/commit/cc390db6596ae55788601ce7826a8c8e41354dae a try?

ssumpf, 2019-08-27:
@nfeske: "Test succeeded" :+1:

nfeske, 2019-08-27:
@ssumpf Cool! Thanks for the great support.

nfeske, 2019-08-28:
Fixed in master.

ssumpf, 2019-12-13:
@nfeske: 8ef6cbc fixes GDB symbol loading after fork/exec.

nfeske, 2019-12-13:
@ssumpf Super fantastic!

chelmuth, 2019-12-20:
Merged to master.

closed 2019-12-20 by chelmuth