genodelabs/genode #3483: Tooling and support for keyboard layouts

chelmuth, 2019-08-23:
The *input_filter* does a pretty good job in supporting a bunch of localized keyboard layouts and Unicode character generators, but has some quirks this issue addresses. Also, related fixes, for example, Qt5 key mappings are subsumed in this issue.

[ ] Caps Lock works different from Shift Lock https://en.wikipedia.org/wiki/Caps_Lock
[ ] Support dead-key/composing key sequences https://en.wikipedia.org/wiki/Dead_key
[ ] Update/amend chargen configs for en_us, de_de, de_ch (sg), fr_ch (sf), fr
[ ] tool to generate keyboard layout and character generator configuration from XKB https://github.com/chelmuth/xkb2ifcfg (integrate into genode source tree)

Localized keyboards can easily be tested online https://docs.microsoft.com/en-us/globalization/windows-keyboard-layouts

nfeske, 2019-08-23:
The commit series looks excellent already! That's a very cool feature.

I'd use the regular Genode license header though. The code is using Genode (AGPLv3) headers anyway. So the MIT headers just muddy the waters while the AGPLv3 applies.

chelmuth, 2019-08-23:
The following commits implement the features ticked above.

5746922cda tool: generate input_filter config from XKB
46cba5a563 input_filter: dead-key sequence support
94dac26e22 input: fix Codepoint::INVALID in Press event
cf50694e9b utf8: non-character U+fffe as invalid codepoint

I will add fixups after review and for the missing bullet points.

chelmuth, 2019-08-24:
I addressed the remarks and the missing documentation.

a6e74eaeb4 fixup "input_filter: dead-key sequence support" (no break with exc)
2a11865516 fixup "input_filter: dead-key sequence support" (README)
b6029a1d44 fixup "tool: generate input_filter config from XKB" (license)

nfeske, 2019-08-26:
Great! Thank you for following the suggestions.

commit 8b3d4f97 (2019-09-30): qt5: rework keyboard handling

Fixes #3483
closed 2019-11-19 by chelmuth