genodelabs/genode #3484: xml_generator: indentation broken with appended content right before node closure

chelmuth, 2019-08-23:
This is only cosmetic but produced unattractive XML files in xkb2ifcfg, where add a `<dummy/>` node after the last comment if needed. See e00929b640296bf276bac7c378cb93c87d509e36 for an example in the xml_generator test.

chelmuth, 2019-08-26:
Moved debug commit to e82f4ebca1aeb05434e6d46b8b2df88159f90b8b.

nfeske, 2023-05-31:
I investigated this glitch today and came to the following explanation.

The missing indentation of the second `<level1>` node is caused by the interplay with the raw append method, which treats whitespace - a newline in this case - as potentially significant data. Hence, when appending raw content, the XML generator's built-in indentation gets deliberately disabled (controlled by the `is_indented` state). The weird (or better: absence of) indentation is caused by the last `\n` character, which is inserted as is.

We could address this visual glitch using heuristics, inspecting the appended data for newline characters. E.g., with commit f3d477d4d5555f5ee020b21774ebdf89a00d2068, the output looks as follows:

```
<level0>
	<level1>
		<level2/> <!-- comment --></level1>
	<level1>
		<level2/> <!-- comment -->

	</level1>
</level0>

```

I must admit that I'm not sure whether it is worth adding this tweak. It is quite specific. E.g., should someone append a newline followed by a space, the indentation of the end tag would look strange again. It is probably best to stick to one notion of "data". Either we treat whitespace as significant and not mangle with the indentation within the node (which would, strictly speaking, mutate data), or we format appended raw content somehow beautifully. Each approach has pros and cons. I lean towards keeping the existing notion of data as opaque, including whitespace. But if commit f3d477d4d5555f5ee020b21774ebdf89a00d2068 is deemed useful - should it make the output of xkb2ifcfg much better on the eyes - I'm not against merging it.

chelmuth, 2023-06-01:
From the background of your description I think the issue can be closed. Thanks.

closed 2023-06-01 by chelmuth