genodelabs/genode #3489: Move smbios.h from os/include/os to os/incude/smbios/smbios.h

nfeske, 2019-08-28:
The header is a utility used by only two components that are local to the os repository. Its use is highly specific to x86. It should therefore not be regarded as an official os-level abstraction of Genode, which is the intention behind the os/include/os location. In fact, I'm trying to trim the content of os/include/os as much as possible.

To allow multiple components to use the utility, let's move it to os/include/smbios and copy it to the packages of each component that requires it - following the pattern of similar headers like those in os/incude/cli_monitor or os/incude/terminal.

@m-stein would you be open to do that?

m-stein, 2019-08-28:
@nfeske No problem, its on my list.

m-stein, 2019-11-04:
Fixed on my merge_to_staging.

nfeske, 2019-11-04:
Thanks. Merged to staging.

commit 98c7f665 (2019-11-04): smbios.h: move to os/include/smbios

Fixes #3489
closed 2019-11-19 by chelmuth