genodelabs/genode #3495: acpi_drv: reduce capability quota requirements

chelmuth, 2019-09-13:
The current implementation requests ACPI table mappings page-by-page with distinct I/O memory connections per page. A drafted solution could adapt https://github.com/genodelabs/genode/blob/2c47cd5c946af1ec585d2758955f8b2c4e919aa0/repos/os/src/drivers/acpi/memory.h#L69-L102 as follows.

1. Use `Allocator_avl::metadata()` for `Io_mem` book keeping.
2. Iterate over all pages of the request
- check for existing mapping with `Allocator_avl::metadata()`
- merge existing region with requested region to compound
- release existing iomem
3. Finally allocate one compound region and create **one** `Io_mem`

As a result we need capabilities for a few compound I/O memory regions only.

This issue complicates running Sculpt on HP Elitebook 840 G5 (with about 300 caps for 108 ACPI table pages).

alex-ab, 2019-09-13:
Just as note, the virtual addresses must stay the same during your re-arrangement which will be not possible if you don't rewrite the surrounding code also.

chelmuth, 2019-09-13:
Do you have a pointer to *surrounding code* which may bring complications? As I read it the allocation is done in this function only and I do not plan to change the virtual addresses of the mapping.

alex-ab, 2019-09-13:
Ah, read the code wrongly, the virtual addresses are all dense within a specific _range and the physical addresses serves as index - forget my comment.

chelmuth, 2019-09-17:
Commit a3eef3f0fa417339e96a97536d40b4d8957094f3 implements the approach drafted above and reduces the number of disjoint `Io_mem_connection`s to 44, which corresponds to ~120 caps on the test hardware menrioned. The number of connections could be reduced even further if consecutive regions would be merged too.

chelmuth, 2019-09-23:
Updated branch includes fix for #3494 too.
fba54fca1c platform/x86: support 64-bit base address registers
4b9cc54af2 platform_drv.inc: reduce acpi_drv caps to 350
9e44cc8b40 acpi_drv: reduce capability quota requirements

commit 2e3f83ea (2019-09-30): acpi_drv: reduce capability quota requirements

Map ACPI tables entirely as one dataspaces (not page-by-page). Note, the
current approach does only merge overlapping but not consecutive I/O
memory regions, which would reduce the amount of capabilities required
even further.

Fixes #3495
closed 2019-11-19 by chelmuth