genodelabs/genode #3499: libc: polling one preserved fd without timeout leads to endless loop

cnuke, 2019-09-20:
If `poll()` is used without timeout, i.e., *timeout == -1*, on (solely) one preserved fd, it will never be ready as fds without a plugin are skipped ([poll.cc#L57](https://github.com/genodelabs/genode/blob/2c47cd5c946af1ec585d2758955f8b2c4e919aa0/repos/libports/src/lib/libc/poll.cc#L57)) . This, in return, leads to calling `Libc::suspend` over and over again ([poll.cc#L78](https://github.com/genodelabs/genode/blob/2c47cd5c946af1ec585d2758955f8b2c4e919aa0/repos/libports/src/lib/libc/poll.cc#L78)).

I encountered this problem while trying to use a program in noux that polls stdin (waiting for user input) using `poll()` and could reproduce the issue by extending the libc test and executing *libc.run* without setting stdin in the test components configuration.

I am not sure how to fix the problem properly yet in a general fashion, though.

chelmuth, 2019-09-20:
The unanswered question (which is open since the libc update) is: Why do we have `poll()` and `select()` in the first place? Or: Which otherness does prevent us from implementing `poll()` on top of `select()`? And, I'd appreciate a substantial test case then.

cnuke, 2019-09-20:
Honestly, I don't remember if there were pressing technical reasons but I haven't checked any libc or rather VFS plugin in this regard. So we might as well revert this change and emulate `poll()` like we did before.

cnuke, 2019-09-23:
Commit d4519de changes the implemenation back to emulating `poll()`. So far this fixes my issue but I have not tested if it introduces any further regressions.

chelmuth, 2019-09-30:
The "poll()" implementation in libc was rolled back by 643f86c370f25c73177 but the plugin interface has still to be removed (maybe in the context of #3497).

cnuke, 2019-11-19:
Original problem was fixed by bringing back the old `poll` emulation.

closed 2019-11-19 by cnuke