genodelabs/genode #3506: stdcxx: add more missing symbols to ABI

chelmuth, 2019-09-24:
The list of missing symbols grew with the Genode Summer projects in progress, so we have to extend the ABI symbols of the library to accommodate protobuf and others.

The following command lines may help to identify the appropriate mangled symbols from your linker errors.

```sh
genode-x86-nm --dynamic --defined-only --format posix build/x86_64/debug/stdcxx.lib.so  | sed 's/ .*//' > /tmp/symbols.mangled
c++filt < /tmp/symbols.mangled > /tmp/symbols.demangled
paste /tmp/symbols.mangled /tmp/symbols.demangled | expand -t150 > /tmp/symbols
less /tmp/symbols
```

Now search your symbol, for example `/std::basic_fstream<wchar_t, std::char_traits<wchar_t> >::basic_fstream\(\)`.

@trimpim @schelmdd

trimpim, 2019-09-25:
@chelmuth 
```diff
diff --git a/repos/libports/lib/symbols/stdcxx b/repos/libports/lib/symbols/stdcxx
index a38ec5fcd..e2d9c94c6 100644
--- a/repos/libports/lib/symbols/stdcxx
+++ b/repos/libports/lib/symbols/stdcxx
@@ -863,3 +863,7 @@ _ZStlsIcSt11char_traitsIcESaIcEERSt13basic_ostreamIT_T0_ES7_RKNSt7__cxx1112basic
 _ZStrsIcSt11char_traitsIcEERSt13basic_istreamIT_T0_ES6_RS3_ W
 _ZTv0_n24_NSoD0Ev W
 _ZTv0_n24_NSoD1Ev W
+
+_ZNSt13basic_fstreamIcSt11char_traitsIcEED1Ev W
+_ZNSt13basic_fstreamIcSt11char_traitsIcEEC1EPKcSt13_Ios_Openmode W
+_ZNSi4readEPcl W
```

senier, 2019-09-26:
> The following command lines may help to identify the appropriate mangled symbols from your linker errors.
> 
> ```shell
> genode-x86-nm --dynamic --defined-only --format posix build/x86_64/debug/stdcxx.lib.so  | sed 's/ .*//' > /tmp/symbols.mangled
> c++filt < /tmp/symbols.mangled > /tmp/symbols.demangled
> paste /tmp/symbols.mangled /tmp/symbols.demangled | expand -t150 > /tmp/symbols
> less /tmp/symbols
> ```
> 
> Now search your symbol, for example `/std::basic_fstream<wchar_t, std::char_traits<wchar_t> >::basic_fstream\(\)`.
> 
> @trimpim @SchelmDD

Alternative, quicker approach:

```sh
$ COLLECT_NO_DEMANGLE=1 make run/test/whatever
```

This will make the linker output unmangled symbols which can go directly into the symbols file.

trimpim, 2019-10-30:
@chelmuth sorry for coming so late.

While porting `grpc` I have found some more missing symbols:

```
diff --git a/repos/libports/lib/symbols/stdcxx b/repos/libports/lib/symbols/stdcxx
index e2f61349f..dbeb7ef95 100644
--- a/repos/libports/lib/symbols/stdcxx
+++ b/repos/libports/lib/symbols/stdcxx
@@ -970,3 +970,7 @@ _ZTv0_n24_NSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED0Ev W
 _ZTv0_n24_NSt7__cxx1119basic_ostringstreamIcSt11char_traitsIcESaIcEED1Ev W
 _ZTv0_n24_NSt7__cxx1119basic_ostringstreamIwSt11char_traitsIwESaIwEED0Ev W
 _ZTv0_n24_NSt7__cxx1119basic_ostringstreamIwSt11char_traitsIwESaIwEED1Ev W
+
+_ZNSt7__cxx1118basic_stringstreamIcSt11char_traitsIcESaIcEEC1ERKNS_12basic_stringIcS2_S3_EESt13_Ios_Openmode W
+_ZTISt12bad_weak_ptr W
+_ZTVSt12bad_weak_ptr W
```

chelmuth, 2019-10-30:
Merged as 866017c4bcb50f67cfb96bb5bcbb75178c030b26.

commit 7e0e6fce (2019-10-10): stdcxx: add more missing symbols to ABI

Fixes #3506
closed 2019-11-19 by chelmuth