genodelabs/genode #3510: trace: subject_id of 0

alex-ab, 2019-09-29:
Subject_id objects can be instantiated without actually referencing a valid Subject, e.g. by the default constructor https://github.com/genodelabs/genode/blob/master/repos/base/include/base/trace/types.h#L68. Subject_id ```0``` is used as kind of invalid ID.

The Subject registry however handles ```0``` as a normal ID,
https://github.com/genodelabs/genode/blob/master/repos/base/src/core/include/trace/subject_registry.h#L353.

To fix the ambiguity either we should remove the default constructor (so not using an invalid id) or don't let the registry use 0 as valid id.

What is your preference ?

m-stein, 2019-09-30:
I would vote for not mixing invalid and valid IDs in the same type and instead add a member ``valid`` which makes it clear whether to de-reference a subject ID or not.

nfeske, 2019-09-30:
@m-stein learning from Ada/SPARK, right? ;-) I like it.

I cannot tell for sure but I guess the default constructor is solely there to support the construction of an array of subject IDs. But that could also be achieved by using an array of `Constructible` objects. Here, the validity would correspond to whether an element is constructed or not.

Maybe we could try removing the default constructor first, and look after the compile issues that follow from that? Let's try to apply the constructible pattern there. If we are lucky, we may not even need the notion of an "invalid" ID at all?

m-stein, 2019-09-30:
@nfeske Got me :)

Constructible would bring the advantage of not being able to ignore the validity state.

alex-ab, 2019-10-04:
The constructible approach seems to be not viable given the design of the trace session interface. The Subject_id is transferred via rpc, see https://github.com/genodelabs/genode/blob/master/repos/base/include/trace_session/client.h#L84. This would require a change of the interface, since you don't want to memcpy constructible objects, do you ?

alex-ab, 2019-10-04:
Given the fact I don't will have time to change anything in this regard in near future, 69c0cd1f701cc4f2cd7eb922d899b2406257dde4 avoids the allocation of 0 as valid id.

nfeske, 2019-10-07:
@alex-ab I completely missed the use of the subject IDs over the RPC interface. This renders the constructible approach impractical, indeed. Thanks for your patch. Maybe we can think though the general problem of managing invalid ID (picking up @m-stein's suggestion) as a separate discussion later.

m-stein, 2019-11-04:
Just as a general question out of curiosity: Why would I refrain from sending a consructible object via RPC given that the payload object is copyable and of sensible size (which should both be the case for plain IDs)? I seem to miss something.

m-stein, 2019-11-04:
Are Constructible/Reconstructible only non-copyable in general to prevent hiding of the potential noncopyable-ness of payload objects? Because Constructible/Reconstructible themselves should be copyable AFAICS.

nfeske, 2019-11-04:
> Are Constructible/Reconstructible only non-copyable in general to prevent hiding of the potential noncopyable-ness of payload objects?

Yes. The noncopyability of the payload object would not be respected otherwise, which is dangerous.

m-stein, 2019-11-04:
> which is dangerous.

Doubtless. One could add another template parameter to (Re)Constructible to adapt to copyable payload objects. But, of course, this is currently not on any critical path.

chelmuth, 2019-12-20:
With 1297a8f merged to master this issue is fixed in my opinion. Can we close it now?

closed 2019-12-20 by alex-ab