genodelabs/genode #3511: File_system: increase resolution of file modification timestamps

cnuke, 2019-09-30:
The newly added modification timestamp support (#1748) uses `seconds` as unit. For practical reasons it makes sense to increase its resolution. Using `nanoseconds` would give us around *292* (and *243* if we keep the Unix epoch as starting point) years in both directions if we stick to using solely `int64_t` as type. I am not aware of current Genode use-cases that would require this kind of granularity, so I rather lean torwards using `milliseconds`.

That being said, the `File_system::Timestamp` is already an object - adding an interface along the lines of `seconds`, `nanoseconds` and `resolution` (which contains the actual resolution of the of the value returned by `nanoseconds`) would allow us to keep the conversion code contained. In addition I would like to add a `valid` method and to keep handling the invalid state hidden as well.

nfeske, 2019-09-30:
Thanks for opening this issue. I agree to all the points you raised.

chelmuth, 2024-04-26:
I'd love to see this issue addressed after its year-long slumber ;-)

nfeske, 2025-03-20:
Addressed by commit https://github.com/nfeske/genode/commit/2b871157b9db7e954c3e9db940e5de99055823de. @cnuke may you have a look?

nfeske, 2025-03-20:
Fixup https://github.com/nfeske/genode/commit/ee93308bc649aa6d55d328df96428d221b67fb22 adds the unit conversion in vfs_rump stat, and silences conversion warnings.

nfeske, 2025-03-20:
I just tested the change on Sculpt based on staging via the following command in the inspect view:
```
$ while true; do echo x > test; ls -la --full-time test; done
```
It turns out that the time still increases in steps of seconds on the ext2-based ahci fs, which is apparently the precision of ext2. When executing the command on the ram-based config fs, the milliseconds granularity can be observed.

So for most practical matters, the increased resolution won't be a visible improvement. @cnuke, @chelmuth shall we still merge this change?

cnuke, 2025-03-20:
> Addressed by commit [nfeske@2b87115](https://github.com/nfeske/genode/commit/2b871157b9db7e954c3e9db940e5de99055823de). @cnuke may you have a look?

Thanks, I will take a more closer look tomorrow.

> So for most practical matters, the increased resolution won't be a visible improvement. @cnuke, @chelmuth shall we still merge this change?

From solely skimming the changes they are clearer stating the intent still, so I would not mind merging them.

chelmuth, 2025-03-21:
As I understand the history of Ext2/3/4, sub-second timestamps where not available before Ext4, which stores additional bits in the second half of 256-byte inodes (e.g., the mtime_extra field). More recent NetBSDs are able to parse the extra bits, but the version used in our Rump port is too old.

commit fe8447df (2025-03-21): os: increase mtime resolution to milliseconds

This patch changes the 'Timestamp' type of the VFS and file-system
session to an unsigned 64-bit value holding the number of milliseconds
since the UNIX epoch (beginning of 1970).

The special case of an invalid timestamp has been removed as it was
never consistently handled anyway.

In contrast to the POSIX timespec, which represents tv_sec and tv_nsec
both as signed values, this patch uses an unsigned value because the
interpretation of negative tv_sec and tv_nsec values is muddy.
Hypothetical modification times older that 1970 are capped at 197 [... cut]