genodelabs/genode #3513: dde_zircon port source no longer available

chelmuth, 2019-10-04:
The mirror of zircon at https://github.com/fuchsia-mirror/zircon.git vanished and I don't know of any alternative download location. @jklmnn do do you have any ideas? Otherwise, I would remove the repository.

jklmnn, 2019-10-07:
I found a Zircon fork on GitHub and forked it myself. So the source is available at  https://github.com/jklmnn/zircon. I added this in https://github.com/jklmnn/genode/commit/6a77e628c18c61ead8fb0f251909857ba942a565.
It seems that Google decided to merge all the different fuchsia repos into a single one so all future zircon code will stay there (although I don't know of a GitHub mirror of this repository).

chelmuth, 2019-10-07:
Thanks, merged the commit just now.

commit 73110197 (2019-10-07): use zircon fork

fixes #3513
closed 2019-11-19 by chelmuth