genodelabs/genode #3516: fix compile warnings in test/vfs_stress

nfeske, 2019-10-07:
There are a couple of compile warnings about possibly unintended fall-though cases in switch statements.

nfeske, 2019-10-07:
Commit https://github.com/nfeske/genode/commit/b13aa8ad626c7e4dfa1dee78ac9ce43f848999c4 removes the warnings. However, I'm not completely sure about possibly intended fall-though cases. @ehmry I wonder, is the patch ok from your perspective?

ehmry, 2019-10-07:
I believe all these fall-throughs are intentional, so I have a fixup to annotate that at 16cabfa76367720d2a79110108343b7ad0bb16c5.

nfeske, 2019-10-08:
@ehmry thank you for the quick response and your patch. I'm grateful that you prevented my ill-conceived patch from entering staging. ;-)

commit ef8bdaf3 (2019-10-08): test/vfs_stress: annotate switch label fall-throughs

This test was written for speed, not correctness.

Fixes #3516
closed 2019-11-19 by chelmuth