genodelabs/genode #3519: libc: implement ioctl operations via pseudo files

nfeske, 2019-10-09:
Currently, ioctl operations of "device" files are passed through the VFS interface, which has a number of limitations:

* As the ioctl arguments are opaque, the interface is muddy.
* Operations cannot transparently be executed across file-system server boundaries because Genode's `File_system::Session` has no notion of ioctls.

To overcome these limitations, I want to establish a scheme that follows the lines of the socket fs. Similar to how the socket fs maps socket calls to pseudo-file accesses, the libc could decode ioctls locally and map them to pseudo-file accesses.

A device file can be accompanied with a (hidden) directory that is named after the device file and hosts pseudo files for triggering the various device operations. For example, for accessing a terminal, the directory structure looks like this:

```
  /dev/terminal
  /dev/.terminal/info
```

The 'info' file contains device information in XML format. The type of the XML node corresponds to the device type. E.g., If the libc receives a `TIOCGWINSZ` ioctl for /dev/terminal, it reads the content of /dev/.terminal/info to obtain the terminal-size information. In this case, the `info` file looks as follows:

```
  <terminal rows="25" columns="80/>
```

Following this scheme, VFS plugins can support ioctl operations by providing an ioctl directory in addition to the actual device file. The layout of the pseudo file systems provided by the VFS plugins is a convention by the libc.

nfeske, 2019-10-09:
Commit https://github.com/nfeske/genode/commit/bcc11c6a4bb116d50a2496361d5294109ed17472 implements the libc part for the `TIOCGWINSZ` operation.

cnuke, 2019-10-10:
I can see the appeal of using XML as it makes parsing easier (and more robust as the code is well tested) but at the some time makes using this interface from tools that do not understand XML difficult. That's why I did not follow down that road but was probably pre-mature as the prominent use-case will not be someone using the shell for doing ioctls.

nfeske, 2019-10-10:
@cnuke thanks for chiming in. I don't see the use of XML as a contradiction. A VFS plugin can provide both an XML representation that can conveniently be processed by a program and a number of pseudo files for properties that may be of interest for shell scripts. For example, the terminal VFS plugin may feature the pseudo files `rows` and `columns` in addition to `info`. The former would simply contain a number.

Anecdotally speaking, in the socket fs, the omission of XML led to the need for custom parsing code, which leaves a rather fragile impression on me. The libc code is certainly more complex than it needs to be for supporting the socket API. The principle feature of performing TCP from shell scripts - while nice in theory - has remained unused in practice. The bottom line is that the restraint from using XML had not benefited us so far in this case.

The approach of using an XML-based info file originally came into my mind because of the need to watch for changes. When changing the terminal size, we need to deliver a POSIX signal to the terminal-using application. It makes sense to deliver such events by watching a pseudo file. If we have the properties scattered in a number of files (e.g., `rows`, `columns`), the application would need to watch all of them. When the terminal size changes, most often than not, both watched files would trigger, resulting in more than one POSIX signal. In contrast, if we have all properties present in an `info` file, we can simply watch this single file and process multiple properties as one atomic step.

cnuke, 2019-10-10:
Fair enough, providing additional pseudo-files would eleminate this difficulty.

nfeske, 2019-11-19:
Fixed in master.

closed 2019-11-19 by nfeske