genodelabs/genode #3526: nic_router: problems with arbitrary downlink buffer sizes

m-stein, 2019-10-15:
When using buffer sizes from 32 x PKT_SIZE to 64 x PKT_SIZE for the Bad_args_nic in the nic_router_stress test, the nic router has an error:
``Error: attempt to allocate zero-size block from heap``

When using a buffer size of 0 these errors occur in the nic router:
``Error: Session_env: less X quota consumed than expected``

nfeske, 2021-02-26:
@m-stein I wonder, is this issue still important?

m-stein, 2021-03-01:
The issue seems to be mitigated now.

When using buffer sizes from 20 * PKT_SIZE to 64 * PKT_SIZE, the test now succeeds and the router merely complains (non-critical):
```
[init -> nic_router] Error: Session_env: more RAM quota consumed than expected
[init -> nic_router] Error: Session_env: more CAP quota consumed than expected
```

When using buffer sizes of 1 * PKT_SIZE to 20 * PKT_SIZE, the router, although not halting, unexpectedly denies the session requests:
```
[init -> nic_router] Warning: unexpected exception during Nic-session creation
[init -> nic_stress_2] Error: Nic-session creation failed (ram_quota=0, cap_quota=0, tx_buf_size=25600, rx_buf_size=25600, label="bad_args")
[init -> nic_stress_2] Error: Construct_destruct_test failed
[init -> nic_router] Warning: unexpected exception during Nic-session creation
[init -> nic_stress_1] Error: Nic-session creation failed (ram_quota=0, cap_quota=0, tx_buf_size=25600, rx_buf_size=25600, label="bad_args")
[init -> nic_stress_1] Error: Construct_destruct_test failed
```

When using a buffer size of 0, the router prints a lot of rather unhelpful but non-critical errors and finally denies the session requests:
```
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_router] Error: Session_env: less CAP quota consumed than expected
[init -> nic_router] Error: Session_env: less CAP quota replenished than expected
[init -> nic_stress_1] Error: Nic-session creation failed (ram_quota=0, cap_quota=0, tx_buf_size=0, rx_buf_size=38400, label="bad_args")
[init -> nic_stress_1] Error: Construct_destruct_test failed
[init -> nic_stress_2] Error: Nic-session creation failed (ram_quota=0, cap_quota=0, tx_buf_size=0, rx_buf_size=38400, label="bad_args")
[init -> nic_stress_2] Error: Construct_destruct_test failed
```

m-stein, 2021-03-01:
That said, I think that this issue is still worth fixing it but admittedly, I have not planned to do it in the near future. Taking in account also the date of issue creation, I'd say we close it and come back to it in the course of our general "NIC-Router-Consolidation" endeavor.

closed 2021-03-01 by m-stein