genodelabs/genode #3537: ARMv8: Improve support on Genode

ssumpf, 2019-10-24:
Packages

  genodelabs/bin/arm_v8a/gnumake/2019-09-28
  genodelabs/bin/arm_v8a/gmp/2019-09-28
  genodelabs/bin/arm_v8a/gcov/2019-09-28
  genodelabs/bin/arm_v8a/test-xml_generator/2019-09-28
  genodelabs/bin/arm_v8a/rump/2019-09-28
  genodelabs/bin/arm_v8a/qt5_webkit/2019-09-28
  genodelabs/bin/arm_v8a/gtotp_report/2019-09-28

  Builds

  make.prg
  gcov-libcpp.lib
  gcov-libiberty.lib
  gmp-mpf.lib
  gmp-mpq.lib
  gmp-mpz.lib
  libgcov.lib
  basic.prg (virtualkeyboard)
  calculatorform.prg
  mixer_gui_qt.prg
  openglwindow.prg
  qt5_jscore.lib
  qt_launchpad.prg
  samegame.prg
  test-qpluginwidget.prg
  test-qt_quick.prg
  test-tiled_wm-app.prg
  test-tiled_wm-overlay.prg
  test-tiled_wm-panel.prg
  testqstring.prg
  tetrix.prg
  textedit.prg
  builtins-rust.lib
  liblibc-rust.lib
  librand-rust.lib
  librustc_unicode-rust.lib
  libunwind-rust.lib

  Tests

  demo
  depot_download
  fb_bench
  fetchurl_lwip
  fetchurl_lxip
  noux_tool_chain_auto
  rump_ext2

ssumpf, 2019-10-29:
Related for GCOV: https://github.com/cproc/genode-generated/issues/1

ssumpf, 2019-11-04:
@chelmuth: ace7a12 fixes the mpfr API package for all platforms, please consider for staging.

ssumpf, 2019-11-04:
@chelmuth: f349198 fixes the LICENSE file for the mpc package.

ssumpf, 2019-11-04:
@chelmuth: 309dea5  removes the bogus include

nfeske, 2019-11-06:
noux-tool_chain_auto on NOVA x86_64 reports the following error:

```
[2019-11-06 07:11:48] [init -> noux -> /libexec/gcc/x86_64-pc-elf/8.3.0/cc1plus] [31mError: LD: jump slot relocation failed for symbol: '__gmpn_invert_limb'[0m[0m
```

ssumpf, 2019-11-06:
@nfeske: Okay, I am looking into it.

ssumpf, 2019-11-06:
c6671e3 adds the `invert_limb` function to x86_64

nfeske, 2019-11-06:
Thanks. I added your fixup to staging.

chelmuth, 2019-12-20:
Only qt5_webkit is still missing, which is another issue.

closed 2019-12-20 by chelmuth