genodelabs/genode #3543: expanding_reporter: configure initial buffer size

cnuke, 2019-10-28:
Currently, the buffer initially starts with `4096` and gets expanded by `4096` bytes on every resize attempt. When I used the expanding reporter to dump the internal state of a component, which generates the report programmatically by traversing a tree, this lead to a somewhat unexpected run time (for better or - in this case definitely - the worse every new reporting attempt started again from the top of the tree).

I briefly tweaked the expanding strategy (instead of using a fixed value double the size on every attempt) but in the end settled on setting the initial buffer size to a value that I know covers my usage pattern.

(Using a report in such a way if one expects a large amount of data is arguably not the proper approach.)

nfeske, 2019-10-28:
This is a reasonable feature. However, I'd like to see the argument passed as a dedicated type `struct Initial_buffer_size { size_t value; };` declared in the class scope. Granted, this is a bit bureucratic, but it makes the value more clear at the caller site.

cnuke, 2019-10-28:
No problem, I adjusted the commit - see a7a4bde.

nfeske, 2019-10-28:
Cool, thanks! Merged to staging.

commit e7d38577 (2019-10-28): expanding_reporter: configure initial buffer size

Fixes #3543.
closed 2019-11-19 by chelmuth