genodelabs/genode #3547: vfs server: possible infinite blocking due to queued packet handling

nfeske, 2019-11-01:
When executing `ls -laR /` in the bash.run scenario, the execution of `ls` can get stuck mid way. I tracked down the issue to the repeated calls of `getdirentries`. At one point, such a call won't return. In most cases, this happens once a complete directory is scanned, when `ls` prints the result, and starts reading a new one.

After further investigation together with @chelmuth, we identified the problem to be at the VFS server side. In the presence of both a read request (reading an entry from a directory) and a large number of write requests (many lines written to `/dev/terminal`), the VFS server would leave the read request unanswered.

nfeske, 2019-11-01:
Commit https://github.com/nfeske/genode/commit/82cc211bc7cc852afaa29eaa6c6240d2963cca3f fixes the issue. Refer to the commit message for further information.

nfeske, 2019-11-05:
My original low-invasive attempt to fix the issue did not work well in other scenarios. In particular, it interfered with file watching in ways that I did not see coming and which are extremely hard to interpret with the given VFS code structure.

This prompted me to simplify the internal workings of the VFS server in commit https://github.com/nfeske/genode/commit/45b7a2bf7696221772108df4ca23d6dab8421cd7. Besides solving the issue at hand, the commit will hopefully ease the future maintenance and optimization of the VFS server.

Note that the commit depends on the two minor commits https://github.com/nfeske/genode/commit/5f3a0aa800d0f3c0bc45bd423dff8109e6159afa and https://github.com/nfeske/genode/commit/8a45af43be46ae321aac77370745db4a58ed8ebb.

There is one aspect I'm bit uncertain about: While restructuring the code, I came across the SEEK_TAIL handling, which did not make any sense to me. I changed this code to always seek to the end of the file but I'm pretty sure I'm missing something. @ehmry Can you remember the reasoning behind overwriting the file from the end of it when SEEK_TAIL is specified as position? The comment at https://github.com/genodelabs/genode/blob/master/repos/os/include/file_system_session/file_system_session.h#L86 is consistent with my intuitive understanding. But I cannot grasp what the code at https://github.com/genodelabs/genode/blob/master/repos/os/src/server/vfs/node.h#L613 accomplishes that seeking to the end would not?

nfeske, 2019-11-06:
The new commit https://github.com/nfeske/genode/commit/2d0158d3bc9efeb89753f81f5eae84e4945a4aac back-pedals my prior change of the symlink-write mechanism and maintains the original approach of enforcing the null termination at the server side. The symlink content must be buffered anyway (for re-attempted writes as response to a saturated I/O buffer) and cannot be taken from the (untrusted because shared memory) packet-stream buffer. However, in contrast to the original code, which allocated the buffer on the stack, the new version allocates the buffer inside the `Symlink` node as the content is needed not just at `submit_job` time but during the `execute_job` stage.

The new version further cleans up the code by removing the dependency from the `file_system/node.h` header, which originates from the ram_fs but provides no useful functionality for the VFS server. The outstanding clarification of `SEEK_TAIL` notwithstanding, I marked the issue as fixed.

ehmry, 2019-11-07:
@nfeske Overwriting the last piece of a file with SEEK_TAIL is wrong. It looks like an erroneous attempted to deduplicate code in `_read` and `_write` at   a635873568e2c7292f95dd3087baa09be8515a8a.

nfeske, 2019-11-08:
@ehmry thanks for investigating and for the clarification.

nfeske, 2019-11-19:
Fixed in master

closed 2019-11-19 by nfeske