genodelabs/genode #3554: Clean up vm_session interface

skalk, 2019-11-15:
While working on issue #3553 I recongnized that _create_vcpu and _exception_handler has to be called by all different platforms on the library side before using a new vcpu. One platform first called the exception handler registration the next one first created the vcpu.
To limit the variance here, in my opinion it is better to combine _exception_handler into _create_vcpu.

Moreover, I saw that several RPC functions in the Vm_session are beginning with an underscore, namely: _cpu_state, _exception_handler, _run, _pause, _create_vcpu
Although, these are public functions, which clashes with Genode's coding style. The intention behind this was as far as I understood that nobody uses those function directly, but uses the library functions inside the base-library. Nevertheless, there are a lot of examples where you should not use code like syscalls, IPC functions etc. directly, but as long as those are public class methods, it should not be prefixed by an underscore.

chelmuth, 2021-02-03:
I approached this issue from a different angle while working on the VirtualBox 6 port and also also the broader scope of *Refactor VM/vCPU API*. In this approach the VM session is just the basis of the VM-specific part of the base library and common denominator of all platforms. It's task is VM-resource management speak memory, vCPUs and the ARM-specific interrupt-controller attachment. Functions on virtual CPUs beyond this are the duty of the native vCPU object, which is used via a dedicated API and not directly via RPCs from the VMMs. This decision stems from the vastly differing implementations for the supported x86 platforms NOVA, Fiasco.OC, and seL4.

What I changed in a nutshell...

- The VM session is pruned to a pure VM RPC interface and supports the creation of `Native_vcpus`, which implement platform-specific RPC interfaces.
- API code is moved from `Vm_session_client` into `Vm_connection` and `Vm_connection::Vcpu`. *vm_session/client.h* is removed.
- vCPUS are created by instantiating `Vm_connection::Vcpu` objects that provide a clean interface and hide the platform specifics from the VMM.
- vCPU exit configuration changes from whitelist-style callback to construction-time omission declaration. (Currently no state omission is implemented but the first candidate is the FPU.)
- `Genode::Vm_state` becomes `Genode::Vcpu_state` and is granted an interface overhaul with documentation.
- `Vcpu_state` is part of the `Vcpu` object.
- `Genode::Vm_handler` becomes `Genode::Vcpu_handler` and is iteratively simplified to a pure interface (see also below).
- The VM library is no longer part of *base-common* (which is used by core too) but *base*.

I plan to also address the following while continuing the vbox6 port.

- Utilities and functionality in `Vcpu_handler` (i.e., signal-context capability, semaphore, member-function pointer) is highly platform dependent and cannot be implemented in a generic way. Therefore all this code will be moved in the several VM libraries. If this introduces code duplication, the common code can be placed in *base/src/include/base/internal* or a similar place.
- I imagine the transition model between VM and VMM to become easier to understand: The vCPU runs after `Vcpu::run()` is requested. On VM exception/exit `Vcpu_handler::handle_exit()` is called and the vCPU state is available via `Vcpu::state()`. The VM is entered again upon return from `handle_exit()`. VM entry loads either all vCPU state or omits some parts blacklisted by the `handle_exit()` return value.
- With `Exit_config` and the `handle_exit()` return value `Vcpu_state` could be simplified to a plain structure of values as the omission information is carried by other means.

My current state of work can be found at b637faecc78f8f65ca90f7143f33c89500efb847. Currently, I finished the implementation for x86 and NOVA only, but the vbox5_vm tests are functional.

@alex-ab @skalk What's your opinion about my proposal?

alex-ab, 2021-02-09:
Thanks for your efforts. I'm fine in principal. The only advise I can give and wish I have, is in the context of the exit configuration look not solely at Vbox, because it is a bad adviser. When you check Seoul carefully, you will see a VMM which exactly knows which state it requires on an exit (instead taking it all as VBox does).

skalk, 2021-02-11:
@chelmuth I'm sorry for the long delay in response. It was not intended.

I appreciate the direction of moving the Vcpu into a separate interface a lot! And I agree with all subsequent steps you've sketched.

alex-ab, 2021-02-16:
@chelmuth: 09b9611c01e637354d137beabf64a531888fa8c6 refactors the seL4 version according to your changes, ae144c298bf45adb74dac286ce930562ac4f7ff9 is additional required to avoid pure virtual exceptions

chelmuth, 2021-02-16:
Could you identify the pure virtual function in this case?

alex-ab, 2021-02-16:
No.

alex-ab, 2021-02-19:
The vm_session branch now contains the foc specific adjustments and seoul is working again. I added for base-hw also proposal commits. With them, vmm_arm on imx8 works again. tz_vmm I did not test by now on hardware. In case the base-hw should be considered, it would be good that @skalk also have a look.

alex-ab, 2021-02-19:
I noticed that in the nova case the Id_space is used to lookup the vcpu object on a vm exit. The used internal Mutex/Lock will corrupt in the contention case the utcb (and thereby the guest state). We must use a lock free way to do the lookup. Additionally, we have to avoid too high level Genode code in the early exit stage to avoid corruption of the fpu state, which is the one of the guest.

alex-ab, 2021-02-19:
> The used internal Mutex/Lock will corrupt in the contention case the utcb (and thereby the guest state). 

After checking again,  the one used member tls (base/internal/spin_lock.h) of the utcb is before the guest state, so it should be save to use a Mutex.

chelmuth, 2021-02-19:
Just for my understanding...

- The previous implementation used `Genode::Registry` and thereby also uses a mutex to protect integrity. Did you suspect the cause of the issues with stalled CPUs in this?
- The use of the mutex is safe in case of contention and without, right?

alex-ab, 2021-02-19:
> * The previous implementation used `Genode::Registry` and thereby also uses a mutex to protect integrity.

Indeed.

> Did you suspect the cause of the issues with stalled CPUs in this?

No, that was not the reason. I just reviewed the code and that came to my mind that it may be a issue.

> * The use of the mutex is safe in case of contention and without, right?

I solely looked at the nova specific spinlock part, and this case a member before the guest state is used on the utcb.

chelmuth, 2021-02-21:
Thanks for your efforts, I'm in the process of review and merge. While reading your adaptions for base-hw I discovered *hw_native_vcpu/hw_native_vcpu.h* is missing, but also got the impression that it's not needed. Am I right?

chelmuth, 2021-02-21:
Now I see I mixed up create_vcpu() result and native_vcpu(). The RPC object is actually used to get hold of the vCPU kernel capability and state. Shouldnt the RPC function be named more like `Native_vcpu::kernel_cap()` then?

alex-ab, 2021-02-21:
@chelmuth: sorry, forgot to add the new file to git, see d4288a3b9af11fa203088456f2a1d777b12ce100

chelmuth, 2023-09-13:
Should we close this issue in favor of #4968?

chelmuth, 2023-10-04:
From my perspective #4968 addresses all topics of this issue and was merged to master. Therefore, I close this issue. Further topics deserve dedicated issues.

closed 2023-10-04 by chelmuth