genodelabs/genode #3555: depot_autopilot: rework log matching

m-stein, 2019-11-19:
The current log-matching algorithm has several drawbacks:
* It does sanitizing mixed into the matching loop and only temporarily.
* It doesn't recognize patterns like "AAB" when the input is "AAAB" because on mismatches it doesn't jump back to an earlier character on the input.
* It is hard to grasp and debug because it consists of several nested loops with a lot of character pointers.
* it iterates the input for each pattern anew.

m-stein, 2019-11-21:
This fixes the core issues:
38e0b722ab depot_autopilot: rework log matching

However, things could get more elegant by, for instance, removing Log_event::Plain_string and re-use Expanding_string instead or by trying to reduce the copy operations (there are more now). But, IMO, this has a low priority for me currently.

m-stein, 2019-11-22:
@chelmuth Thanks for the detailed review!
I've added a fix-up on my merge_to_staging branch.

chelmuth, 2019-12-20:
Merged to master.

closed 2019-12-20 by chelmuth