genodelabs/genode #3557: depot_query: pagefault with 19.11

alex-ab, 2019-11-21:
```
[runtime -> extract] extracted '/public/alex-ab/bin/x86_64/menu_view/19.11.tar.xz' to '/depot/alex-ab/bin/x86_64/menu_view'
[runtime -> extract] extracted '/public/alex-ab/bin/x86_64/top_view/19.11.tar.xz' to '/depot/alex-ab/bin/x86_64/top_view'
[runtime -> extract] extracted '/public/alex-ab/bin/x86_64/graph/19.11.tar.xz' to '/depot/alex-ab/bin/x86_64/graph'
[runtime -> update -> dynamic] child "extract" exited with exit value 0

[core] no RM attachment (READ pf_addr=0x20000 pf_ip=0x1018a2a from pager_object: pd='init -> runtime -> depot_query' thread='ep')
[core] page fault, pd='init -> runtime -> depot_query' thread='ep' cpu=0 ip=0x1018a2a address=0x20000 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1

[runtime -> update -> manager] installation complete.
```

objdump leads to:

```
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:524
 1018a0d:       48 83 c1 01             add    $0x1,%rcx
 1018a11:       48 83 85 00 fa ff ff    addq   $0x1,-0x600(%rbp)
 1018a18:       01 
 1018a19:       48 8b 85 d8 f9 ff ff    mov    -0x628(%rbp),%rax
 1018a20:       48 8b bd 00 fa ff ff    mov    -0x600(%rbp),%rdi
 1018a27:       48 89 c2                mov    %rax,%rdx
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:519
 1018a2a:       0f b6 44 38 ff          movzbl -0x1(%rax,%rdi,1),%eax
 1018a2f:       48 01 fa                add    %rdi,%rdx
 1018a32:       48 89 95 d0 f9 ff ff    mov    %rdx,-0x630(%rbp)
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:520
```

The ep of depot_query dies during handling some VFS related content:

```
510 .   .   template <typename STRING, typename FN>
511 .   .   void for_each_line(FN const &fn) const
512 .   .   {             
513 .   .   .   char const *src           = _buffer.ptr;
514 .   .   .   char const *curr_line     = src;
515 .   .   .   size_t      curr_line_len = 0;
516                             
517 .   .   .   for (size_t n = 0; ; n++) {
518                             
519 .   .   .   .   char const c = *src++;
520 .   .   .   .   bool const end_of_data = (c == 0 || n == _buffer.size);
521 .   .   .   .   bool const end_of_line = (c == '\n');
```

alex-ab, 2019-11-27:
It happened again. This time the pf_addr is 0, but the source code line is the same in the os/vfs.h file:

```
[runtime -> verify] /public/genodelabs/pkg/rom_filter/2019-11-18.tar.xz: good signature
[runtime -> extract] extracted '/public/genodelabs/pkg/rom_filter/2019-11-18.tar.xz' to '/depot/genodelabs/pkg/rom_filter'
[runtime -> update -> dynamic] child "extract" exited with exit value
[leitzentrale -> manager] genodelabs/pkg/chroot/2019-11-18 incomplete or missing
[core] no RM attachment (READ pf_addr=0x0 pf_ip=0x100fba0 from pager_object: pd='init -> runtime -> update -> dynamic -> depot_query' thread='ep')
[core] page fault, pd='init -> runtime -> update -> dynamic -> depot_query' thread='ep' cpu=0 ip=0x100fba0 address=0x0 stack pointer=0xa02fef18 qualifiers=0x4 irUwp reason=1
```

```
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:524
 100fb98:       48 83 c2 01             add    $0x1,%rdx
 100fb9c:       49 83 c4 01             add    $0x1,%r12
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:519
 100fba0:       43 0f b6 44 25 ff       movzbl -0x1(%r13,%r12,1),%eax
 100fba6:       4f 8d 74 25 00          lea    0x0(%r13,%r12,1),%r14
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:520
 100fbab:       84 c0                   test   %al,%al
 100fbad:       75 d9                   jne    100fb88 <_ZN11Depot_query4Main28_collect_binary_dependenciesERKN6Genode6StringILm100EEERNS_12DependenciesENS_15Recursion_limitE+0x398>
/home/alex/genode/sculpt_release/repos/os/include/os/vfs.h:528
 100fbaf:       48 85 d2                test   %rdx,%rdx
```

The interesting part is that the chroot package was not downloaded at all, even not attempted, according to the log.

alex-ab, 2019-11-27:
The setup is to use a fresh usb stick and copy sculpt via dd to it. Booting Sculpt, expanding the usb stick partition, using the usb partition as depot location and then select shared_fs in the runtime menu.

chelmuth, 2019-11-27:
I spotted one possible issue: If the size of the inspected file is `0` then the buffer pointer may become `nullptr`. Could you look into the given directory if there exists a null-sized file?

alex-ab, 2019-11-27:
I don't know which directory, but in case the genodelabs/pkg/chroot/2019-11-18 was tried, chroot and chroot/2019-11-18 did not exist on the usb stick.

nfeske, 2019-11-29:
Thanks @alex-ab for the steps! Even though I was not able to directly reproduce it with exactly these steps, I eventually found a way to manually provoke it: When "unusing" the Genode partition and then overwriting an existing `archives` file with a zero-length file `echo -n "" > pkg/chroot/2019-11-21/archives`, and then "using" the partition again, the page fault occurs.

nfeske, 2019-11-29:
Commit https://github.com/genodelabs/genode/commit/586d249186341b54587de60fa7eb679e1bc96224 triggers the problem synthetically.
Commit https://github.com/genodelabs/genode/commit/3a789047e861bc4ee4a137cd5247a7f59f455df5 fixes it.

alex-ab, 2019-11-29:
Great !

commit 3a789047 (2019-11-29): os/vfs.h: handle zero-sized files

The 'File_content' class failed to consider zero-sized files.

Fixes #3557
commit f85ec313 (2019-12-19): os/vfs.h: handle zero-sized files

The 'File_content' class failed to consider zero-sized files.

Fixes #3557
closed 2019-12-19 by chelmuth