genodelabs/genode #3568: TCP server using lwip as well as lxip blocks after repeated connections

sidhussmann, 2019-11-29:
A simple TCP server blocks after repeated connections. The server can handle at most 1029 connections. Sometimes less.
`run/tcp_stress_lwip` and `run/tcp_stress_lxip` show this problem. [1]

Occasionally I get following additional log message using lxip:
```
[init -> init -> tcp_stress_client] echo 51
[init -> init -> tcp_stress_server] closed
[init -> init -> tcp_stress_server] read to accept
[init -> init -> tcp_stress_server] accepted 165
[init -> init -> tcp_stress_server] readed echo 52
[init -> init -> tcp_stress_server] written
[init -> init -> tcp_stress_client] echo 52
[init -> init -> tcp_stress_server] BUG: failure at /home/user/genode/contrib/dde_linux-234e76f366dc583176c0695d33886fb6576765c2/src/lib/lxip/include/linux/skbuff.h:2217/skb_orphan()!
```

To debug this using gdb on base-linux, I created in `genode/build/x86_64/debug` a symbolic link `ld.lib.so` to `ld-linux.lib.so` and started gdb in this directory using `sudo gdb -p <PID> tcp_stress_server`. Here its output:

```
Reading symbols from tcp_stress_server...
Attaching to program: /home/user/genode/build/x86_64/debug/tcp_stress_server, process 52128
[New LWP 52131]
[New LWP 52132]
[New LWP 52133]
[New LWP 52134]
Cannot access memory at address 0x482feb02ebe0ff67
Cannot access memory at address 0x482feb02ebe0ff5f
Cannot access memory at address 0x482feb02ebe0ff67
Cannot access memory at address 0x482feb02ebe0ff67
Cannot access memory at address 0x482feb02ebe0ff5f

warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
0x00000000010d9530 in ?? ()
(gdb) bt
#0  0x00000000010d9530 in ?? ()
#1  0x00000000010a5492 in ?? ()
#2  0x0000000000000038 in ?? ()
#3  0x000000000000000a in ?? ()
#4  0x00000000400ffe90 in _stack_area_start () from ld.lib.so
#5  0x00000000400ffec0 in _stack_area_start () from ld.lib.so
#6  0x00000000064d0040 in ?? ()
#7  0x0000000001062ba9 in ?? ()
#8  0x00000000010efd98 in ?? ()
#9  0x0000000c0000000b in ?? ()
#10 0x0000000000000000 in ?? ()
(gdb) 
```

[1] https://github.com/throwException/genode/commits/tcp_stress_test

ssumpf, 2019-12-04:
@sidhussmann: I could reproduce your issue. The problem arises when nfds > FD_SETSIZE for `select`, which POSIX states is undefined behaviour. We need to change the `Fd_allocator` in libc to fix this issue, which will limit the number of open libc files to FD_SETSIZE (currently 1024).

sidhussmann, 2019-12-04:
@ssumpf thank you for investigating this. This sounds reasonable.

ssumpf, 2019-12-04:
@sidhussmann: a3ba453 should fix the connection issue.

ssumpf, 2019-12-04:
@chelmuth: Please review a3ba453 + a66ad2e and consider for staging. Thx.

sidhussmann, 2019-12-04:
@ssumpf thank you for the quick fix. I just pulled your branch and tested your fixes. The scenario with `lxip` works like a charm. However, `lwip` segfaults after a few hundred connections with:

```
no RM attachment (READ pf_addr=0x18 pf_ip=0xad2a114 from pager_object: pd='init -> init -> tcp_stress_server' thread='runner')
``` 
Occasionally, instead of segfault this error message occures:
```
[init -> init -> tcp_stress_server]  Error: Assertion "tcp_receive: valid queue length" /home/user/genode/contrib/lwip-6e0661b21fde397041389d5d8db906b5a6543700/src/lib/lwip/src/core/tcp_in.c:1122
```

May this be a different issue?

ssumpf, 2019-12-04:
@sidhussmann: The reason of lwip  failing seems to be a race between the EP and the Runner thread of your test applications. In case of lwip, the library registers a periodic timer ( repos/libports/src/lib/lwip/sys_arch.cc:Sys_timer), the handler function directly calls `sys_check_timeouts` of lwip in the EP context, while the runner thread may also access lwip at the same time.
Executing:
`Libc::with_libc([] () {`
 `  RunServer();});` directly in `Server_main` and client does not lead to faults or assertions to be triggered here.

sidhussmann, 2019-12-04:
@ssumpf we implemented it this way because we first noticed the problem while implementing the gRPC server and we then successively removed complexity until we had a tcp server and tcp client. Your suggestion does make sense in this scenario. And it now works perfectly. However, in the gRPC server case it results to a dead-lock described in the [mailing-list](https://www.mail-archive.com/users@lists.genode.org/msg00884.html) . But this is to be discussed in a different issue. Consider this issue resolved from my part.

chelmuth, 2019-12-04:
> The reason of lwip failing seems to be a race between the EP and the Runner thread of your test applications.

That's a severe observation as it is also true for regular Pthreads using the VFS concurrently to signal handling in VFS plugins... quite something to chew on.

chelmuth, 2019-12-13:
@ssumpf I merged the following commits from your topic branch to staging. Please rebase before proceeding with the lxip developments. Did I miss a patch?

2d55ac7174 libc: thread local errno support
8b31c73337 vfs_lxip/lxip: Lxip::lock for concurrent access
10298d4b9e vfs_lwip/lwip: Lwip::lock for concurrent access

chelmuth, 2019-12-13:
Almost forgot: Any opinions about integrating the TCP-stress test into our code base and the nightly autopilot runs?

commit 0e6b5453 (2019-12-04): libc: limit fd id allocations to FD_SETSIZE

fd > FD_SETSIZE cannot use 'select' or 'poll' within our libc.
Therefore, we added a bit allocator in order to allocate fd < FD_SETSIZE
(1024).

fixes #3568
closed 2019-12-19 by chelmuth