genodelabs/genode #3569: depot_autopilot: close connection to load process

rite, 2019-12-03:
As mentioned in the mailing list [1], I observed a weird behavior when using the `depot_autopilot`: `log/serial` fails to reconnect to the target after a reboot: `picocom` is still running and blocks access to the serial port. As this only happens when the run script is executed within a docker container and I don't quite understand why it solves the problem, I hesitated to submit my patch.

Nevertheless, we used this patch for some time now and nothing bad happened ;)

--
[1] https://lists.genode.org/pipermail/users/2019-October/006938.html

chelmuth, 2019-12-04:
I also adapted *tool/run/run* and merged 5682dd50b4ee833769d9e415f84941dd77578cd2. Thereby I got the feeling this *safer kill* may be worth a dedicated run-tool function (e.g., `proc kill_pid {pid}`) and refactorization. What do you think?

rite, 2019-12-04:
> Thereby I got the feeling this _safer kill_ may be worth a dedicated run-tool function (e.g., `proc kill_spawned {spawn_id}`) and refactorization. What do you think?

Indeed. Fixup commit 1ec0ab8 implements `kill_spawned`. I'm not sure what's best practice here, but I didn't include the check for -1 but leave that part to the caller. But of course, this can easily be changed.

commit f8cc4abd (2019-12-04): run: close connection to load process

From the man page of expect:

> Both expect and interact will detect when the current process exits
> and implicitly do a close.  But if you kill the process by, say, "exec
> kill  $pid",  you  will need to explicitly call close.

Fixes #3569
commit 5682dd50 (2019-12-04): run: close connection to spawned processes on kill

From the man page of expect:

> Both expect and interact will detect when the current process exits
> and implicitly do a close.  But if you kill the process by, say, "exec
> kill  $pid",  you  will need to explicitly call close.

Fixes #3569
commit 6145cdcf (2019-12-19): run: close connection to spawned processes on kill

From the man page of expect:

> Both expect and interact will detect when the current process exits
> and implicitly do a close.  But if you kill the process by, say, "exec
> kill  $pid",  you  will need to explicitly call close.

Fixes #3569
closed 2019-12-19 by chelmuth