genodelabs/genode #3570: nic_router* tests: recent failures

m-stein, 2019-12-04:
Several NIC-router related tests frequently fail again on different kernels and architectures. It is not clear so far what caused this regression.

m-stein, 2019-12-04:
I've tested nic_router_stress and nic_router_flood on qemu/x86_32/pistachio.

With nic_router_stress, the test timeout was to restrictive but after fixing that I couldn't observe further problems. I don't know what causes the decreased performance though.

With nic_router_flood, I was not able to reproduce the ongoing
```
[init -> drivers -> nic_drv] Error: slab backend exhausted!
[init -> drivers -> nic_drv] Error: backend allocator exhausted
```
that seems to be the problem. The test finished just fine although incredibly slow. Of course, it causes a lot of traffic on the NIC driver but this hasn't be a problem so far.

skalk, 2020-02-19:
Tonight, there was a page-fault of the nic_router's ep on i.MX53 QSB with the hw kernel:
```no RM attachment (READ pf_addr=0x0 pf_ip=0x100bee0 from pager_object: pd='init -> nic_router' thread='ep')```

m-stein, 2021-01-19:
The description of this issue seems to be stale. I'd vote for closing it.

closed 2021-01-19 by m-stein