genodelabs/genode #3573: rpi3: uart output not working

tomga, 2019-12-09:
I have a problem with making uart output working with rpi3 on real hardware or maybe with starting system - I can't tell right now.

Do you have a working configuration on which Genode is starting on Raspberry Pi 3 in 64 bit mode and get some output via uart? Release notes mention regular testing only with qemu so I can't be sure and I'd like to avoid this stage of kernel initialization if I can.

I can add that I'm sure that cables are ok because I have it working with my changes for rpi3 with 32 bit. For 64 bit version I have only a message from u-boot that says that it starts kernel.

If it was not tested on real hardware I'll probably try to debug bootstrap like I did for 32bit.

tomga, 2019-12-09:
I forgot to mention that this problem is only on `hw` and not on `foc`. With the latter log test is working.

skalk, 2019-12-09:
@tomga the uart is working fine. I've tested it on hardware when first enabling rpi3. In the meantime however we add SMP support for armv8, on hardware for i.MX8 and for rpi3 using qemu only. When I set the NR_OF_CPUS define in `repos/base-hw/lib/mk/spec/arm_v8/*-hw-rpi3.mk` to 1 instead of 4 it is working fine again.
The problem here is that on Qemu all CPUs are running from the very beginning, and we stop non-boot CPUs in the assembler startup code by a `WFE` instruction. When waking up the secondary CPUs, we just send a `SEV` event to wake them up on rpi3/QEMU. But on real hardware, you have to set the initial instruction pointer for the secondary CPUs first.

skalk, 2019-12-09:
@tomga with commit e1c2632 I've tried to add the missing code. Unfortunately, it is still not working starting the remaining 3 cores. Or they stuck somewhere else in the startup code. Anyway, for the time being it may help you as a starting point.

tomga, 2019-12-10:
@skalk thanks for quick response. I could verify your diagnosis is correct and setting NR_OF_CPUS to 1 actually allows get output from uart. Should I close this issue and open another one with not starting secondary cores on rpi3 or maybe rename this one? Probably I will not work on this issue in near future (it stopped being a blocker for me) so unless you are going to fix it shortly it shouldn't stay open with misleading title. What do you think?

skalk, 2019-12-10:
Sure, it is a good idea to conserve it. I'll rename it.

tomga, 2020-01-07:
@skalk I found some information about waking up cores on rpi3 in 64bit mode and (according to https://www.raspberrypi.org/forums/viewtopic.php?t=232895#p1427323) it differs from the way it is in aarch32 mode.
Address to jump has to be written to specific memory locations (0xe0, 0xe8, 0xf0) and later `sev` should be executed. I tried it but for some reason `sev` causes exception and cores do not wake up. I feel that resolving this may require much digging for me (as I have little experience with assembly on arm) and can be obvious for you so maybe you'll be able to check this and make it work. If not I'll get back to it in some future.

tomga, 2020-01-07:
> for some reason sev causes exception

I couldn't stop thinking about it and I think I found what causes it - although I didn't find reference for it. It seems that if `sev` is executed in EL1 and other cores are in EL2 than exception is trapped. This situation happens because primary core's exception level is changed from EL2 to EL1 just before attempt to wake up non-primary cores. Moving call to start other cores before changing exception level seems to help.

Additionally I think that in proposed patch change from `_crt0_enable_fpu` to `_start` as a starting point for non-primary cores is unfortunate because the code commented as hack for qemu is executed and it would require additional `sev` instruction and with proper timing.

With quick fixes for above problems I could start at least one non-primary core and execute c++ code. I'll continue investigation in the next days.

skalk, 2020-01-08:
@tomga great to hear how though you dig into this!

tomga, 2020-01-09:
@skalk, I'd like to ask you for some advice as I don't understand what happens. It seems that I'm able to pass the bootstrap but `kernel_init()` function is executed only by first core that finishes bootstrap (depending on delays I add it can be a primary core or not). Subsequent cores that finish bootstrap somehow end up executing `kernel()` without entering `kernel_init()`.

My wild guess would be the commented out part about stack per cpu in `repos/base-hw/src/core/spec/arm_v8/crt0.s` (similar one is used in `repos/base-hw/src/core/spec/arm_v8/crt0.s`). But then I can't see how this file (from `arm_v8`) is working properly for `imx8q_evk`.

Do you have some advice? I feel I'm close but...

tomga, 2020-01-09:
In 29282563a9ae14dc704870cbbb2ca757f508c10c I pushed relevant changes for starting secondary cores.

I'm sure that with those changes `Bootstrap::Platform::enable_mmu()` in `cortex_a53_mmu.cc` executes for all cores almost to the end of this function. "Almost" because I couldn't get any log on uart after the line which enables mmu and caches (`Cpu::Sctlr_el1::write(sctlr);`).

Like I wrote yesterday first executing core enters the `kernel_init()` from (`base-hw/src/core/kernel/init.cc`) but others do not and they fail later with assertion:
```
Error: Assertion failed: id < _count && _cpus[id].constructed()
Error:   File: /projects/genode/genode/repos/base/../base-hw/src/core/kernel/cpu.cc:186
Error:   Function: Kernel::Cpu& Kernel::Cpu_pool::cpu(unsigned int)
```
I'm out of ideas now.

skalk, 2020-01-10:
>  My wild guess would be the commented out part about stack per cpu in repos/base-hw/src/core/spec/arm_v8/crt0.s (similar one is used in repos/base-hw/src/core/spec/arm_v8/crt0.s). But then I can't see how this file (from arm_v8) is working properly for imx8q_evk.

That commented code is a misleading artefact, and was accidentally commited. It should not be the problem, because the cpus run on a dedicated stack initially assigned in bootstrap's crt0 code. The correct final stack is used whenever the kernel is entered the first time via its exception vector. For initialization purposes it still uses the bootstrap stack.

> Error: Assertion failed: id < _count && _cpus[id].constructed()
Error:   File: /projects/genode/genode/repos/base/../base-hw/src/core/kernel/cpu.cc:186
Error:   Function: Kernel::Cpu& Kernel::Cpu_pool::cpu(unsigned int)

Ok, either the wrong cpu count got propagated from bootstrap to core (_count variable), or even more probable: the cpu id returned by 'Cpu::executing_id()' is duplicating or outside the range of 0-_count. At best you can check the value of 'Cpu::Mpidr::read()' inside bootstrap (as long as log still works) for each cpu. We simply return 'Cpu::Mpidr::read() && 0xff' inside 'Cpu::executing_id() '.

tomga, 2020-01-10:
> Ok, either the wrong cpu count got propagated from bootstrap to core (_count variable), or even more probable: the cpu id returned by 'Cpu::executing_id()' is duplicating or outside the range of 0-_count. 

I'm almost sure I've verified that it's not true. I checked those values. It is the second part of the assertion is not true. And it is because `kernel_init` is not executed. It is executed only for one core and as I wrote earlier it doesn't have to be a primary core. I don't know why. After setting `Sctlr_el1` in `enable_mmu` either uart stops working or execution is transfered somewhere else and I don't know where so I'm not able to trace what happens.

skalk, 2020-01-10:
@tomga the assertion which fails is in the code path of kernel_init only. It would be helpful if you can print out the values of MPIDR from the different CPUs within bootstrap before turing of the MMU, just for us to know what is different than expected.

tomga, 2020-01-10:
I feel you don't believe me :-) Maybe you are right in it but I can't see it yet.

> the assertion which fails is in the code path of kernel_init only.

Maybe you are right, but I have the impression that it is not that obvious. It is called from `executing_cpu()` and it is called from `thread.cc` and `irq.cc`. Are they really called only from `kernel_init`?

I pushed my current working branch to https://github.com/tomga/genode/commits/rpi_all_19.11

On top of it are changes with experiments to run those cores and the last one enables delay loops (between starts secondary cores and a longer one for secondary cores to wait until primary core quits `enable_mmu`) and output you requested and it gives (comments inside):
```
Starting kernel ...

cpu: 0
S1
S2
S3
cortex_a53_mmu.cc:2022147483648        // line 202 and Mpidr value put here by mistake
            Mpidr:2147483648
     Mpidr & 0xff:0
ki1: 0                    // kernel_init, primary core
cpu: 1
cortex_a53_mmu.cc:2022147483649
            Mpidr:2147483649
     Mpidr & 0xff:1
id:1,cnt:4,c:0.1     // values before assertion c:0 means that constructed returned false
Error: Assertion failed: id < _count && _cpus[id].constructed()
Error:   File: /projects/genode/genode/repos/base/../base-hw/src/core/kernel/cpu.cc:186
Error:   Function: Kernel::Cpu& Kernel::Cpu_pool::cpu(unsigned int)
cpu: 2
cortex_a53_mmu.cc:2022147483650
            Mpidr:2147483650
     Mpidr & 0xff:2
cpu: 3
cortex_a53_mmu.cc:2022147483651
            Mpidr:2147483651
     Mpidr & 0xff:3
```

I'm not sure what you expected to find in those Mpidr values but I suspect they are different from what you expected.

I checked state of cores with JTAG debugger. Enabling of caches and mmu seem to succeed. But why after that logs stop being written? Maybe some reset/clear is required? There was something like this in `cortex_a15_cpu.cc`
```
	/* invalidate branch predictor */
	Bpiall::write(0);
```

Log from openocd after halting cores. Let me know if I can provide more info.

```
Open On-Chip Debugger
> targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0  rpi3.a53.0         aarch64    little rpi3.tap           running
 1  rpi3.a53.1         aarch64    little rpi3.tap           running
 2  rpi3.a53.2         aarch64    little rpi3.tap           running
 3* rpi3.a53.3         aarch64    little rpi3.tap           running

> halt
rpi3.a53.3 cluster 0 core 3 multi core
target halted in AArch64 state due to debug-request, current mode: EL1H
cpsr: 0x200003c5 pc: 0xffffffc000000204
MMU: enabled, D-Cache: enabled, I-Cache: enabled
> targets rpi3.a53.2
> halt
rpi3.a53.2 cluster 0 core 2 multi core
target halted in AArch64 state due to debug-request, current mode: EL1H
cpsr: 0x800003c5 pc: 0xffffffc000000204
MMU: enabled, D-Cache: enabled, I-Cache: enabled
> targets rpi3.a53.1
> halt
rpi3.a53.1 cluster 0 core 1 multi core
target halted in AArch64 state due to debug-request, current mode: EL1H
cpsr: 0x600003c5 pc: 0xffffffc0000579bc
MMU: enabled, D-Cache: enabled, I-Cache: enabled
> targets rpi3.a53.0
> halt
rpi3.a53.0 cluster 0 core 0 multi core
target halted in AArch64 state due to debug-request, current mode: EL1H
cpsr: 0x400003c5 pc: 0xffffffc000058a90
MMU: enabled, D-Cache: enabled, I-Cache: enabled
```

tomga, 2020-01-11:
@skalk I think that something is wrong with the stack after enabling mmu. In yesterdays log from openocd it could be seen that all caches were enabled so - given that I couldn't trust logging on uart - I started to add code to disable instruction cache in some places. I've verified that disabling instruction cache right after enabling it in `enable_mmu` works but if I moved this code to `base-hw/src/bootstrap/init.cc`:
```
extern "C" void init()
{
	Bootstrap::Platform & p = Bootstrap::platform();
	auto cpu_id(p.enable_mmu());

	using Cpu = Hw::Arm_64_cpu;
	Cpu::Sctlr::access_t sctlr = Cpu::Sctlr_el1::read();
	Cpu::Sctlr::I::set(sctlr, 0);
	Cpu::Sctlr_el1::write(sctlr);

	p.start_core(cpu_id);
}
```
it works only for first core. So returning from `enable_mmu` doesn't work as expected. I have no idea why.

Additonally I found that `enable_mmu` in `cortex_a53_mmu.cc` always returns 0. It is not related to problem here but unless it is passed through some functions and later ignored it would suggest that this code couldn't work correctly in multiprocessor environment. Can you say on which configurations `smp` test is working and is tested regularly? Should I upgrade to current master with my work?

tomga, 2020-01-12:
SUCCESS!!

I wrote earlier:

> I think that something is wrong with the stack after enabling mmu

so I checked SP after enabling MMU in a busy loop and its value was exactly the same on all cores.

After that it was quite easy to find that my earlier change that moved starting point for secondary cores from `_start` to `_crt0_enable_fpu` in bootstrap's `crt0.s` (which skipped special code for qemu)   skipped writing cpu id into register `x0` which was later used for setting stack properly for each core.

Now I have `log` and `smp` tests working on `rpi3`. I'll prepare clean patch soon.

tomga, 2020-01-12:
@skalk changes in the above commit are minimal changes in my opinion required for starting secondary cores on rpi3. Comment in the commit contains description of changes. I tested it on real hardware and now `log` and `smp` tests pass. I think I didn't break running on `qemu` but I did not test it.

One controversy is the return value from `enable_mmu` for Cortex A53 which always was zero. This value is passed through some functions but probably it is not used in the end. I fixed returned value anyway.

And in the end I'm sorry for the noise I made in this issue over the weekend. Until I found the problem with stack pointer I really couldn't understand behavior I was observing and hoped for some ideas from someone.

skalk, 2020-01-13:
@tomga I'm quite happy that you succeeded to run multiple cores on rpi3 hardware, congratulations! I'll investigate your patch quite soon, and incorporate it to staging. Regarding the return value for enable_mmu: you are right, although it is not evaluated in the ARMv8 core code right now it should return the correct value.

>  And in the end I'm sorry for the noise I made in this issue over the weekend. Until I found the problem with stack pointer I really couldn't understand behavior I was observing and hoped for some ideas from someone.

You do not have to apologize at all! You are always welcome to discuss progress or road-blocks in the issue tracker. I'm honestly impressed by your patience when dealing with those low-level issues, and your steady progress. Thank you!

skalk, 2020-02-06:
@tomga I've rebased your commit to current staging, and did minor cosmetics (macro, comments). From my side it is ready for staging.

nfeske, 2020-02-10:
Merged to staging

skalk, 2020-02-21:
The commit solution is already on master.

closed 2020-02-21 by skalk