genodelabs/genode #3575: depot_deploy: support heartbeat settings in deploy config

sidhussmann, 2019-12-09:
To guarantee availability of a system even with complex sub-systems, I want to be able to explicitly tell which packages should be watched for missing heartbeats. [Genode 18.11](https://genode.org/documentation/release-notes/18.11#Component_health_monitoring) introduced the heartbeat feature to init configurations. This patch adds the functionality to deploy configurations in the same way:

```
<config arch="} [depot_spec] {" verbose="no">

	<static>
		<parent-provides>
			<service name="ROM"/>
			<service name="CPU"/>
			<service name="PD"/>
			<service name="LOG"/>
			<service name="Timer"/>
			<service name="File_system"/>
			<service name="Nic"/>
		</parent-provides>
	</static>

	<common_routes>
		<service name="ROM" label_last="ld.lib.so"> <parent/> </service>
		<service name="ROM" label_last="init">      <parent/> </service>
		<service name="CPU">   <parent/> </service>
		<service name="PD">    <parent/> </service>
		<service name="LOG">   <parent/> </service>
		<service name="Timer"> <parent/> </service>
	</common_routes>

	<report delay_ms="1000"/>
	<heartbeat rate_ms="2000"/>

	<start name="really-complex-app" pkg="} [depot_user]/pkg/[really-complex-app] {">
		<heartbeat/>
		<route>
			<service name="Nic"> <parent/> </service>
		</route>
	</start>
</config>
```

sidhussmann, 2020-01-21:
Is this something you guys would consider or would you choose a different solution?

nfeske, 2020-01-21:
Sorry for the lack of response. I just haven't taken the time to form an opinion. The feature is certainly useful and the patch looks fine.

nfeske, 2020-01-21:
After having a closer look at the patch, I'd like to change one thing:

For memory safety, we try to avoid the returning of references or pointers. An `Xml_node` is actually a glorified pointer. Hence, use of `config.sub_node()` is dangerous. Within the API, we gradually move to a coding style that is less risky.

Instead of
```
if (config.has_sub_node("report")) {
  const Xml_node& report_config = config.sub_node("report");
  xml.append(report_config.addr(), report_config.size());
}
```

is is better to use

```
config.with_sub_node("report", [&] (Xml_node const &report) {
  xml.append(....);
});
```

The pointer is strictly captured within the scope of the lambda, which eliminates a while class of potential dangling-pointer issues.

In the same spirit, the use of `Xml_node::addr` is discouraged. It will eventually be removed. You may use `Xml_node::with_raw_value` instead, which scopes the pointer.

Regarding the `xml.append`, I'd prefer to make the XML generation more explicit by generating concrete nodes and attributes instead of copying bytes verbatim from the `report`. This way, the code reveals the complete form of XML that can come out of it.

sidhussmann, 2020-01-22:
@nfeske there are two important things you just highlighted.

1. Memory safety
2. Regarding `xml.append`

I agree with both arguments. Argument number two lead internally to discussions. Let me try to sum it up. I understand your point to make xml generation explicit. In this specific case we convert a `deploy` configuration to an `init` configuration. Since these two interfaces are very similar, one could argue that it would be cumbersome to iterate through all attributes from let's say the `heartbeat` node and add them explicitly when the expected ones occur. Also, what if the interface of `init` changes and an attribute is added to the `heartbeat` node? You would have to change the implementation of `depot_deploy` as well. Arguably, for a feature `depot_deploy` is not really concerned about.

On the other hand, one could argue that the usage of the `deploy` configuration should abstract complexity away from the `init` configuration.

Furthermore, the `report` node is ambiguous. Does it mean to tell `depot_deploy` to report its state or is it `init` that should report its state, whose configuration `depot_deploy` provides?

How about following solution: We remove the ambiguous `report` node from the `deploy` configuration and just use `heartbeat`, since heartbeat does not make sense without being able to report it. If the `heartbeat` node is set, `depot_deploy` then writes the `init` configuration with `heartbeat` node and its `rate_ms` attribute, and sets the `report` node with a reasonable `delay_ms` attribute. E.g. `delay_ms=heartbeat_rate_ms/2`.

What do you think?

nfeske, 2020-01-24:
I very much appreciate your response.

What you said about the pros and cons of coping XML content as-is versus making the generation explicit is spot on. It's not a clear-cut black-and-white decision but must be considered case by case. I generally lean towards the explicit approach unless it would result in a lot of boilerplate code. In the specific case, I see the deploy config as something different from the init config rather than some kind of wrapper. Given that the <heartbeat> config is so simple, I find it worth to be explicit here. Boilerplate complexity is apparently not a problem in this case.

Regarding the <report> node. I think we could leave the rate of the reports up to the user instead of hard-wiring the heuristics of `delay_ms=heartbeat_rate_ms/2`. Both <report delay_ms="..."/> and <heartbeat rate_ms="..."/> could be present in the deploy config.

As a minor nitpick - I hope that I'm not too pedantic - let me suggest to change `const size_t` to `size_t const`. Throughout Genode, we always put `const` keywords to the right of the type to be marked as const. This is the general rule of how `const` works in C++. In contrast, the use of `const` at the beginning is a special case and breaks the consistency of the general pattern.

chelmuth, 2020-02-06:
The branch was updated by @sidhussmann (Would you mind to squash your commits?). Is it time for the fixed label already?

sidhussmann, 2020-02-06:
@chelmuth I just force-pushed the squashed commit in 495b031. I'd consider this feature as fixed.

sidhussmann, 2020-02-10:
@nfeske , I neglected to reply to your feedback. It is in my interest to contribute the cleanest, most maintainable code possible. So no, your comment was not at all received as pedantic.

nfeske, 2020-02-10:
Thanks for clarifying, Sid. I'm relieved. :-) Your patch just entered staging now.

nfeske, 2020-02-28:
Merged to master.

closed 2020-02-28 by nfeske