genodelabs/genode #3578: Genodians.org without Noux

nfeske, 2019-12-12:
With the new libc support for execve and fork, we can in principle run scenarios like the static CMS of genodians.org directly on Genode without the need of the Noux runtime. However, this use case stresses the libc mechanisms in new ways compared to the rather basic bash.run scenario.

- Rather deep fork hierarchies
- GNU make
- Tcl
- Intensive use of pipes

nfeske, 2019-12-12:
@chelmuth I'd appreciate a review of the commit series. I merged it to staging to see what the autopilot has to tell.

chelmuth, 2019-12-12:
No objections from my side.

nfeske, 2019-12-12:
Thanks for the quick feedback!

sidhussmann, 2019-12-12:
@nfeske I'm interested in how you use the vfs/pipe plugin but I can't find the run script. Is it available somewhere else?

nfeske, 2019-12-12:
Right now, the best examples are the bash.run scenario (https://github.com/genodelabs/genode/blob/master/repos/ports/run/bash.run#L87) and the libc_pipe test (https://github.com/genodelabs/genode/tree/master/repos/libports/recipes/pkg/test-libc_pipe).

The genodians.org scenario is located at a separate repository. You can find the relevant commits on my topic branch https://github.com/nfeske/genodians.org/commits/remove_noux

sidhussmann, 2019-12-12:
@nfeske perfect! Thank you!

nfeske, 2020-02-10:
Done.

closed 2020-02-10 by nfeske