genodelabs/genode #3583: vfs: new plugin for named pipes

sidhussmann, 2019-12-15:
The VFS named_pipe plugin exposes a control file for creating named pipes. The
goal is to be able to transfer data via file handles from one component to
another. E.g. you can attach stdin of a libc component to a named pipe and write
data using standard fwrite() calls.

Here a sample scenario where stdin of a component is attached to a named pipe:

```
<start name="vfs">
	<provides> <service name="File_system"/> </provides>
	<resource name="RAM" quantum="4M"/>
	<config>
		<vfs>
			<dir name="upstream">
				<named_pipe/>
			</dir>
		</vfs>
		<default-policy root="/" writeable="yes"/>
	</config>
</start>

<start name="my-libc-app">
	<resource name="RAM" quantum="4M"/>
	<config>
		<vfs>
			<dir name="dev">
				<dir name="pipe"> <fs/> </dir>
				<log/>
			</dir>
		</vfs>
		<libc stdin="/dev/pipe/upstream/out" stdout="/dev/log" stderr="/dev/log"/>
	</config>
</start>
```

The directory named `upstream` represents a pipe and contains an `in` and `out`
file for writing and reading respectively to the pipe.

A remote component can now write to `/dev/pipe/upstream/in` so the data written
is then accessible on stdin within "my-libc-app". Once the remote component
closes the `/dev/pipe/upstream/in` file, an EOF is received on stdin of
"my-libc-app".

Reads and writes are non-blocking and will complete short operations without
error, with the exception of reads on an empty pipe, which return READ_QUEUED.
The read and write capacity of a pipe may be queried by stat'ing the size of
`out` and `in` files.

sidhussmann, 2019-12-18:
4198c8d adds `watch` functionality to the `out` file handle so that the receiving client gets notified when new data is available.

chelmuth, 2019-12-19:
I'm wondering if the features added by this additional VFS plugin could easily be added to *vfs_pipe* by an additional configuration mechanism like the following. The advantages would be less code duplication, less test efforts and less divergence between this and the original pipe.

```
<pipe> <fifo name="upstream"/> <pipe>
```

Also, I did not find the implementation of the following sentence from the *README*, or did I get it wrong and it is actually available?

> The VFS named_pipe plugin exposes a control file for creating named pipes.

sidhussmann, 2019-12-23:
@chelmuth I like your idea. However, I'm not so sure about the "easily"... It does take some redesign which will take some time. I'm looking into it.

sidhussmann, 2020-01-10:
@chelmuth I added the fifo feature to _vfs_pipe_ and removed the _vfs_named_pipe_ like you suggested. It took some refactoring which I handled in separate commits. Currently the feature can be used with following `vfs` configuration: 

```
<dir name="upstream">
  <pipe><fifo/></pipe>
</dir>
```

I'm having trouble figuring out how to achieve the same functionality with the `name` attribute, and not with embedding pipe within a `dir` node. Could you help me out with this last step?

chelmuth, 2020-01-13:
I think you got me wrong. What I meant was a pipe that works as currently documented with a control file named *new* and dynamically created subdirectories for pipes with *in* and *out* files. FIFOs are just pre-created subdirectory with *in* and *out* files named after the `name` attribute of the declared `<fifo>` node. Your current implementation completely changes the plugin behavior with the `<fifo>` subnode. I'd like to create a bunch of FIFOs on VFS startup if needed like follows.

```xml
<dir name="pipes">
  <pipe>
    <fifo name="fifo0"/>
    <fifo name="fifo1"/>
    <fifo name="fifo2"/>
  </pipe>
</dir>
```

So, the `Vfs_pipe::Fifo_file_system` from your implementation implements one subdirectory in the `Vfs_pipe::File_system`.

sidhussmann, 2020-01-16:
@chelmuth I didn't consider my last commit as final solution. FYI I'm still working on it when I find the time.

sidhussmann, 2020-01-19:
@chelmuth I finished implementing the features according to your feedback. Also, since we are using this feature extensively to stream large amounts of data to libc applications, I added a stress test that validates piping 4M of random data in 7119f83af4780952ab5296ca87a9d5cf3aba9c99.

sidhussmann, 2020-10-21:
These four commits above re-implement the fifo feature for the vfs/pipe plugin with minimal differences to the original pipe plugin implementation.

sidhussmann, 2021-01-04:
@nfeske as discussed on the mailing list, IMHO this feature is ready for review. The following commits show my approach:

6a90a979b2a05565fdbb664f09434e9795bf9792
63160d4ce459233ef86d7fc7c8e6c613d4bd8d21
a9c286442f031870a41b0799ee665cb4d54438f8
7c762c88017388ce8b35f21787f7f8e5ad5f8e34
2175c4b18e39b81143a0302e3e00415af2115826
6ab540e786874979c2fc57e1835a1489c855c403
8f266f584142498884888230efe2f60ef77c9ea4
cf15eb01a464fba61c23e301699f16cb70cb92e1

I haven't squashed these commits yet because I figured it might be easier to review. However, I'm not so sure anymore. Should I squash these to one commit?

nfeske, 2021-01-08:
Thanks @sidhussmann for the convenient list of commits. I gave them a rough review. Please see my comments below. For further process, I agree that squashing the steps into one commit would make life easier. So once the points below are addressed, I'd like to review the whole change as one patch again, and possibly provide further feedback.

My remarks:

There a a few minor coding-style issues (see https://genode.org/documentation/developer-resources/coding_style for reference).

* Indent initializers of `Fifo_file_system` constructor, `Vfs_pipe::File_system` constructor

* Rename `_get_pipe_id` to `_pipe_id` (our coding style omits set_ and get_ from accessors)

* Please omit braces in simple cases like

  ```
  if (notify) {
    submit_signal();
  }
  ```
  This should be
  ```
  if (notify)
    submit_signal();
  ```

The code shows recurring patterns of:

```
try {
  _pipe_space.apply<Pipe&>(...{
    ...
  });
catch (Pipe_space::Unknown_id) { }
```

To reduce the depth of nesting levels, please consider introducing a
a private method that wraps `Id_space::apply`, like sketched as follows:

```
template <typename FN>
void _try_apply(Pipe_space::Id id, FN const &fn)
{
  try { _pipe_space.apply<Pipe &>(id, fn); }
  catch (Pipe_space::Unknown_id) { }
}
```

This will not only simplify the code, but also presumably reduce the patch size.

Please consider using a `Registry<Fifo_element>` instead of a `List<Fifo_element>` because the `Registry` is much safer. E.g., one cannot forget to remove an element.

I very much appreciate the update of the README. In the README, please quote verbatim words with "'" instead of "`" because Genode's README and documentation files are based on GOSH syntax. (see https://github.com/nfeske/gosh/blob/master/README) Code blocks are marked with a "!" at the start of each line. You can find plenty of examples in the source tree.

It is very good that you added a test scenario, exercising the feature. By looking at the scenario, I think it would be a good candidate for the depot autopilot because it is fairly freestanding. You may generate the of test data in the recipe (content.mk) of the test case, e.g., by using something like `seq 5000`. Depot-autopilot tests are preferable over run scripts because they are all executed within one boot cycle, and - since depot-autopilot has the role of Genode's basis test bench - the included tests get the highest attention from Genode's core team.

I think that explicit naming of `/in` and `/out` files is preferable over a builtin policy of selecting one of the two files based on attributes. It is more transparent to the user and in line with the functioning of the existing pipe mechanism.

sidhussmann, 2021-01-12:
Thank you for your feedback @nfeske and my apologies for the coding-style issues that slipped by. Also, I'm really glad that you pointed out that the `Registry` feature is way more suitable and safer instead of using a `List` of references.

I have applied your feedback and added autopilot tests in a7e005fe30d6ef59f06301d5c2d7331176504ad4. I have not squashed all commits just yet. There are currently 3 commits in branch https://github.com/sidhussmann/genode/tree/3583-vfs-pipe-fifo-feature , because I would like to discuss the following subject a bit more:

> I think that explicit naming of `/in` and `/out` files is preferable over a builtin policy of selecting one of the two files based on attributes. It is more transparent to the user and in line with the functioning of the existing pipe mechanism.

At first, I was very much in favor of this approach as well. However, I believe there are more arguments for the approach how it is implemented in d8a94ce0c69d7b9592bda7c1633914f1095a3e73 : 
1. IMO the argument regarding of the transparency to the user does not quite hold. For the existing pipe mechanism, the user (in this case the libc user) does not have any notions of `/new`, `/in` or `/out`. They just use the `pipe` syscall and handle the endings of the pipe via file descriptors. 
2. The current implementation resembles more how the libc user is used to the named pipe feature of linux. On linux you open the fifo file, and you get the right end of the pipe by the `O_WRONLY` or `O_RDONLY` flag provided with the `open` syscall.
3. IMO the old implementation (with `/in` and `/out`) is more complicated to the user and more error prone. For instance, when you have a scenario where you want to talk to a libc component via `stdin` and `stdout` we'd have the following situation:

```
<start name="vfs">
	<provides>
		<service name="File_system"/>
	</provides>
	<resource name="RAM" quantum="4M"/>
	<config>
		<vfs>
			<pipe>
				<fifo name="upstream"/>
				<fifo name="downstream"/>
			</pipe>
		</vfs>
		<default-policy root="/" writeable="yes"/>
	</config>
</start>

<start name="echo">
	<binary name="test-libc_fifo_pipe"/>
	<resource name="RAM" quantum="2M"/>
	<config type="echo">
		<vfs>
			<dir name="dev">
				<dir name="pipe-in"> <fs/> </dir>
				<dir name="pipe-out"> <fs/> </dir>
				<log/>
			</dir>
		</vfs>
		<libc stdin="/dev/pipe-out/upstream/out" stdout="/dev/pipe-in/downstream/in" stderr="/dev/log"/>
	</config>
	<route>
		<service name="File_system"> <child name="vfs"/> </service>
		<any-service> <parent/> <any-child/> </any-service>
	</route>
</start>

<start name="fifo-pipe-test">
	<binary name="test-libc_fifo_pipe"/>
	<resource name="RAM" quantum="64M"/>
	<config iterations="10">
		<vfs>
			<dir name="ro">
				<rom name="test-data.bin"/>
			</dir>
			<dir name="dev">
				<dir name="pipe-in"> <fs/> </dir>
				<dir name="pipe-out"> <fs/> </dir>
				<log/>
			</dir>
		</vfs>
		<libc stdout="/dev/log"/>
	</config>
	<route>
		<service name="ROM"    label="init_template">  <child name="fs_rom"     label="init_template"/> </service>
		<service name="ROM"    label="empty_template"> <child name="fs_rom"     label="empty_template"/> </service>
		<service name="Report" label="init.config">    <child name="report_rom" label="init.config"/> </service>
		<service name="File_system">                   <child name="vfs"/> </service>
		<service name="Timer">                         <child name="timer"/> </service>
		<any-service> <parent/> <any-child/> </any-service>
	</route>
</start>
```

In the mailing list [1]  you kindly pointed out that to avoid a potential deadlock we would need to have a separate `file_system` session for `stdin` and `stdout` respectively. This forced us to use names such as `pipe-in`, `pipe-out`, `upstream`, and `downstream` - combined with `/in` and `/out`, which ended up in having file paths like `/dev/pipe-out/downstream/out`. This is some serious plumbing :-). Additionally, not to fall into the same trap of the dead-lock, the user needs profound know-how of core Genode internals. With the current approach, it comes more natural and you are even forced to use two `file_system` sessions:

```
<start name="echo">
	<binary name="test-libc_fifo_pipe"/>
	<resource name="RAM" quantum="2M"/>
	<config type="echo">
		<vfs>
			<dir name="dev">
				<dir name="pipe-in"> <fs/> </dir>
				<dir name="pipe-out"> <fs/> </dir>
				<log/>
			</dir>
		</vfs>
		<libc stdin="/dev/pipe-out/upstream" stdout="/dev/pipe-in/downstream" stderr="/dev/log"/>
	</config>
	<route>
		<any-service> <parent/> <any-child/> </any-service>
	</route>
</start>
```

I'm curious about further feedback from you.

Also, should I remove the `vfs_fifo_pipe.run` script, since we now have a depot-autopilot test?

[1] https://lists.genode.org/pipermail/users/2020-December/007389.html

chelmuth, 2021-01-13:
I got the feeling your point 3 stems from the fact that the VFS server configuration lacks label-based policies. Wouldn't it be practicable to map the intention of upstream/downstream to labels and then set the `root` attribute properly? For example

```
<policy label_suffix="upstream"   root="/upstream" writeable="no"/>
<policy label_suffix="downstream" root="/downstream" writeable="yes"/>
...
<dir name="pipe-in"> <fs label="upstream"/> </dir>
<dir name="pipe-out"> <fs label="downstream"/> </dir>
...
<libc stdin="/dev/pipe-out/out" stdout="/dev/pipe-in/in" stderr="/dev/log"/>
```

So, the session can also be limited to read-only in case of standard input.

sidhussmann, 2021-01-13:
Read/Write policies on vfs server side is definitely something we should consider.
Just to make sure I get your point. What you described was only one side of the two pipes (the interaction between `vfs` and `echo`).

[image]

So when we introduce policies with the `/in /out` approach it would look something like this:

```
<start name="vfs">
...
<policy label_suffix="fifo-pipe-test -> upstream"   root="/upstream" writeable="yes"/>
<policy label_suffix="echo -> upstream"   root="/upstream" writeable="no"/>
<policy label_suffix="echo -> downstream" root="/downstream" writeable="yes"/>
<policy label_suffix="fifo-pipe-test -> downstream" root="/downstream" writeable="no"/>
...
</start>

<start name="echo">
...
<dir name="pipe-in"> <fs label="upstream"/> </dir>
<dir name="pipe-out"> <fs label="downstream"/> </dir>
...
<libc stdin="/dev/pipe-out/out" stdout="/dev/pipe-in/in" stderr="/dev/log"/>
...
</start>

<start name="fifo-pipe-test">
...
<dir name="pipe-in"> <fs label="downstream"/> </dir>
<dir name="pipe-out"> <fs label="upstream"/> </dir>
...
</start>
```

And in `fifo-pipe-test` the two file paths would be `SEND_FILE=/dev/pipe-in/in` and `RECEIVE_FILE=/dev/pipe-out/out`. 

@chelmuth, is this what you meant? If yes, introducing policies is IMO something that is orthogonal to whether we address the ends of the fifo pipe via `/in /out` or via the `O_WRONLY/O_RDONLY` flag. 

What I also now notice is that within `fifo-pipe-test` the actual file name e.g. `upstream` is abstracted away. Which to me makes it even more confusing. But maybe we chose the wrong names for the dirs, fifo files, and pipe-endings?

@nfeske, I do see your point. And I don't want to come off as too stubborn. Critically reflecting on my point 2, it seems this boils down to whether which perspective should be more concise: The applications perspective vs the perspective on the configuration (session routing and policies). My goal is to design an example scenario that makes the fifo-pipe feature easy to use and understand. We are not there yet.

nfeske, 2021-01-13:
@sidhussmann Thank you for the great care you are putting in the issue and for your response to the points I raised during my initial review. The line of thoughts behind the user-interface choices is important to know. Thanks for sharing it. Admittedly, I have not analyzed the pro and cons very deeply yet.

> And I don't want to come off as too stubborn.

No worries. I can assure you, that's not the case at all. :-)

A few thoughts crossed my mind.

* Regardless of the way to pick, we should strive finding a scheme that is then consistently used for both named and unnamed pipes.

* Maybe we can consider supporting both a precise specification of an /in or /out path, and the opening of a convenient pseudo-device file that has some convenient semantics (like distinguishing the pipe ends based on open attributes)?
  
  The terminal plugin works this way. The file /dev/terminal works like a character device. You can open it, read it, and write it. Additionally, under /dev/.terminal/, the plugin hosts a bunch of pseudo files that give access to the under-the-hood plumbing side of the plugin. The libc implicitly accesses those files for providing ioctl and signal semantics that are missing from the /dev/terminal file. So if we put /in and /out files in a corresponding dot-directory for the pipe plugin, we would not bother people using the convenience frontend with too many details, while still allowing for the explicit use of /in and /out paths for people that need it (e.g., for rigid read-write policy scenarios where one wants to hand out only one end of the pipe).

* I am definitely open for changes of the plugin's file structure. So if you have ideas for improvements, I don't want to hold you back. For these considerations, note that placing /in and /out in distinct directories has the advantage that the chroot component (or the root attributes of VFS policy nodes) can readily be used to isolate the access to either end of the pipe. I'm not saying that I have an immediate use case but I think that this natural fit is quite beautiful.

nfeske, 2021-01-14:
> Also, should I remove the vfs_fifo_pipe.run script, since we now have a depot-autopilot test?

Yes, please drop it. For executing the new pkg/test-libc_fifo_pipe as a stand-alone test, it would be worth trying the os/run/test.run script.

sidhussmann, 2021-01-15:
@chelmuth the commit above introduced explicit policies, which made it possible to remove all `pipe-in` and `pipe-out` directories. I believe it is now much simpler - and safer.

> For these considerations, note that placing /in and /out in distinct directories has the advantage that the chroot component (or the root attributes of VFS policy nodes) can readily be used to isolate the access to either end of the pipe. 

@nfeske from a security point of view, what is the advantage of using `chroot` instead of using explicit policies on `vfs` server side? By explicit policies I mean something like this:

```
<vfs>
	<pipe>
		<fifo name="upstream"/>
		<fifo name="downstream"/>
	</pipe>
</vfs>
<policy label="init -> echo -> upstream" root="/" writeable="no" />
<policy label="init -> echo -> downstream" root="/" writeable="yes" />
<policy label="fifo-pipe-test -> upstream" root="/" writeable="yes" />
<policy label="fifo-pipe-test -> downstream" root="/" writeable="no" />
```

> For executing the new pkg/test-libc_fifo_pipe as a stand-alone test, it would be worth trying the os/run/test.run script.

I was not aware of this great feature. Since `test.run` does not evaluate the `content.mk` file of the raw recipe, I was forced to add `raw/test-libc_fifo_pipe/test-data.bin` rather than executing `seq 5000 > $@` on the fly. Is this the intended way to provide test data for test package archives?

> Regardless of the way to pick, we should strive finding a scheme that is then consistently used for both named and unnamed pipes.

I agree. It seems the solution space is quite big. So, I believe we should focus on the user-interface first. I would love to discuss this in front of a whiteboard with you two :-). How should we proceed?

nfeske, 2021-01-19:
> @nfeske from a security point of view, what is the advantage of using chroot instead of using explicit policies on vfs server side?

Let me try to rephrase what I wanted to say: By placing each end of the pipe in a distinct directory, access control to each end of the pipe can be enforced via access control of the respective directory (access control by controlling visibility). Both the VFS root attribute and the chroot component work at the granularity of directories. This speaks in favor of a nested directory structure.

> I was not aware of this great feature. Since test.run does not evaluate the content.mk file of the raw recipe, I was forced to add raw/test-libc_fifo_pipe/test-data.bin rather than executing seq 5000 > $@ on the fly. Is this the intended way to provide test data for test package archives?

That's a good point. However, checking-in such a chunk of test data is not nice. Since the pattern of the data is not important, couldn't the test use some arbitrary other piece of data that is already present in the scenario. Let say, the init binary?

sidhussmann, 2021-01-24:
@nfeske I didn't see the security implications with my previous approach at first. It seemed to me that the vfs policy `<policy label="fifo-pipe-test -> downstream" root="/" writeable="no"/>` should be enough. However, the problem is that an untrusted component that has e.g. read access to one end of the pipe, automatically also gets write access to the other end. 

> So if we put /in and /out files in a corresponding dot-directory for the pipe plugin, we would not bother people using the convenience front end with too many details, while still allowing for the [...]

The last commit adds this missing feature while preserving the opening of a convenient pseudo-device file with the convenient open attributes. `test-libc_fifo_pipe` now also tests this access control mechanism. 

> Since the pattern of the data is not important, couldn't the test use some arbitrary other piece of data that is already present in the scenario. Let say, the init binary?

That's a good idea! I removed the raw archive again and used exactly this approach. 

There is one thing that I don't fully understand: The role of directories on the client side. E.g. when the two files `upstream` and `downstream` on `test-libc_fifo_pipe` are within the same directory (right under `/dev` and not within `/dev/reader` and `/dev/writer`), the scenario behaves differently. For one, I noticed in some rare cases a dead-lock, even though each file has its own File_system session. The other problem is that having `upstream` and `downstream` within the same directory, yields a failure of the `test.access_control()` test: `Error: should not have read access to /dev/upstream`. Is there a rationale behind the different behavior, or is there a bug?

sidhussmann, 2021-02-01:
Hi @nfeske, is there anything I can further do on my side to get this feature staging ready?

nfeske, 2021-02-01:
Please bear with me. I haven't managed to review your latest change yet.

Regarding your observation reported in the last paragraph, I have unfortunately no clear explanation but a vague guess. The obvious difference between both configurations is that in the one case, the files provided by the two instances of the pipe file systems share one directory, which treats the files like a union file system. That means, when a file is accessed, both instances are asked. To see if this ambiguity is the problem, it might help to instrument the functions by printing the `this` pointer of the pipe-file-system instance. If only one pointer is present in the log output, we know that only one instance responds to all requests and the other just sits there idling.

In the other case where each instance lives in a different directory, no directory entries are shared, which rules out the chance of ambiguities.

sidhussmann, 2021-02-01:
No worries. I fully understand. Thank you for your additional suggestions. I will take a look.

sidhussmann, 2021-02-02:
@nfeske I have pinpointed a problem in my access control mechanism. I'll keep you posted.

sidhussmann, 2021-02-03:
@nfeske thank you again for your last hint. With commit 2b39eb7b4323c03f3c6188652c9c48c78826f3cd access control is now correctly enforced, even when two `fs` sessions share one common directory on client-side. From a users perspective, access control can be done rather intuitively with the pattern `root="/pipe_name/.out" writeable="no"`. Internally this is verified with the pattern `/pipe_name/.out/pipe_name`.

Additionally, I was able to pinpoint the dead-lock I mentioned previously. I noticed that the dead-lock did not only happen when two `fs` sessions share one common directory on the client-side - it just happened more often than with separate directories. For this, I refactored the read/write synchronization in 635f4310362c06092b81b9e690161ce90d9e538a. As noted by @ehmry in his original work, the synchronization between readers and writers was a temporarily hack. It resulted that all pipe-ends were notified at the same time, which leads to dead-locks when the buffers are congested. I solved this by having a signal handler for each pipe-end.

Note that in case there is no policy defined, in combination with two `fs` sessions sharing one common directory on the client-side, dead-locks can still happen with larger data transfers. However, I believe this happens purely on the client by congesting the `File_system` session buffers. This particular scenario can be mitigated with separate directories, as you described in [1].

[1] https://lists.genode.org/pipermail/users/2020-December/007389.html

nfeske, 2021-02-05:
@sidhussmann Thank you for the update and the good news. I skimmed over the latest changes and think that the pipe-plugin shapes up nicely. I also appreciate the refactor commit, which makes the code easier to understand.

I have one remark/question regarding the naming of the in/out paths though. In an earlier posting (https://github.com/genodelabs/genode/issues/3583#issuecomment-759589069), I pointed you to the convention of our notion of control directories for pseudo devices, named after the device but with a prepended dot. You can find this idea also illustrated in the picture https://genode.org/documentation/release-notes/cmus.png for the terminal plugin and OSS plugin.

Now seeing that you prepended the out directory with a dot, I'm wondering what prompted you choosing this scheme over the existing convention? If I understand correctly, in your latest version, the directory structure looks as follows
```
/pipe_name                  <- can be written and read
/pipe_name/.out/pipe_name   <- can only be read
/pipe_name/.in/pipe_name    <- can only be written
```
With the convention of the other plugins, it would instead look as follows
```
/pipe_name              <- can be written and read
/.pipe_name/            <- is a directory (note the dot)
/.pipe_name/in/in       <- can only be read
/.pipe_name/out/out     <- can only be written
```

I think the latter has the following advantages:
* It is consistent with other VFS plugins.
* The `.pipe_name/` is not a hybrid of a directory and pseudo file. It is a regular directory.
* The pseudo files inside the `in/` and `out/` directories have generic names, which do not reveal the name of the pipe. The consumer and producer does not need to know the name of the pipe. So they shouldn't know. But it is certainly good to name each end distinctly. This way the ends of different pipes can be mounted side by side (think of /dev/in, /dev/out) and the role of the pseudo file is clear from glancing at its name.

sidhussmann, 2021-02-08:
@nfeske I took your comment regarding pseudo devices and control directories merely as an encouragement to change the directory structure. It was not clear to me that there was a convention. Admittedly, when looking at your illustration, it makes much more sense now. I must also point out that my desire to keep two files within one directory on the client had lead me to a rather rocky path. I was under the impression that the ambiguity you mentioned [1] can be solved on the server-side within the pipe plugin. The mechanisms of how files and directories map to file-system sessions were not so obvious to me. Tweaking the directory structures left me sometimes with working scenarios, sometimes not. It had lead me to the (false) conclusion that we'd need the actual pipe-name at the end of the path (e.g `/pipe_name/in/pipe_name`). I learned that the order of the client-side files within the same directory matters. I now have a clearer picture of how the client-side of the file-system session works. It is always safer to separate the files in different directories. However, I have the feeling that this ambiguity you mentioned in [1] is quite a pitfall to trap into - not _only_ regarding file-system sessions to fifo pipes. I'm not sure how this can be solved, though.

Nevertheless, thank you for sharing your ideas and suggestions! I'm pretty sure we now share a common view of how the plugin should be structured. Commit fe955b6667f28554ea63f661de63b5bd2b637980 introduces the pattern you suggested. 

[1] https://github.com/genodelabs/genode/issues/3583#issuecomment-770975098

nfeske, 2021-02-11:
@sidhussmann Thank you for taking my feedback to heart and for reworking the code accordingly.

I gave your fixup a read and left two remarks. My overall impression is that the patch has reached a very good state. So I'd encourage you to squash the sequence of commits into a form that you deem suitable for staging. I'll then give the overall result a thorough review again.

> I'm pretty sure we now share a common view of how the plugin should be structured.

Even though this issue took quite many iterations, I think it greatly reinforced our mutual understanding. I very much appreciate your perseverance and your openness to my feedback.

> However, I have the feeling that this ambiguity you mentioned in [1] is quite a pitfall to trap into - not only regarding file-system sessions to fifo pipes. I'm not sure how this can be solved, though.

You are not alone. I agree that the current implementation is still too complex. The problem is that the single plugin entangles two aspects that should better be handled separately:

1. The handling of dynamic directory content. In contrast to plugins that provide only a fixed set of pseudo files, the pipe plugin supports multiple pipes appearing and disappearing during runtime.
2. The provisioning of a nested directory structure.

For the second aspect, we already have a solid implementation in the form of the `Dir_file_system`. So this is strictly speaking a  solved problem whereas the pipe plugin solves the same problem again but in a more ad-hoc manner. For example, the handling of reading directory entries of the pseudo directory structure remains merely an afterthought. For several modern VFS plugins, we managed to employ the `Dir_file_system` for this dirty work. E.g., look at the nice way how the terminal plugin implements its pseudo-directory structure (https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/vfs/terminal_file_system.h). There is no bug-prone fiddling with file paths at all. Unfortunately the aspect (1) prevents us from using this pattern for the pipe plugin today.

The same observation applies to the lxip plugin, the lwip plugin, and the trace plugin. In the longer term, we should strive to solve the aspect (1) in the form of a reusable utility. There are several design options. But that discussion would fall outside the scope of this issue.

Even though there is room for improvement (there always is), I think it is safe to say that the current state of the plugin is pretty cool! :-)

sidhussmann, 2021-02-11:
@nfeske I squashed the commits to these two: d74ace2ed383ada6c305036cfe724071feabbd53 and 6f4c87c6eca14f435b82666ca4d45f12d7b13699

> Even though this issue took quite many iterations, I think it greatly reinforced our mutual understanding. I very much appreciate your perseverance and your openness to my feedback.

I agree. From my perspective it was worth the effort.

> For the second aspect, we already have a solid implementation in the form of the `Dir_file_system`.

I did notice this class in the terminal file-system plugin. It is indeed quite elegant. I wasn't sure how much effort it would cause, to use it. I'm glad to hear that it would have been a dead end. 

> In the longer term, we should strive to solve the aspect (1) in the form of a reusable utility. There are several design options. But that discussion would fall outside the scope of this issue.

Agreed.

> Even though there is room for improvement (there always is), I think it is safe to say that the current state of the plugin is pretty cool! :-)

Thanks! Please let me know if you find anything further. I'm willing to walk that extra mile.

nfeske, 2021-02-12:
While test-driving the commit, I observed a problem with the bash.run scenario.

When issuing the following command

```
$ ls -1 | wc -l
```

it gets stuck. In the log, I see the following message:

```
[init -> /bin/bash -> 1] Error: Uncaught exception of type 'Libc::Symlink_resolve_error'
[init -> /bin/bash -> 1] Warning: abort called - thread: ep
```

sidhussmann, 2021-02-12:
> While test-driving the commit, I observed a problem with the bash.run scenario.
>
> When issuing the following command
>
> ```
> $ ls -1 | wc -l
> ```
>
> it gets stuck. In the log, I see the following message: [...]

@nfeske fixed in d83a0e18930797e41297bb4adbe5abd1b34b96b4

nfeske, 2021-02-15:
Thanks @sidhussmann for the fixups. I can confirm that the bash.run script works fine now.

I squashed the fixups to their corresponding commits and merged both to staging now.

nfeske, 2021-02-17:
Commit https://github.com/genodelabs/genode/commit/efc7c282ed951019f791ba6097c1a68c2fd7d6db fixes the depot_autopilot test on various platforms.

nfeske, 2021-02-17:
Unfortunately, on Fiasco.OC, the test gets stuck early on. The problem can be reproduced via test.run on Qemu/x86_32 and Qemu/x86_64.

```
Genode 20.11-190-gff05e91d12
471 MiB RAM and 9237 caps assigned to init
[init -> test -> fifo-pipe-test] test started with 15 iterations
[init -> test -> fifo-pipe-test] --- test iteration 0 started ---
[init -> test -> fifo-pipe-test] re-starting echo
[init -> test -> fifo-pipe-test] starting thread to send data to pipe
[init -> test -> fifo-pipe-test] starting thread to receive data from pipe
[init -> test -> fifo-pipe-test] joining sender thread 
[init -> test -> init -> echo] echo started
[init -> test -> fifo-pipe-test] sent 401408 bytes
[init -> test -> fifo-pipe-test] joined sender thread
[init -> test -> fifo-pipe-test] joining receiver thread 
[init -> test -> init -> echo] piped 401408 bytes
[init -> test -> fifo-pipe-test] received a total of 401408 bytes
[init -> test -> fifo-pipe-test] joined receiver thread
[init -> test -> fifo-pipe-test] --- test iteration 1 started ---
[init -> test -> fifo-pipe-test] re-starting echo
[init -> test -> fifo-pipe-test] starting thread to send data to pipe
[init -> test -> fifo-pipe-test] starting thread to receive data from pipe
[init -> test -> fifo-pipe-test] joining sender thread 
Error: Test execution timed out
Makefile:392: recipe for target 'run/test' failed
```

@sidhussmann May you have a look?

nfeske, 2021-02-17:
I'm sorry to report that we are not completely there yet.

The ports/tool_chain_auto.run script fails on almost all tested platforms (linux, nova, hw on x86_64, as well as foc, fiasco, nova, okl4 on x86_32, as well as hw on ARM). Strangely, it works on one single platform that is foc on x86_64.

The symptom of the failure looks always the same:

```
[init -> vfs] creating build directory...
--- start of time measurement ---
[init -> /bin/bash -> 3] Warning: unsupported ioctl (request=0x80106678)
[init -> /bin/bash -> 3] Warning: unsupported ioctl (request=0x80106678)
[init -> /bin/bash -> 3 -> 2 -> 11] Warning: utimes: utimes not implemented
```

After these messages, the scenario gets stuck.

I double checked that the "vfs/pipe: add fifo feature to pipe plugin" commit is the troublemaker. When reverting this commit (and its fixups) from staging, the tool_chain_auto scenario works.

Unfortunately, this is a show stopper. @sidhussmann Would you be able to investigate the issue? I think the best way to investigate is the use of the base-linux platform, where GDB can be used to attach to the blocked processes. The hints I gave in https://github.com/genodelabs/genode/issues/4000#issuecomment-767544755 may be of help.

sidhussmann, 2021-02-17:
@nfeske I agree. I will take a look.

sidhussmann, 2021-02-18:
@nfeske commit c3dbeefa7366794c3e4bd442e8c93487a74fcbb8 fixes `tool_chain_auto.run`. I was able to reproduce the deadlock you mentioned for Fiasco.OC also on base-linux, but only after ~2000 iterations. Commit d0b5c5991a3d8d34377e36685bcb7e4ffd8751c0 fixes this issue. My stress test just succeeded with 15k iterations. 

I also noticed that a few fixup patches didn't make it to staging: f3c0814cee d7911b5cb4 aa596ee024 and most importantly e3dc4469f9.

nfeske, 2021-02-19:
You really owned it! This makes me super happy. Congratulations, @sidhussmann!

> I also noticed that a few fixup patches didn't make it to staging ...

Don't worry. I already squashed this commit series into the original commit (mentioned in https://github.com/genodelabs/genode/issues/3583#issuecomment-779238819). This was needed because the fixups for the "add fifo" commit collided with the changes of "refactor" commit. I wanted to resolve these conficts before they would enter staging.

The bottom line: I can confirm that the fixes work. The showstopper is gone. :-)

sidhussmann, 2021-02-19:
@nfeske thank you! I just noticed that actually, commit d0b5c5991a3d8d34377e36685bcb7e4ffd8751c0 and c3dbeefa7366794c3e4bd442e8c93487a74fcbb8 should be squashed with 5717b8dae642a258f1fe755eb749a75dc25f0c20 (vfs/pipe: refactor read/write synchronization) and not 6c0f81044b0bc064e94ee06df259591da85be05b (vfs/pipe: add fifo feature to pipe plugin). Do you want me to adjust the commit messages, so it's easier to squash?

nfeske, 2021-02-19:
@sidhussmann Ah, I see. I merely took the commit messages as a guide for squashing.

Could you produce two clean commits out of the existing series? I could then replace the existing commits on staging with your blessed commits.

sidhussmann, 2021-02-19:
@nfeske done: 8e1e55cb19bd2764b4dd8ee8939fffa482b55e88 and d708f79a431eb4dad85ebd5c7cb9cd6ac8b37116

sidhussmann, 2021-02-19:
@nfeske I noticed that sending a signal each time the buffer is full during write [1] has quite a performance impact: On base-linux `fifo-pipe-test` takes 3 seconds longer with 15 test iterations. It seems like we can skip sending the signal. I'll let you know when my tests have confirmed this assumption.

[1] https://github.com/sidhussmann/genode/blob/3583-vfs-pipe-fifo-feature-for-staging/repos/gems/src/lib/vfs/pipe/plugin.cc#L215

nfeske, 2021-02-19:
Thanks. I force-pushed staging with your two commits on top.

> I'll let you know when my tests have confirmed this assumption.

If you find ways to improve the performance, we can still add changes, preferably as new commits.

Speaking of performance, you may consider calling `Signal_handler::local_submit` instead triggering a regular signal. This way, the signal would be handled completely local to the component, which has lower overhead (it comes down to a semaphore-up operation). This optimization was very effective for the libc, so maybe it would improve the performance of the pipe plugin as well? But this is just an (untested) idea.

Another - rather non-creative  idea - would be adding a way to increase the pipe buffer size. For the block-session interface, I observed that the effect of the I/O buffer size on the throughput (between RAM-block server and test program w/o actual I/O) was quite significant (until a buffer size of 4 MiB). That is, using 128 KiB instead of 8 KiB could be quite effective. It would certainly be the best to make the buffer size configurable. Speaking of the buffer, the use of `Ring_buffer` is probably also worth reconsidering. It is convenient but operates byte by byte. But these ideas would better go to a separate issue. ;-)

sidhussmann, 2021-02-19:
@nfeske this sounds a plan. `Signal_handler::local_submit` did improve performance additionally a few percent. Commit is comming up. Also configurable buffer sizes has been in the back of my head as well for a while. Will look into it.

Speaking of a replacement for `Ring_buffer` did you have `Magic_ring_buffer` in mind?

nfeske, 2021-02-19:
> Speaking of a replacement for Ring_buffer did you have Magic_ring_buffer in mind?

As far as I know, this utility does not work on Linux (I'm not sure). It definitely creates a dependency from core's RM service, which is better to avoid. So I'd not consider it.

sidhussmann, 2021-02-19:
Commit 921492a8f6db0c4264f504b30cec4e7c67bf337b adds the performance improvements regarding signal handling.

nfeske, 2021-02-19:
Cool. Merged to staging right away.

nfeske, 2021-02-22:
With the latest version (current staging), the test-libc_pipe fails:

```
[init -> test -> sequence -> libc_pipe_local] Error writing to pipe (bytes_written=-1, BUF_SIZE=16384)
[init -> test -> sequence -> libc_pipe_local] Error: no plugin found for read(4)
[init -> test -> sequence -> libc_pipe_local] Error: no plugin found for fcntl(2)
[init -> test -> sequence -> libc_pipe_local] Error: no plugin found for write(2)
[init -> test -> sequence] Warning: child "libc_pipe_local" exited with exit value 1
```

This is reproducible via

```
build/x86_64$ make run/test PKG=test-libc_pipe KERNEL=linux BOARD=linux
```

sidhussmann, 2021-02-22:
@nfeske commit 01229989ea349d92e5e28f9e4822201f6c69b05e fixes this bug. However, I'm not sure if this fixup is sound. From my perspective, it would be better to return `Write_result::WRITE_ERR_WOULD_BLOCK` during write when buffer is full. This doesn't play well with libc and instantiating the plugin locally. What do you think? Should I investigate further within the libc or is returning `Write_result::WRITE_OK` in this case the way to go?

This patch succeeded following tests on my machine:
- test-libc_fifo_pipe on base-linux (15k iterations)
- test-libc_fifo_pipe on base-foc
- test-libc_pipe on base-linux
- run/tool_chain_auto on base-linux

nfeske, 2021-02-22:
Thank you @sidhussmann for the quick response.

> However, I'm not sure if this fixup is sound.

I cannot give you a definite answer straight away, but since WRITE_ERR_WOULD_BLOCK was never returned in the previous version, it is sensible to keep it that way. Hence, your fixup looks fine to me.

On another note, we have scheduled the Genode 21.02 release for Thursday this week. Are you interested to contribute a few words about the pipe plugin to the official release notes?

sidhussmann, 2021-02-22:
> On another note, we have scheduled the Genode 21.02 release for Thursday this week. Are you interested to contribute a few words about the pipe plugin to the official release notes?

@nfeske sure:

The vfs/pipe plugin now has a named pipe feature. The main motivation was to easily stream data from pure Genode components to libc components via file-system sessions that can be attached to stdin, stdout, and stderr. This feature further makes it possible to pipe several components together, similarly to how it is done on Unix. Additionally, the thread synchronization has been improved so that large data chunks can be transferred without blocking.

nfeske, 2021-02-22:
Thanks for your input. :)

nfeske, 2021-02-23:
The feature entered the master branch, hurray!

closed 2021-02-23 by nfeske