genodelabs/genode #3590: ssh_terminal: ambigouity of return values of `Ssh::Server::auth_pubkey`

nfeske, 2020-01-03:
On staging, there is the following compile warning:

```
repos/gems/src/server/ssh_terminal/server.cc: In member function ‘bool Ssh::Server::auth_pubkey(ssh_session, const char*, ssh_key_struct*, char)’:
repos/gems/src/server/ssh_terminal/server.cc:549:10: warning: enum constant in boolean context [-Wint-in-bool-context]
   return SSH_AUTH_PARTIAL;
          ^~~~~~~~~~~~~~~~
```

The corresponding code (https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/ssh_terminal/server.cc#L513) confuses me. The method somehow maps `ssh_auth_e` enum values to a boolean return value. The enum values are defined as follows:

```
enum ssh_auth_e {
	SSH_AUTH_SUCCESS=0,
	SSH_AUTH_DENIED,
	SSH_AUTH_PARTIAL,
	SSH_AUTH_INFO,
	SSH_AUTH_AGAIN,
	SSH_AUTH_ERROR=-1
};
```

In the first check (session validity), `false` is returned in the error case.
In one branch (SSH_PUBLICKEY_STATE_NONE), `SSH_AUTH_PARTIAL` is returned, which is mapped to a `true` return value.
In the failure branch (at the end of the method), `SSH_AUTH_DENIED` is returned, which is mapped to a `true` return value.
In the success branch (SSH_PUBLICKEY_STATE_VALID), `SSH_AUTH_SUCCESS` is returned, which is mapped to `false`.

I am unable to see a clear pattern. When inspecting the caller site (https://github.com/genodelabs/genode/blob/master/repos/gems/src/server/ssh_terminal/ssh_callbacks.cc#L244) I see that `true` is mapped to SSH_AUTH_SUCCESS whereas `false` is mapped to SSH_AUTH_DENIED. This seems to contradict with the implementation of `auth_pubkey`. I wonder how can the authentication work in the first place because SSH_AUTH_SUCCESS seems to be mapped to SSH_AUTH_DENIED?

How to go about fixing the warning? Intuitively, I think that tunneling the `ssh_auth_e` value through a bool value is the problem here. Changing the return value of `auth_pubkey` to `ssh_auth_e` would make things more clear.

cnuke, 2020-01-06:
@nfeske thanks for pointing out the issue. Commit 2a97a33 fixes the problem. I went for using `bool` and documenting why `true` is returned at certain places as otherwise the callback function would have to handle the results additionally.

That being said, what's more concerning is that without the fix even when providing the wrong publickey the authentication attempt is successful AFAICT.

nfeske, 2020-01-08:
Thanks @cnuke for the fix.

@sidhussmann Please note this issue.

sidhussmann, 2020-01-08:
@nfeske and @cnuke I fixed the compiler warnings in #3475, which is still pending for review. 
The function  `Ssh::Server::auth_pubkey()` is called from libssh via the `session_auth_pubkey_cb()` callback which expects an `int`. So I'm not so sure if returning a `bool` is correct...

Btw, I'm guessing that the page fault should be fixed with the lwip/lxip fixes. The changes in #3475 had several more improvements.

chelmuth, 2020-01-08:
@cnuke I'd much appreciate you looking into this keeping @sidhussmann's remarks in mind. In contrast to you, @nfeske and me don't know much about libssh.

@sidhussmann if you have improvement independent of the pipe feature in #3475 you may open a separate pull request.

sidhussmann, 2020-01-08:
@chelmuth the title of #3475 may be misleading. It does not really have to do anything with pipe on Genode side. The title comes from the need to use pipe on the client (linux) side. The improvements are all worth considering.

cnuke, 2020-01-09:
@sidhussmann you are right that the callback function returns an `int`. Since `auth_pubkey` returns a `bool` the result must be translated (that was already done). Merely changing the signature will fix the warning but the bug @nfeske pointed out is still there.

I think the code ended up the way it currently is because I simply copied the *password* callback implementation when adding *pubkey* support. It stands to reason if it would be better to return the proper value in `auth_pubkey` and remove the translation from the callback function (e.g., `SSH_AUTH_PARTIAL` will not work).

(Sorry for not paying attention to #3475 - please feel free to nudge again if an issue is stale for longer period of time. I'll look into it.)

sidhussmann, 2020-01-09:
All good @cnuke. Thanks for letting me know to nudge again, if I believe one of my issues deserve more attention ;). I have to admit, I never tested SSH with _pubkey_. Please let me know how I can assist with #3475 since it is based on an older Genode version. Btw, there is a new version of libssh. Trying to port it to Genode was not as straight forward as I initially thought (didn't try very hard though).

nfeske, 2020-02-10:
Merged to master

closed 2020-02-10 by nfeske