genodelabs/genode #3596: core: page fault after insane affinity parameters provided during thread construction

alex-ab, 2020-01-07:
```
Warning: unsupported location 2x1
Error: 
PAGE-FAULT IN CORE addr=0xa10068 ip=0x10fe44 (read)

stack pointer 0xa01fdf90, qualifiers 0x4 irUwp
faulter utcb 0xa01ff000, last message item count 0
  #1 0x00000000a01fdf90 0x000000000010fe44
  #2 0x00000000a01fdf98 0x0000000000162d3
```

alex-ab, 2020-01-07:
The undissolved object in [Cpu_thread_componet L149](https://github.com/genodelabs/genode/blob/master/repos/base/src/core/include/cpu_thread_component.h#L149) are the root cause, if in the context of the Rm_client constructor an exception is thrown.

alex-ab, 2020-01-07:
8b1abe388aa9743e5954be770230aef83b2e4973 is my proposal

chelmuth, 2020-01-07:
Related to #1412 ?

alex-ab, 2020-01-07:
> Related to #1412 ?

Kind of. The actual sanitize code detected the in-proper input and denied it by throwing an exception. Unfortunately the generic code failed to handle it. In that sense it is a fallout of #1412.

nfeske, 2020-01-07:
It is probably the best to wrap the `ep.manage` into a dedicated guard object that performs the `dissolve` in the destructor. With commit https://github.com/alex-ab/genode/commit/3611d7bd41cf76db9a6dccab71fdb90e53ea2d64 I see the problem that an exception may be thrown before the call of `ep.manage` which would then result in a `dissolve` call with a not-managed object as argument.

alex-ab, 2020-01-08:
Ok, please have a look at 9511fd36ae2a672bb329e190d7572a4e816342e4

alex-ab, 2020-01-08:
9d7aa41e31a24fa55dda25caf746f3a74af0c35c addresses the comments by @nfeske

nfeske, 2020-01-08:
Thanks for the super quick response. Merged to staging.

commit 26358f88 (2020-01-08): core: dissolve thread object on failed creation

Fixes #3596
commit 5b633a83 (2020-02-04): core: dissolve thread object on failed creation

Fixes #3596
closed 2020-02-04 by chelmuth