genodelabs/genode #3599: sculpt: configure root init according to actual available physical cpus

alex-ab, 2020-01-08:
Up to now in [sculpt.run](https://github.com/genodelabs/genode/blob/master/repos/gems/run/sculpt.run#L36) the available affinity space is set fixed to width of 2. On machines with more than 2 physical cpus, this leads to the effect than not all physical CPUs can be chosen to place components by configuration on it.

Since we don't know the target machines and their CPU configuration count for the Sculpt images, we should add an option to create the affinity space depending on the real available physical CPUs.

Just to show the effect, e.g. a configuration of <affinity-space width="2" height="1"/> leads to

4 core physical machine:

```
<affinity-space width="2" height="1"/>
<start name="component 0"> <affinity xpos="0" width="1"/> ...    -> physical core 0
<start name="component 1"> <affinity xpos="1" width="1"/> ...    -> physical core 2
```
Core 1 and 3 can't be directly used by configuration

8 core physical machine:

```
<affinity-space width="2" height="1"/>
<start name="component 0"> <affinity xpos="0" width="1"/> ...    -> physical core 0
<start name="component 1"> <affinity xpos="1" width="1"/> ...    -> physical core 4
```
Core 1,2,3 and 5,6,7 can't be directly used by configuration

alex-ab, 2020-01-08:
acac60740d4ac65df25087da5fc954a9a53729ee and 5168ea4438dc44b8b201a4b8adb7244faae8cda5 are my proposal to implement such a feature. We may also add an explicit option to the affinity-space configuration node to use the native/hardware space or to fall back to width/height 1,1 as done up to now when no configuration at all was given - i'm open for your suggestions.

alex-ab, 2020-01-16:
Tested the both commits, which leads to a affinity-space runtime configuration matching the actual available physical CPUs. With #3597 one can now place components as desired.

nfeske, 2020-01-17:
I merged your commit (with a minor alignment change) and added another fixup needed for leitzentrale.run

alex-ab, 2020-01-17:
Thanks, please add also ae697efadcba6fe37b8d140d9742b1c089ed9868, otherwise the sculpt commit will have no effect.

nfeske, 2020-01-17:
Thank you for the reminder. Now, I merged this one also.

nfeske, 2020-02-28:
Fixed in master.

closed 2020-02-28 by nfeske