genodelabs/genode #3610: trace: optimize requesting execution time of all threads

alex-ab, 2020-01-16:
Currently to request for all threads the execution time at one reading point, as done by cpu_load_monitor, top and top_view, one has to do 1+n RPC calls (1 x subjects() + n x subject_info()). In case of Sculpt n becomes quickly large (>500). If requesting this data in short time internals (< 100ms),  the overhead  becomes not negligible anymore.

alex-ab, 2020-01-16:
@nfeske: please review, for test purposes at moment i adjusted solely top

alex-ab, 2020-03-18:
May I ask what blocks this feature to be considered ?  I have several follow up issues I would address but I'm blocked since I got no response since over 2 months.

nfeske, 2020-03-18:
Until the release of Sculpt 20.02 I did not consider the feature because of ABI-stability. The priorities of Sculpt and Genode 20.02 were elsewhere.

Thanks for bringing the issue back to my attention. Now is indeed a good time to proceed with it.

nfeske, 2020-03-19:
Now I had a look at the patch.

* Why is the new type `Dynamic_info` needed? There seems to be a lot of redundancy with the existing `Subject_info` type. What is the rationale behind the name of the type? As far as I can see, there is nothing more dynamic about a `Dynamic_info` than about a `Subject_info`. If possible, I'd like to avoid introducing the new type.

* Please pay attention to the vertical whitespace (two empty lines between subsequent class definitions).

* Please let us avoid C-style interfaces (as usual, for memory-safety reasons). I'm generally trying the clean up the Genode API in this respect. The method `Trace::Session_client::dynamic_info` goes in the opposite direction. Please consider an interface like the following instead:

```
template <typename FN>
void with_subject_infos(FN const &fn) const
{
  /* request batch of subject infos marshalled in the shared-memory buffer */
  ...

  /* call 'fn' with an index and 'Subject_info const &' for each subject info */
  ...
}
```

With such an interface, there is no need for any memcpy or the use of any pointer at the public API. The copy-out is the responsibility of the caller and can be performed by a plain assignment.

alex-ab, 2020-03-19:
> Why is the new type Dynamic_info needed? There seems to be a lot of redundancy with the existing Subject_info type. What is the rationale behind the name of the type? As far as I can see, there is [...]

The idea is that Dynamic_info contains no static data, like thread name and session label. If you compare the sizes of Dynamic_info vs Subject_info you get 48 vs 256 Bytes. Imagine 700 and more threads like in Sculpt, you have to spent 145k+ more memory to core just to request the execution time (dynamic part) compared to using Subject_info. Requesting the dynamic data may happen quite often (<50ms), so copying as less (useless static info) as possible is desired to not waste CPU time for others.

I did not change the Subject_info in the first place, in order to keep the API stable. I can break of course the API by putting into Subject_info also the Dynamic_info, e.g.

```
--- a/repos/base/include/base/trace/types.h
+++ b/repos/base/include/base/trace/types.h
@@ -122,10 +122,8 @@ class Genode::Trace::Subject_info
 
                Session_label      _session_label  { };
                Thread_name        _thread_name    { };
-               State              _state          { INVALID };
                Policy_id          _policy_id      { 0 };
-               Execution_time     _execution_time { 0, 0 };
-               Affinity::Location _affinity       { };
+               Dynamic_info       _dynamic        { };
 
        public:
```

What do you think ?

nfeske, 2020-03-20:
Thanks for this good explanation. Now the picture has become much more clear.

That said, I don't share your ambition for optimizing the memory bandwidth here. Your patch eliminates 700 RPCs - that is a fantastic improvement! Shaving off a few KiB from the memory copy will be negligible in comparison. Let's numbers speak. A copy of 700 Subject_info objects is still less than 200 KiB. At a (high) rate of 50ms, this results in 4 MiB/s memory bandwidth. Regular Sculpt systems have a memory bandwidth of more than 3 GiB/s. So we speak of less than 0.15% of the available bandwidth - at a 50 ms rate!

What the numbers are trying to tell: Let's not try to optimize here but instead keep the interface as simple as possible. Hence I argue to stick with only the `Subject_info` type. This has the added benefit that the new mechanism can eventually substitute the original RPC-based one (possible because it covers full feature set, not merely a subset) once the day has come, there is no need to match `Dynamic_info` objects with `Subject_info` objects at the client side, and there is no need for any explanation of the distinction between dynamic info vs. subject info. Should we indeed end up seeing the fat `Subject_info` objects as a bottleneck, we can still optimize it then.

As for API/ABI compatibility, since Genode/Sculpt 20.02 is out, we can move freely now. So let us change the API/ABI where it makes sense. There are not too many TRACE clients after all.

alex-ab, 2020-03-20:
Admittedly, I don't share the same mindset as you about using available compute resources efficiently. In my mindset tracing should be as less as noticeable as possible. But  before we go into a discussion leading nowhere, I will try to come up with another patch which is aligned with your view asap.

nfeske, 2020-03-20:
> But before we go into a discussion leading nowhere, ...

You may reconsider leaving such salty statements in the issue tracker then.

alex-ab, 2020-04-28:
@nfeske: I reworked the commits according to your suggestions.

nfeske, 2020-04-28:
Thanks. I appreciate it. Merged to staging.

nfeske, 2020-05-18:
Merged to master.

closed 2020-05-18 by nfeske