genodelabs/genode #3612: base/cxx: deadlock during __cxa_allocate_exception

alex-ab, 2020-01-21:
The run/thread test cases fails reliable on various hw platforms. After debugging it turns out that the allocation of an exception 

https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/lib/base/ipc.cc#L106
leads to allocation calls in
https://github.com/genodelabs/genode/blob/master/repos/base/src/lib/cxx/malloc_free.cc#L67.

The heap in this specific case tries to throw an exception holding an internal heap lock. cxx/malloc_free.cc is used to allocate that new exception which ends up in the same cxx_heap and deadlocks on the already taken lock.

alex-ab, 2020-01-21:
With the provided debug commits, the following piece is of interest:
```
100: Kernel: 0x401fdf54 malloc 132 136 [0xda6a8-0xdb704)
101: Kernel: 0x0 malloc 132 136 [0x800f707c-0x800f80d8)
102: Kernel: 0x0 malloc done 132 0x800f80e4
103: Kernel: 0x0 malloc free 0x800f80e4
104: Kernel: 0x401fdf54 malloc 132 136 [0xda6a8-0xdb704)
105: [init -> test-thread] Error: xxxxxxx 0xda6ac
106: [init -> test-thread] 0x61b50
107: Kernel: stop me helper(0x401fdf54) owner=0x401fdf54
108: Kernel:  -> helper
```

Thread 0x401fdf54 starts an allocation at 100. At 104 the same thread tries an new allocation, without leaving the heap.alloc() call with the "malloc done" message in the trace. The thread tries to re-take the lock and finds itself as lock holder/owner and goes to sleep. All remaining 9 threads in this thread test block at the same lock eventually.

alex-ab, 2020-01-24:
I added 4 prototype commits of the idea to separate the lock usage cases as discussed today in the morning. With the commits, the deadlock becomes apparent. Please review whether it was what all had in mind.

nfeske, 2020-01-24:
Cool that you went forward already!

A few remarks after glancing over the commits:

* I'd prefer to host the `Lock` as a member of `Mutex` and `Blockade`. In general, I think it is best to avoid inheritance whenever possible.
* `Mutex::lock` could be named `Mutex::acquire`, I guess that you kept the names to stay compatible with `Lock::Guard`. But with the prospect of eventually hiding the `Lock`, I would instead introduce a new `Guard` in the scope of `Mutex`.
* `Mutex::unlock` could be named `Mutex::release`
* The implementation of the `Mutex` methods should go to the `cc` files of the base library. Otherwise, the log template would be instantiated at each caller site. Also the log calls would make `mutex.h` and `blockade.h` unnecessarily depend on `log.h`.

alex-ab, 2020-01-28:
I adjusted the commits on https://github.com/alex-ab/genode/commits/issue_3612 according to your suggestions, thanks. Please re-check.

I adjusted only some cases with the new Genode::Mutex / Genode::Blockade mechanism. If wanted, I may go all over the code and replace Genode::Lock. Or do we don't want to rush ?

nfeske, 2020-02-13:
Thanks for update. I merged your commit series to staging.

alex-ab, 2020-02-17:
It turns out that the second Out_of_memory exception is actually not caused by the -client- nor the generic -core- part, but by the -hw- kernel.

During an IPC to core (by cxx_heap()) the call actually never reaches the generic core part (Pd_session_component::alloc()), but instead is canceled by an Out_of_memory exception during [capability allocation](https://github.com/genodelabs/genode/blob/staging/repos/base-hw/src/core/kernel/thread.cc#L44) which [is translated to an aborted syscall](
https://github.com/genodelabs/genode/blob/staging/repos/base-hw/src/core/kernel/thread.cc#L789) by the -hw- kernel. On [client side](https://github.com/genodelabs/genode/blob/master/repos/base-hw/src/lib/base/ipc.cc#L106)  it is turned again into an Out_of_memory exception and we ran into the deadlock.

Increasing the support maximum capability count helps 4c07bf68e7590391d7681977bb02aeb339493b13, but potentially it can be solved more clever @skalk  ?

skalk, 2020-02-18:
Well this is just a cosmetique workaround. As far as I understood it correctly, your first analysis is still valid. A client provokes an exception, during which the CXX library's heap needs an additional allocation. This results in an IPC call to core. During the call within the kernel, a capability needs to be allocated (during all IPC calls this needs to be done eventually for capability reception), but the capability slab is empty at this point. Therefore, the IPC returns with an error, which results in throwing a new exception in user-land... Increasing the capability slab just moves the race-window. Am I right?

alex-ab, 2020-02-18:
Yes, you are right. The obvious observation is that one never ever must throw an second exception during memory allocation of the first exception. For me the conclusion is that we can't use any Genode allocators which throws exception in the CXX library (e.g. heap). The second conclusion also is, that any allocator which don't throw but make IPC to -others-, e.g. parent/services/core, are also unusable. The -others- may throw exceptions for whatever reason, like in this case the -hw- kernel because of insufficient space for capabilities.

alex-ab, 2020-03-25:
I updated the commits regrading replacement of Lock with Blockade/Mutex in core, base lib, os as far as possible. If possible please consider the commits of https://github.com/alex-ab/genode/commits/issue_3612 next time as appropriate. The commits were on merge_to_staging already for some weeks to avoid breakage.

nfeske, 2020-03-25:
Thanks for the reminder.

The patches look very good to me. From merely looking at the patch, I haven't grasped the intention behind the change regarding the `_exit_handler.exit`. I'll review the code again to find out.

Thanks for the thorough cleanup work. I'll merge your commits in the next round of staging. I hope that will be tomorrow.

alex-ab, 2020-03-25:
I added a fixup which is hopefully easier to grasp. The weeks before when I did the patch initially I tried to avoid to introduce a bool in the generic rpc_server.h. I let the decision to you.

chelmuth, 2020-03-26:
Also with the fixup I'm still wondering if this is just an optimization for the unsynchronized startup or actually required in a corner case unknown to me.

alex-ab, 2020-03-26:
The intention is to avoid semantic misuse of the Blockade mechanism. You may also do something like:

```
ep._delay_start.block();
ep._delay_start.unblock();
```
on each incoming IPC for the Entrypoint. On review one would come to the conclusion, that this not the intention of the _delay-start Blockade.

chelmuth, 2020-03-26:
Now I get what you mean: Formerly the code used `{ Guard { _delay_start }; }` as blockade effectively calling lock/unlock on each RPC. Nevertheless, I'd not introduce your optimization with the Mutex/Blockade patch but vote for removal of support for `start_on_construction == false` after Noux vanished.

alex-ab, 2020-03-26:
> Nevertheless, I'd not introduce your optimization with the Mutex/Blockade patch

As said, it is not about an optimization, but instead about usage of Blockade as intended.

> but vote for removal of support for start_on_construction == false after Noux vanished.

That's a good point, but a different issue and not related to this one.

alex-ab, 2020-03-26:
The new staging breaks all the commits of this issue, so I will have to redo this again. What is now the decision finally regarding the _delay_start issue ?

chelmuth, 2020-03-26:
> The new staging breaks all the commits of this issue, so I will have to redo this again.

That's unfortunate and also not expected as the only staging commit touching base could be the `wait_and_dispatch_one_io_signal` fix.

> What is now the decision finally regarding the _delay_start issue ?

Please adapt this change to the sketeched block/unblock pair without additional flag. We will remove the indecent use of `Blockade` after Noux was removed.

alex-ab, 2020-03-26:
Ok, I see.

alex-ab, 2020-03-30:
I adjusted the commits and according to the merge_to_staging result the branch can be considered next time as appropriate.

nfeske, 2020-03-30:
Thanks. I merged the three commits to staging.

alex-ab, 2020-03-31:
Please add a93e86dfdf2cd58d0e1531ab85af9da2a0b89a81  and 4da062915a9b4741f04e6ae0ee0a7d094722553c to staging

alex-ab, 2020-03-31:
> Please add [a93e86d](https://github.com/genodelabs/genode/commit/a93e86dfdf2cd58d0e1531ab85af9da2a0b89a81) and [4da0629](https://github.com/genodelabs/genode/commit/4da062915a9b4741f04e6ae0ee0a7d094722553c) to staging

Sorry, please use a93e86dfdf2cd58d0e1531ab85af9da2a0b89a81 and 50d040b50093955ba94b92de1ad6c7b2f4e4be9e. (The fixup of the second commit was wrongly named)

chelmuth, 2020-03-31:
Merged both commits to staging.

nfeske, 2020-04-01:
The following depot-autopilot tests trigger mutex-related messages on all platforms:

libc_vfs_audit, libc_vfs_fs, libc_vfs_ram

@alex-ab May you have a look at them?

alex-ab, 2020-04-01:
I will check.

alex-ab, 2020-04-01:
In [vfs/ram_file_system](https://github.com/genodelabs/genode/blob/staging/repos/os/src/lib/vfs/ram_file_system.h#L834:L843) two guards are used with the same lock/mutex. The question is, is it considered good usage ?

alex-ab, 2020-04-01:
With the proposed commits the message goes away.

commit 056500ea (2022-01-24): base-hw: remove throw from ipc_call code path

This eliminates the dependency of the IPC-call operation from the cxx
heap.

Fixes #3612
commit 3e4af3a5 (2022-02-15): base-hw: remove throw from ipc_call code path

This eliminates the dependency of the IPC-call operation from the cxx
heap.

Fixes #3612