genodelabs/genode #3620: Virtio device models for console and network in ARM VMM

skalk, 2020-01-23:


skalk, 2020-01-23:
@ssumpf there seems to be a minor issue left in the virtio model. when using the modified run-script in my issue#3620 branch, one can connect to a TCP terminal running on the i.MX8 board that listens on port 4321. After that when issueing a constant network flow to the guest, e.g. by opening a telnet connection, or by using netcat, I can observe that after some time obviously the guest OS does not receive any network packet anymore, either ARP, ping or tcp responses, although the nic_router sends them to the vmm.

ssumpf, 2020-01-24:
@skalk: Okay, will have a look.

skalk, 2020-02-21:
@chelmuth @nfeske: This feature is finally ready for staging. The following commit series is needed: 8de7138 e3039e3 04db80b 7a5fef6 195e562 8642677

nfeske, 2020-02-21:
Thanks for posting the commit series. Merged to staging.

commit 93fac419 (2020-02-21): vmm: ARMv8 virtio for network and console

Simple console and network implementation of the Virtio protocol.

Fix #3620
commit f7753113 (2020-02-27): vmm: ARMv8 virtio for network and console

Simple console and network implementation of the Virtio protocol.

Fix #3620
closed 2020-02-28 by chelmuth