genodelabs/genode #3623: hw/x86: add write combining support

alex-ab, 2020-01-29:


alex-ab, 2020-01-29:
@m-stein: please try. I was not able to use the Mt: Bitset_3, maybe you can convince the compiler.

m-stein, 2020-01-29:
Wow, cool :) I'll give it a try it in an hour when I'm in the office.

chelmuth, 2020-02-06:
I added the fixed label as d69bf82262ebb7a8c21d8b421e1a93767b263af1 includes the *magic word*. Would you still like to change something (e.g., `Bitset_3`) before merge?

alex-ab, 2020-02-06:
I plan from my side no changes, I would the decision up to @m-stein.

m-stein, 2020-02-06:
@chelmuth I made some minor adaptions and pushed it on my ``merge_to_staging``:
f220a54559 hw/x86: use PAT for write combining support
5ccba8a400 hw/x86: add cpuid register macro
b1d5e6b709 hw/x86: implement wrmsr in macro

@alex-ab I didn't change the places were you wanted to use bitset ``Mt`` as I think it's quite understandable this way and I didn't get the point what would be the benefit of using the bitset, to be honest.

chelmuth, 2020-03-11:
Merged to staging again.

commit c634314d (2020-02-28): hw/x86: use PAT for write combining support

Fixes #3623
commit 4da72c24 (2020-03-11): hw/x86: use PAT for write combining support

Fixes #3623
commit e096202b (2020-03-26): hw/x86: use PAT for write combining support

Fixes #3623
closed 2020-03-26 by nfeske