genodelabs/genode #3625: nic_router: fix Ethernet destination MAC in domain broadcasts

m-stein, 2020-01-29:
When the router receives a packet for a domain-local IP that is not the routers IP at that domain it broadcasts this packet to all interfaces of the domain. Unfortunately, it does only adapt the Ethernet source MAC in this case. The Ethernet destination MAC stays the same, the same as the new source MAC, the router MAC.

It stands to reason whether the router should instead start an ARP request to find the correct MAC (and thereby the correct interface which would prevent broadcasting) or set the destination MAC to the Ethernet broadcast MAC.

m-stein, 2020-01-29:
A third option would be to implement an ARP proxy for such packets.

m-stein, 2020-01-29:
This is the quickest fix by simply using a broadcast MAC:
15cb223be2 nic_router: fix Eth dest MAC in domain broadcasts

skalk, 2020-02-06:
This commit breaks at least nightly depot_autopilot tests of test-libc_connect_lxip and test-libc_connect_vfs_server_lx.

chelmuth, 2020-02-06:
The problem statement above confuses me because the router seems to change MAC addresses in a pure packet-switching scenario. Please correct me if I'm wrong but I until today understood *domain-local* interfaces share an ethernet segment.

chelmuth, 2020-02-06:
For now, I reverted the commit on staging to return to a state with the above-mentioned tests working.

Also, I did a quick test validating my assumption of the packet-switching resp. hub scenario by commenting out the following lines. The test also worked with the change.

https://github.com/genodelabs/genode/blob/a6e6f38330cb40c1d04e521b6c3c4530c2cba54d/repos/os/src/server/nic_router/interface.cc#L793-L794

m-stein, 2020-02-07:
@chelmuth You were absolutely right, and removing those lines was the only thing needed to make the router really act as hub on domain local traffic. @ssumpf tested it with the Linux VMM + Ping scenario for me - thanks! Here we go:

 ad31e7cd6f nic_router: do not change ethernet header as hub

commit dae3dc82 (2020-02-07): nic_router: do not change ethernet header as hub

When forwarding domain-local traffic like a hub, the NIC router shouldn't
modify packets at all.

Fixes #3625
commit d56b21d3 (2020-02-10): nic_router: do not change ethernet header as hub

When forwarding domain-local traffic like a hub, the NIC router shouldn't
modify packets at all.

Fixes #3625
closed 2020-02-10 by nfeske