genodelabs/genode #3626: base-hw x86 sculpt: uncaught exception in core

m-stein, 2020-01-29:
When I'm running genodelabs/staging Sculpt with base-hw on my t430s, the system sometimes gets completely stuck with this error:

```
Error: Uncaught exception of type 'Genode::Reconstructible<Kernel::Core_object<Kernel::Thread> >::Deref_unconstructed_object'
Warning: abort called - thread: pager_ep
Error: thread 'pager_ep' tried to self de-struct - sleeping forever.
```

The problem can be triggered pretty reliably by freshly downloading the sticks-blue-backdrop package (to a RAM FS using wired connection).

I was able to observe the error in rare cases even on startup of Sculpt.

m-stein, 2020-01-30:
I just found that the situation is not as bad. The problem seems to be introduced by a commit for base-hw that I added as preparation for the implementation of Spunky in #3308 (but it is not related to Spunky). I don't know why I could observe the error yesterday with staging - now I cannot reproduce it with staging anymore.

skalk, 2020-01-31:
@m-stein you took a load off my mind ;-). So can we close this issue again?

m-stein, 2020-01-31:
May I close it as soon as I managed to track the bugs down? So, I can be sure to be right this time :)

skalk, 2020-02-21:
@m-stein: I'm pretty sure you found the issue in one of your commits soon after this discussion, so I'll close this issue.

closed 2020-02-21 by skalk

m-stein, 2020-03-05:
@skalk Ah, thanks! Yes, this was the case :)