genodelabs/genode #3635: nova: use multi dimensional Genode::Affinity to group hyperthreads in y-axis/height

alex-ab, 2020-02-05:


alex-ab, 2020-02-19:
@chelmuth: i updated the issue_3635 branch according to our discussion

chelmuth, 2020-02-28:
I rebased the commits on master and created a topic branch in my repo. Still don't know if the test program could just be dropped or renamed (from `aff` which sounds like an insult in Rheinland).

alex-ab, 2020-02-28:
Thanks, I think the test commit can be dropped.

alex-ab, 2020-02-28:
Just as note, the top_view packages are not yet prepared in case you want to use it with Sculpt.

chelmuth, 2020-03-11:
I'd like to merge this soon, but need your acknowledgement that it's tested with Sculpt (or had to add the `TEMPORARY` tag.

alex-ab, 2020-03-11:
Acknowledged - sending this message from a running Sculpt 20.02 and the patches of this issue.

commit 96f0338c (2020-03-23): nova: support multidimensional affinity space

Fixes #3635
closed 2020-03-26 by nfeske