genodelabs/genode #3636: ahci: switch to Request_stream API

ssumpf, 2020-02-05:
Remove the dated `Block::Driver` front end and update to `Block::Request_Stream` API.

ssumpf, 2020-02-05:
@everyone: Whoever is brave enough to test this on their hardware box, I would appreciate it. Be aware, it might swipe your hard disk :wink:

m-stein, 2020-02-07:
@ssumpf I'll give it a try.

cproc, 2020-02-07:
@ssumpf: I didn't see problems with ahci_block.run anymore with your latest version

ssumpf, 2020-02-07:
> @ssumpf: I didn't see problems with ahci_block.run anymore with your latest version

Super! Thx for testing! Also thanks to @m-stein for trying out Sculpt with the latest version.

ssumpf, 2020-02-10:
I have also successfully tested the driver on Sculpt with a Debian VM and Seoul Firefox.

ssumpf, 2020-02-10:
@chelmuth, @nfeske: Please review 92a0559, 24a7fbb, and 52d7af5 and consider for staging.

nfeske, 2020-02-10:
The commits look very good to me. I left two minor notes at the commits.

It is really good to see the AHCI driver becoming so tidy and neat. Thanks @ssumpf for the profound restructuring work!

ssumpf, 2020-02-10:
@nfeske: 2284878 implements our findings

nfeske, 2020-02-10:
Thanks for the fixup. It does indeed look much better now. Cool!

I merged your commit series to staging now.

ssumpf, 2020-02-11:
@nfeske: de9f0ab should fix the discovery for x260

nfeske, 2020-02-11:
Thanks to your fixup, Sculpt works nicely on my x260 again. Merged your fixup to staging.

ssumpf, 2020-02-19:
I have added `sync` command support to the AHCI driver with 5f8666a, 04696df expands the block tester to support sync for `replay`. With part_block's sync support also enabled (issue #3652), we now support the `sync` operation for block access all the way down to hardware level. Therefore, @chelmuth, @nfeske: Please consider for staging.

chelmuth, 2020-02-20:
Please reopen the issue or move sync to a dedicated one as this issue was closed by 8a7deae238c1d883b3e0e583a3fff1c3ab14be40.

ssumpf, 2020-02-20:
> Please reopen the issue or move sync to a dedicated one as this issue was closed by [8a7deae](https://github.com/genodelabs/genode/commit/8a7deae238c1d883b3e0e583a3fff1c3ab14be40).

Thanks for the notice.

ssumpf, 2020-02-20:
36b2f2a changes `read` to `write` in `Ata::Protocol::submit` in order to make the code more understandable.

chelmuth, 2020-03-11:
Merged to staging.

nfeske, 2020-03-26:
Merged to master.

commit 3acbb36a (2020-02-10): ahci: switch to request stream API + structural changes

- Remove dated 'Block::Driver' front end and implement
  'Block::Request_stream' front end
- Remove all dynamic memory allocations but DMA memory allocations
- Remove 'Platform_hba' and implement platform specific functions in
  'spec/<platform>/*'
- Ata and Atapi don't inherit from 'Port' any more, but are a member of
  'Port' as a protocol implementation
- Use platform driver for MMIO mappings (x86)
- Exchange stateful initialization of Ata/Atapi with a sequential
  initialization using 'wait_for_any' and 'retry' patterns
- Fix Atapi i [... cut]
commit 8a7deae2 (2020-02-20): ahci: switch to request stream API + structural changes

- Remove dated 'Block::Driver' front end and implement
  'Block::Request_stream' front end
- Remove all dynamic memory allocations but DMA memory allocations
- Remove 'Platform_hba' and implement platform specific functions in
  'spec/<platform>/*'
- Ata and Atapi don't inherit from 'Port' any more, but are a member of
  'Port' as a protocol implementation
- Use platform driver for MMIO mappings (x86)
- Exchange stateful initialization of Ata/Atapi with a sequential
  initialization using 'wait_for_any' and 'retry' patterns
- Fix Atapi i [... cut]
closed 2020-03-26 by nfeske