genodelabs/genode #3640: hw: optimize ipc receive capability allocation

skalk, 2020-02-06:
On each ipc wait or call, when there is the need to potentially receive capabilities, those need to get allocated in advance. Until now this was done simply each time, and unsed ones were freed after it. To limit the burden of frequent allocations, we can instead always hold a level of allocations, and re-allocate new ones next time when a thread waits or calls.

nfeske, 2020-02-10:
The patch does not seem to apply cleanly to the current staging branch. @skalk could you please check?

skalk, 2020-02-11:
I had synchronized it with the changes done by @m-stein on his issue #3308 branch. I again rebased those changes with my commit on current staging. You can use my merge_to_staging branch therefore.

nfeske, 2020-02-11:
Thanks a lot, Stefan. I merged the commit series to staging now.

commit a114faca (2020-02-11): hw: optimize ipc capability allocation

In preparation of ipc receive, by now a bunch of capabilities got
allocated to be ready iin case of capability receiption. After that
unuse slots were freed again. This overhead of senseless (de-)allocations
is replaced in this commit by just restock capability slots that got
used by the last receive.

Fix #3640
closed 2020-02-20 by chelmuth