genodelabs/genode #3646: window_layouter: dynamic scree

skalk, 2020-02-11:


skalk, 2020-02-11:
Although, there is already rudimentary support for having more than one "screen" in the window_layouter's configuration, it does not allow to dynamically switch in between of those. When having more than 3 or 4 windows, which overlap heavily or stay fullscreen it is tiring to use the switch-in-between-windows shortcut. Therefore, it would be nice to first switch in between different screens via keyboard shortcuts.

nfeske, 2020-02-13:
Thanks for this cool feature. Can't wait to merge (and use) it. Please find my nitpicks as inline comments.

skalk, 2020-02-13:
@nfeske ok, I've incorporated the points you made

nfeske, 2020-02-13:
Very good. I merged your new patch along with my minor fixup https://github.com/genodelabs/genode/commit/a67be171cb76be7b82202172e8d1be0d31cd025d. May you give the result another spin?

skalk, 2020-02-17:
I gave it a try, and it seems to work as expected.

closed 2020-02-20 by chelmuth

nfeske, 2020-03-27:
Unfortunately, the screen-handling feature introduced a regression that I noticed today while experimenting with the tiling features of the window layouter. The tiling no longer works. (This made me wonder during my FOSDEM presentation actually, but back then, I brushed it aside as a typo or something).
 
In https://github.com/genodelabs/genode/commit/3c830c18e7c305841305345b477fe32e1d37e082, I reverted the offending part of the patch and thereby restored the original behavior. The dynamic screen handling still seems to work well, which suggests that the dropped condition was probably an accident.

@skalk Can you remember the thoughts behind this particular change? Do you think that restoring the condition will break the dynamic screen handling in some way?

nfeske, 2020-03-27:
BTW, I updated my Sculpt 20.02 index with the fixed version. So you can easily give it a try.

nfeske, 2020-03-30:
Some further testing with my commit reveals that it introduces a problem: In the rules, the geometric attributes of several assign nodes become zero, which is not intended. I'll investigate this further.

nfeske, 2020-03-30:
Commit https://github.com/genodelabs/genode/commit/c6efef1d894335c5e4edbdfbd13582f52baa3cb4 fixes the problem by implementing the support for multiple screen targets and making the distinction between visible and invisible targets explicit. The original version of the dynamic-screen feature created only one screen target. With only the first screen present in the list of targets, windows on invisible screens would naturally never match any target. The dropped condition covered up this missing part of the implementation.

commit c9ed0382 (2020-02-13): window_layouter: add dynamic screen handling

This commit adds dynamic switching in between different defined
screens via keyboard shortcuts, or by editing the rules file.
In the window_layouter package it defines per default ten screens,
reachable by pressing the screen key along with numbers 0-9.

Fix #3646