genodelabs/genode #3649: sculpt_manager: accept uplink from local network too

skalk, 2020-02-13:
This was brought up by Colin Parker on the mailing list: when trying to use a passed-through wifi-card not supported by Genode as uplink, by appropriately configuring the nic_router, the sculpt_manager does not provide depot download functionality neither, because it checks for an uplink that is server of the nic_router. This limitation is not necessary. As long as there is network at all, and the uplink got an IP address, the sculpt_manager might allow depot downloads.

skalk, 2020-02-13:
commit 1eec61843905ef85c12be08c8405f7a04741a3f7 implements this test-wise, feedback from Colin is welcome.

skalk, 2020-02-24:
Colin Parker confirmed on the mailing list that the solution works for him.

nfeske, 2020-02-24:
Very good. Given this confirmation, I merged your patch to staging now.

nfeske, 2020-02-28:
Merged to master.

closed 2020-02-28 by nfeske