genodelabs/genode #3651: hw: synchronize ARMv7 context switch appropriately

skalk, 2020-02-14:
Recent changes, e.g. issue #3640, in the generic parts of the base-hw kernel shortened the kernel-path or better reduced branches taken in between. Therefore, a problem got obvious on the i.MX 6Quad SABRElite board. When switching in between different PDs, we have to be more accurate in synchronizing address-space ID and page-table changes. Otherwise, a corrupted branch-predictor is the result. Actually, this is a regression, we had a more conservative implemention long time ago.

skalk, 2020-02-14:
@nfeske please apply commit eca606d to staging. It should resolve the current nightly issues of hw on imx6q_sabrelite. At least it fixes the depot_autopilot tests, which I've tested.

nfeske, 2020-02-14:
Wow! I'm happy that you tracked down this painful issue in such a short time. Thank you for providing the fix, which I merged to staging just now.

commit 9944fd6d (2020-02-14): hw: be more accurate in synchronizing ASID/Pages

Fix #3651
closed 2020-02-20 by chelmuth