genodelabs/genode #3652: pat_block: switch to Request_stream and Job APIs

ssumpf, 2020-02-17:


ssumpf, 2020-02-17:
@chelmuth, @nfeske: I have tested commit d04a34a with `test/block/client` and `block_tester` in parallel. Please review the commit and consider it for staging.

ssumpf, 2020-02-18:
@nfeske: e28ce3b add for fairness + `io_buffer` configuration support. The performance is on par with the former version of `part_block`.

nfeske, 2020-02-18:
Very nice! As a minor change, could you please replace the static `start_index` local variable by a proper `_start_index` member variable? This would make the state more obvious.

Generally, there should be no need for any static local variable (or global variables in general) when using the Genode API (with the component's `Main` instance as the sole exception).

ssumpf, 2020-02-18:
> Very nice! As a minor change, could you please replace the static `start_index` local variable by a proper `_start_index` member variable? This would make the state more obvious.
>
> Generally, there [...]

done with f055304

nfeske, 2020-02-19:
Its cool to see the new block-level APIs at work. Thanks @ssumpf for the big refactoring work.

I continued reviewing the new version of part_block and have a few more suggestions.

@ssumpf while you are at cleaning up the implementation, could you please replace the `to_string` methods by a `print` method compliant the Genode's output.h interface? This would eliminate the use of `snprintf` (a legacy) and the nasty static local buffer. The latter is a dangerous pattern because the buffer is the same for multiple instances of the object. So subsequent calls of `to_string` for different objects change the content pointed-to by the pointer returned by the first call. In general, whenever we spot a local static variable, let's try to eliminate it.

It would be nice to align the `read` calls in the `dump_hdr` method to make the code easier to read. There are also a few other places where horizontal alignment would be welcome. Maybe you could skim over the code again looking out for such refinements?

Please change the overly long line in `Partition_table::Sector::Sector` into multiple lines, i.e., by laying out the `Operation` members in a table-like way.

I spotted a few places where a reference is returned. In my opinion, it would be worth considering changing them into the `with_...` pattern where the method takes a lambda as argument, which is then called with the reference to the requested object. This pattern avoids the returning of pointers/references and thereby rules out many potential use-after-free problems.

Granted, most of these remarks are rather cosmetic. But now that we touch the component with your major overhaul, I think this is a good time to apply such changes as well.

ssumpf, 2020-02-19:
@nfeske: da448d5 addresses your suggestions

nfeske, 2020-02-19:
This looks super! Thanks a lot for taking the time for these finishing touches.

I'll merge your commits to staging immediately after the next master update, presumably tomorrow.

ssumpf, 2020-02-21:
@chelmuth: 8a2289b make `part_block` compile on 32 bit platforms.

chelmuth, 2020-02-21:
Patch applied, thanks.

commit ffb33e55 (2020-02-20): part_block: switch to Request_stream and Job APIs

- use Job API as block connection back end
- use Request_stream API as front end
- use Mmio framework for gpt and mbr parsing
- implement sync correctly

fixes #3652
commit 3995d2f4 (2020-02-27): part_block: switch to Request_stream and Job APIs

- use Job API as block connection back end
- use Request_stream API as front end
- use Mmio framework for gpt and mbr parsing
- implement sync correctly

fixes #3652
closed 2020-02-28 by chelmuth