genodelabs/genode #3653: black hole - a dummy implementation of common session interfaces

nfeske, 2020-02-18:
As a commonly requested feature for Sculpt (e.g., by @mewmew), it would be nice to have the ability to wire up various sessions of a deployed component to a dummy version of the required service. This way, the user could easily start an application that would normally require a NIC session but connect this to a "black hole" network that would simply drop all packets.

This feature does not need to be builtin into Sculpt but it can be implemented as a regular component to be installed as a package. As its runtime description would list the provided services, the sculpt manager would present the "black hole" for all the covered services once the black-hole component is deployed. It would be sensible to offer this component by default in the depot of genodelabs in the "Tools" section.

For the first version, I propose to cover only the NIC session because I see this as the most prominent use case of black-holing. Further down the road, we should add a ROM service (just serving an "<empty/>" ROM module), an Audio_out service (so that games can run even if no audio driver is present), and a (read-only) file-system service with no content.

nfeske, 2020-02-18:
Note that even though the component sounds mundane, it is security critical because one instance would certainly shared by multiple (mutually distrusting) components in practice. For this reason, the component must be implemented with the same care as a resource multiplexer, not allowing for any (covert) crosstalk between sessions.

mewmew, 2020-02-18:
> Note that even though the component sounds mundane, it is security critical because one instance would certainly shared by multiple (mutually distrusting) components in practice. For this reason, [...]

Potentially, the same component could spawn a unique instance for every user of the service. In that way, even if an attacker were to gain control of one of these spawned components, they would not be able to see e.g. the network requests being sent by other application to the compromised component.

nfeske, 2020-02-18:
@mewmew Yes, that would be the way to go. Each client has a distinct session object anyway (with the way the Genode API works). The black hole component just must ensure that there is no state shared by any two sessions. This is actually not hard but it must be considered.

ttcoder, 2020-02-23:
+1 to providing a "null audio-out" sometime over the course of 2020 :-)  This would help me with testing my app on the Qemu side (where I could make audio work under the host) without resorting to #if-defing in my code, but rather just a different .run file for running in Qemu or running on bare metal.

cproc, 2021-08-24:
Commit b9670e1 implements a first version of the black hole component, providing a dummy 'Audio_out' service.

nfeske, 2021-11-30:
The component has become part of Sculpt. Let's close this issue and open a new issues later when extending the component to further session interfaces.

closed 2021-11-30 by nfeske