genodelabs/genode #3655: vfs_import error when importing large files in to a <fs/> node

trimpim, 2020-02-18:
I want to import the content of a tar file in to a `<fs/>` node. This fails with the following error:
```
[init -> vfs] Error: failed to create <import> VFS node
[init -> vfs] Error: 	overwrite="true"
```

Debugging the problem revealed that an exception is thrown at https://github.com/genodelabs/genode/blob/master/repos/os/src/lib/vfs/fs_file_system.h#L514 when the file is "too big".

trimpim, 2020-02-18:
@nfeske I have a fix that solves the problem for me, but I'm not happy with it. I think there should be a better solution for this, but I can't see it at the moment.

nfeske, 2020-03-02:
Thanks @trimpim for reporting. To me it seems that a proper fix would be a write loop implemented in the vfs-import plugin.

trimpim, 2020-03-03:
@nfeske Is there a place where I can look how such a write loop is implemented? Then I would update my patch accordingly.

nfeske, 2020-03-03:
You may find inspiration at https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/text_area/new_file.h

Eventually, a utility like this will become part of `os/vfs.h` but before doing that, we need to gather more practical experience.

trimpim, 2020-03-19:
@nfeske I'm currently trying to implement the write loop in `lib/vfs/fs_file_system.h`. I'm not sure, if this is the location you intended the write loop to go, as the current code in the `_write()` method uses the `packet_stream` interface to write the data.

To copy and modify the code so it compiled was not that difficult. But when I execute the test, `handle.fs().write()` never returns.

nfeske, 2020-03-19:
Sorry for not being precise enough. The write loop belongs in the vfs-import plugin that is only active at initialization time. The fs-file-system must not implement a write loop but always immediately return to the caller.

trimpim, 2020-03-19:
@nfeske no problem. I think this is ready for your review now.

nfeske, 2020-03-19:
From glancing over the patch, it looks very nice. Only I wonder, shouldn't you better cancel the outer loop when a write error occurs?

trimpim, 2020-03-19:
@nfeske True, I have updated the second commit.

nfeske, 2020-03-20:
Thanks for reworking the patch. I have one more question: You removed the `Vfs_handle::Guard` and `Flush_guard` variables. Can you explain why? AFAIK, without these guards in place, we will leak file handles and miss the triggering of sync operations.

```
Vfs_handle::Guard guard(dst_handle);
...
Flush_guard flush(env, *dst_handle);
```

trimpim, 2020-03-24:
@nfeske this was not intentional, I have fixed it and rebased to latest master.

nfeske, 2020-03-24:
Alright. Thanks for the update. I'll merge your new version to staging after the next master update, presumably on Thursday.

nfeske, 2020-03-26:
I merged the commit to staging along with a few spelling fixes of the commit message.

@trimpim even though I appreciate your contribution, I am not entirely happy about the process. The amount of guidance needed and the attention required from my side for reviewing the intermediate results is certainly more effort than I'm comfortable with. Regardless of whether you or someone at Genode Labs changes any code of the framework infrastructure, the burden of quality control is on Genode Labs' side. By accepting contributions, we are taking over this responsibility. But if I have to fix spelling mistakes, repeatedly pointing out our coding-style rules, or hinting at glaring omissions like the missing closing of the file descriptors, my confidence slips. Maybe we can reach a different mode of working?

I very much appreciate that you successfully find workable solutions for problems and limitations you encounter within Genode and share them with us. For us, it is important to know about these limitations and requirements, to keep them in the back of our heads, and come up with sustainable solutions over time. Such a sustainable solution often looks quite different from an immediate fix that you created for your problem. You may take this issue as a good illustration. Sometimes a sustainable solution takes *much* more effort than expected, e.g., when it calls for a partial software redesign. It should not be approached hastily.

Whenever someone creates a pull request featuring an intermediate fix, I feel the social pressure to mold the intermediate fix into a sustainable solution because this would be a prerequisite for merging it. But this urge overthrows the priorities of my other work. If the topic is neither on the road map nor otherwise critical for my current working topics, I suffer under a priority inversion problem. Glimpsing at #3712, I cannot help but see this pattern repeat.

Long story short, would you be fine with me taking your intermediate fixes as mere illustrations but deferring the actual solution as a topic for Genode Labs to work on? It goes without saying that I will still try to provide assistance. But it would be a relief to not enter the unspoken agreement to craft a solution for upstream in a timely manner together.

trimpim, 2020-03-26:
@nfeske Sorry for causing that many interruptions on your side, this was certainly not my intention! 

I'll improve further contributions in this regard.

That said I'm definitely fine with your suggestion to take my intermediate fixes as an illustration for the problem while deferring the actual solution as a topic for Genode Labs!

nfeske, 2020-03-26:
I'm grateful for your understanding.

commit fa63c668 (2020-03-26): vfs_import: implement write loop for copy_file

This is required to allow import of large file.

Fixes #3655
commit 76187331 (2020-03-27): fixup "vfs_import: implement write loop for copy_file" (fix build on 32-bit)

This commit fixes the following build error on 32-bit platforms:

  error: narrowing conversion of ‘bytes_from_source’ from
  ‘Vfs::file_size’ {aka ‘long long unsigned int’} to ‘Genode::size_t’
  {aka ‘long unsigned int’} inside { } [-Werror=narrowing]

Fixes #3655
commit 9da428dc (2020-04-17): vfs_import: implement write loop for copy_file

This is required to allow import of large file.

Fixes #3655
closed 2020-04-17 by chelmuth