genodelabs/genode #3657: timeout framework odd warning leads to failure

skalk, 2020-02-19:
Under rare circumstances, we can observe the warning:
```Warning: timestamp value too big```
Apart from the unknown reason why this happens, the warning repeatedly floods the serial output, which seriously affects runtime behaviour. Tonight on two different i.MX boards the ethernet driver got cancelled due to MDIO timeouts that were happening due to this behaviour.
Of course, it would be better to make the imx ethernet driver robust against this, or at least provided CPU quota to it in all run-scripts, but apart from that this timestamp warning is weird and should be fixed or if not serious removed.

skalk, 2020-03-03:
@m-stein do you have an explanation for the observed behaviour?

m-stein, 2020-03-05:
@skalk I cannot give a quick explanation but I'll have a look at it this week..

alex-ab, 2020-06-18:
I see the same on the imx8_eqk board for an application, eventually failing apart:

```
[init -> scrcpy] Warning: timestamp value too big ts=0xffffffffffffffb8  us=0x200 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x7fffffffffffffdc us=0x100 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x3fffffffffffffee us=0x80 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x1ffffffffffffff7 us=0x40 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0xffffffffffffffb us=0x20 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x7fffffffffffffd us=0x10 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x3fffffffffffffe us=0x8 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0x1ffffffffffffff us=0x4 shift=9 pthread.1
[init -> scrcpy] Warning: timestamp value too big ts=0xffffffffffffff us=0x2 shift=9 pthread.1
[init -> scrcpy] Error: overflow A - 0xfffffffffffffe00 0x2a5ae6a
[init -> scrcpy] Error: Uncaught exception of type 'Genode::Duration::Overflow'
[init -> scrcpy] Warning: abort called - thread: pthread.1
```

alex-ab, 2020-06-18:
According to 6ee5f7bc3ed4cbb122abf576d07065dd8fd4af8e the _timestamp() values may be smaller than the previous _ts value.

```
init -> scrcpy] Error: pthread.1 negative 0x170ac71d - 0x170aca22 0x30c -> 0x0
[init -> scrcpy] screen_update_frame update 4111 invocation 4111/0 rotation=0
[init -> nic_drv] Warning: usleep 1
[init -> scrcpy] INFO: 11 fps (+7 frames skipped)
[init -> scrcpy] screen_update_frame update 4121 invocation 4121/0 rotation=0
[init -> scrcpy] Error: pthread.1 negative 0x171c0282 - 0x171c1131 0xeb3 -> 0x0
[init -> nic_drv] Warning: usleep 1
[init -> scrcpy] INFO: 12 fps (+8 frames skipped)
[init -> scrcpy] screen_update_frame update 4131 invocation 4131/0 rotation=0
[init -> nic_drv] Warning: usleep 1
[init -> scrcpy] Error: pthread.1 negative 0x172e9ff5 - 0x172eaeb0 0xec1 -> 0x0
[init -> scrcpy] INFO: 12 fps (+8 frames skipped)
[init -> scrcpy] screen_update_frame update 4141 invocation 4141/0 rotation=0
[init -> nic_drv] Warning: usleep 1
[init -> scrcpy] INFO: 12 fps (+8 frames skipped)
[init -> scrcpy] screen_update_frame update 4151 invocation 4151/0 rotation=0
[init -> scrcpy] Error: pthread.1 negative 0x1743f3ef - 0x174402cf 0xee5 -> 0x0
[init -> scrcpy] Error: pthread.1 negative 0x17477b22 - 0x17478827 0xd07 -> 0x0
```

alex-ab, 2020-06-18:
With 341a006f3540d9282ff933e935c69205a5dcf2f8 the application survives and runs.

It seems that that the timeout framework can't handle _timestamp() values coming from different CPUs. This seems to must be expected, which articles like https://lwn.net/Articles/388188/ and  https://lwn.net/Articles/209101 shows.

chelmuth, 2020-06-18:
Didn't the original error posted above occur on an uniprocessor system (or without the use of multiple cores)?

alex-ab, 2020-06-18:
The observation is what matters, that backwards running _timestamp() values confuse the timeout framework, may it be uni or multi processor.

nfeske, 2020-06-26:
@alex-ab Should I merge https://github.com/alex-ab/genode/commit/f63e741c64c6277f139a3996c31bc72b4bd08570 to staging? 

To me it looks valuable to have this sanitizing in place in any case. It would likely have prevented sporadic failures of the pthread depot-autopilot test since the test spans multiple CPUs. E.g., this night, we got the following result on x86_64.pc.hw:
```
[2020-06-26 04:20:38] [init -> depot_autopilot] 3.054 [init -> test-pthread] Warning: timestamp value too big[0m[0m
[2020-06-26 04:20:38] [init -> depot_autopilot] 3.063 [init -> test-pthread] Error: Uncaught exception of type 'Genode::Duration::Overflow'[0m[0m
[2020-06-26 04:20:38] [init -> depot_autopilot] [0m[0m
[2020-06-26 04:20:38] [init -> depot_autopilot]  test-pthread                    failed     3.063  log[0m[0m
```

alex-ab, 2020-06-26:
> @alex-ab Should I merge [alex-ab@f63e741](https://github.com/alex-ab/genode/commit/f63e741c64c6277f139a3996c31bc72b4bd08570) to staging?

Admittedly, this question must be answered by you or @m-stein, which knows the timeout framework better.

nfeske, 2020-06-26:
Ok then. So just merged it to staging.

alex-ab, 2020-06-30:
The warning still happens on nightly test on arm_v7a hw qsb. The reason is that the internal used Timestamp type (orginiated from Trace::Timestamp) is 32bit. According to a offline discussion, the timeout framework somehow assumes that the type is 64bit, which is not in reality on all platforms.

According to some discussion and different opinions,

* either the timeout framework should not make assumption about Timestamp being ever 64bit.
* or we have to make sure the type Trace::Timestamp is indeed ever 64bit
* or, if we don't want or can't ensure this, the Timestamp type should not be identical to the Trace::Timestamp type

I added some commits fixing it in one or another way to showcase the impact.

m-stein, 2020-06-30:
@alex-ab As already mentioned during the offline discussion I would vote for introducing
```
namespace Timeout {
   struct Timestamp {
      uint64_t value;
      explicit Timestamp (uint64_t value) ...
   };
}
```
and maybe statically check that it has the same size as microseconds values. Maybe I missed it, but it seems that this solution is not covered by your commits.

alex-ab, 2020-06-30:
> @alex-ab As already mentioned during the offline discussion I would vote for introducing

The commits are meant as direction proposal. Depending on the decision, your fixup is appreciated.

chelmuth, 2020-10-09:
@m-stein I had the wrong assumption this issue was also addressed by the commits in #3884 but learned from the nightly logs that we still suffer from a vast amount of `timestamp value too big` messages. I'd much appreciate if we could close this issue before Genode 20.11.

m-stein, 2020-10-12:
@chelmuth No, indeed #3884 is not related to timestamps and time-interpolation. I will investigate this.

m-stein, 2020-10-19:
@chelmuth I've created a quickfix that reduces the number of printed warnings significantly:

d47145152e timeout: do not warn "ts value too big" too often

chelmuth, 2020-10-19:
I merged the commit to staging.

commit 64487ded (2020-10-19): timeout: don't warn "timestamp value too big" too often

Fixes #3657
closed 2020-10-27 by chelmuth