genodelabs/genode #3662: vbox/nova: freeze with overmap attempt

nfeske, 2020-02-22:
I'm using Sculpt based on the current staging branch. Today, I experienced the following problem twice:

```
[runtime -> vbox5-nova-sculpt -> vbox] [34mWarning: attempt to handle the same signal context twice (nested)[0m
[kernel] [ 1] overmap attempt MEM - tree - PD:0xffffffff81d3b280->0xffffffff815feb80 SB:0x001b3600 RB:0x00137a00 O:0x09 A:0x7 SUB:2
[kernel] [ 2] overmap attempt MEM - tree - PD:0xffffffff81d3b280->0xffffffff815feb80 SB:0x001c8e00 RB:0x00122200 O:0x09 A:0x7 SUB:2
```

At this point, the VM freezes.

I cannot say if the first message is related two the kernel messages or if some time has passed between them.

@alex-ab Do you have an interpretation of the kernel messages?

nfeske, 2020-02-22:
Another freeze:

```
[0m[runtime -> vbox5-nova-sculpt -> vbox] [34mWarning: attempt to handle the same signal context twice (nested)[0m
[kernel] [ 0] overmap attempt MEM - tree - PD:0xffffffff8a2fe280->0xffffffff8a451b80 SB:0x001b8a00 RB:0x00134200 O:0x09 A:0x7 SUB:2
[runtime -> vbox5-nova-sculpt -> vbox] VMM: flag 4 pending[0m
[core] no RM attachment (READ pf_addr=0x50 pf_ip=0x844b6 from pager_object: pd='init -> runtime -> vbox5-nova-sculpt -> vbox' thread='EMT-3') [0m
[core] page fault, pd='init -> runtime -> vbox5-nova-sculpt -> vbox' thread='EMT-3' cpu=3 ip=0x844b6 address=0x50 stack pointer=0xa0bfe730 qualifiers=0x4 irUwp reason=1[0m

```

nfeske, 2020-02-22:
Another crash. This time, there was *no* overmap-attempt message but the same page fault again, with the `pf_ip=0x844b6` and `pf_addr=0x50` like above.

The surrounding code in the dynamic linker is as follows:

```
/plain/no/genode.git/repos/base/src/lib/base/lock.cc:41
{
   8449d:	41 57                	push   %r15
thread_check_stopped_and_restart():
/plain/no/genode.git/repos/base-nova/src/include/base/internal/lock_helper.h:39

static inline bool thread_check_stopped_and_restart(Genode::Thread *thread_base)
{
	Genode::addr_t sem = thread_base ?
	                     thread_base->native_thread().exc_pt_sel + Nova::SM_SEL_EC :
   8449f:	49 89 d7             	mov    %rdx,%r15
/plain/no/genode.git/repos/base-nova/src/include/base/internal/lock_helper.h:38
	Genode::addr_t sem = thread_base ?
   844a2:	48 85 ff             	test   %rdi,%rdi
   844a5:	74 29                	je     844d0 <_ZN6Genode15Cancelable_lock9Applicant7wake_upEv+0x50>
/plain/no/genode.git/repos/base-nova/src/include/base/internal/lock_helper.h:39
	                     thread_base->native_thread().exc_pt_sel + Nova::SM_SEL_EC :
   844a7:	48 b8 b8 47 f9 ff ff 	movabs $0xfffffffffff947b8,%rax
   844ae:	ff ff ff 
   844b1:	48 01 d0             	add    %rdx,%rax
   844b4:	ff d0                	callq  *%rax
/plain/no/genode.git/repos/base-nova/src/include/base/internal/lock_helper.h:38
	Genode::addr_t sem = thread_base ?
   844b6:	48 8b 78 08          	mov    0x8(%rax),%rdi
   844ba:	48 83 c7 1d          	add    $0x1d,%rdi
_ZN6Genode15Cancelable_lock9Applicant7wake_upEv():
/plain/no/genode.git/repos/base-nova/include/spec/64bit/nova/syscalls.h:88
		              : "rcx", "r11", "memory");
   844be:	31 d2                	xor    %edx,%edx
```

nfeske, 2020-02-22:
Rebuilt everything to rule out any potential ABI incompatibility.

Git hit with another flavor of a vbox freeze, while casually browsing the web:

```
[runtime -> vbox5-nova-sculpt -> vbox] VMM: flag 4 pending[0m
[runtime -> vbox5-nova-sculpt -> vbox] [31mError: deadlock ahead, mutex=0xaff8ce8, return ip=0x880e6[0m
```

Shortly after rebooting the VM, another time:

```
[0m[runtime -> vbox5-nova-sculpt -> vbox] VMM: flag 4 pending[0m
[runtime -> vbox5-nova-sculpt -> vbox] [34mWarning: fchmod: fchmod not implemented[0m
[kernel] [ 1] overmap attempt MEM - tree - PD:0xffffffff81dbeb80->0xffffffff81914b80 SB:0x00166800 RB:0x00186400 O:0x09 A:0x7 SUB:2
[runtime -> vbox5-nova-sculpt -> vbox] VMM: flag 4 pending[0m
[runtime -> vbox5-nova-sculpt -> vbox] [34mWarning: attempt to handle the same signal context twice (nested)[0m
[kernel] [ 1] overmap attempt MEM - tree - PD:0xffffffff81dbeb80->0xffffffff81914b80 SB:0x0019c000 RB:0x0014ec00 O:0x09 A:0x7 SUB:2
[runtime -> vbox5-nova-sculpt -> vbox] VMM: flag 4 pending[0m
```

alex-ab, 2020-02-24:
> @alex-ab Do you have an interpretation of the kernel messages?

An mapping data base entry in one of the upper levels are tried to be added to the tree and this fails. I have seen this when two concurrent faulting ECs are trying to resolve the fault quasi concurrently.

nfeske, 2020-02-24:
@alex-ab that's interesting. In this case, the overmap attempt would refer to the same mapping parameters as the existing one, right? So it won't be critical at all. I wonder, could the kernel detect this case and feature this information in the message? This would allow one to distinguish such a non-critical case from a highly critical memory inconsistency that I intuitively suspected when seeing the message.

Assuming that the overmap message refers to the uncritical case and showing merely a symptom, do you think the freezes / crashes may rather be an effect of a race condition within vbox?

alex-ab, 2020-02-24:
> @alex-ab that's interesting. In this case, the overmap attempt would refer to the same mapping parameters as the existing one, right? So it won't be critical at all. I wonder, could the kernel [...]

I will look into it whether the cases can be distinguished certainly and come up with a change.

> Assuming that the overmap message refers to the uncritical case and showing merely a symptom, do you think the freezes / crashes may rather be an effect of a race condition within vbox?

The deadlock ahead message would of course explain the freeze.
The page faults before your rebuild look strange and the code disassembly of lock_helper.h somehow makes me clueless about why rax should become 0.

nfeske, 2020-02-24:
The page faults still occur after my complete rebuild. I have seen exactly the situation above three times today already. I haven't seen the overmap messages in those cases.

@chelmuth and me came to a logical explanation of the `pf_addr` of 0x50. If the `void *_stack` member of the `Thread` object is a nullptr, `Thread::native_thread` would return a reference pointing to 0x48. The `exc_pt_sel` is located at offset 8 within the `Native_thread`. Hence, the de-reference of this member triggers a fault at 0x48 +  8 = 0x50.

The question is why can the `Thread::_stack` become a nullptr? The value is initialized at `Thread` construction time and  there is no legitimate code path setting this value to nullptr.

The only explanation would be someone overwriting an existing `Thread` object with zeros.

BTW, in all crashes I observed, the faulting thread is an EMT thread.

nfeske, 2020-02-24:
I instrumented the `native_thread` method in commit 7d0d9898bdb666e85b81c3682cd94a5bad60fc02 and could validate the claim that `Thread::_stack` is nullptr.

However, the this pointer of the offending `Thread` object turned out to be very strange:
```
[runtime -> vbox5-nova-sculpt -> vbox] [31mError: Thread::_stack is unexpectedly a nullptr![0m
[runtime -> vbox5-nova-sculpt -> vbox] [31mError: this=0x2928[0m
```

There is definitely no `Thread` object at address 0x2928. Hence, I suspect that a `Thread` pointer becomes corrupted / or is not properly initialized. E.g., lock `Applicant` objects come into mind, which use to store a thread pointer.

nfeske, 2020-02-24:
Following a hint by @cproc, I tried reverting 5bbaa30655791694dc5c41286a77dbc45ea6ffe4 and even though I stressed the system during the past two hours quite intensively, there was no stability problem since then. Of course, the commit is not the cause of the problem. It rather brings the issue to the surface.

alex-ab, 2020-02-24:
@nfeske, @cproc: thanks for hint. After brooding over the Mutex now I found one semantic difference. The to_wake_up member of Applicant object is in the Lock case ever initialized to 0 and never re-used. In the Mutex case the Applicant object is reused on a catched Blocking_exception and the _to_wake_up member is not reset but pointing potentially to another Threads stack.

Please check an re-try with the commit.

alex-ab, 2020-02-25:
@chelmuth: added a19d09bd5ec70501d2339d404b3c7b8901e09eb5 version as suggested by you

nfeske, 2020-02-25:
I have cherry-picked @alex-abs's commit https://github.com/alex-ab/genode/commit/a19d09bd5ec70501d2339d404b3c7b8901e09eb5 and successfully used the system for one hour now.

I merged the fix to staging.

ssumpf, 2020-02-25:
I have stress tested VirtualBox extensively. Without Alex's commit I can reproduce the page fault at 0x50, with a19d09b the fault disappears.

nfeske, 2020-02-25:
@ssumpf Thank you for the confirmation.

nfeske, 2020-02-28:
Fixed in master.

closed 2020-02-28 by nfeske