genodelabs/genode #3669: Handle different motion event sources in one Input event stream

cnuke, 2020-02-26:
While updating my `usb_gamepad_input` component I stumpled upon the problem that it is not possible to differentiate between multiple event sources for motion events within one Input session. Modern gamepads have multiple (analog) axis, each generating its own (absolute) motion events and previously I could abuse the `code` member to store the ID of an axis. With the current Input session that is no longer possible.

One obvious fix would be to add an `id` member to the motion events like it is already done for `Touch` events. If that is indeed the preferred solution I would prepare such a commit (after the release).

nfeske, 2022-02-01:
Commit https://github.com/nfeske/genode/commit/4656cbdac8e8ad6b08cd02cfb8672d518d611251 add a new event type `Axis` that carries an axis ID and a float value.

To make it usable in practice, we will need to adjust nitpicker to forward the event to the focused client. @cnuke would that input-routing policy (applying the keyboard focus) be appropriate for this type of event? Is the form of predefining the ID values axis X and Y useful and sufficient?

cnuke, 2024-01-03:
> Commit [nfeske@4656cbd](https://github.com/nfeske/genode/commit/4656cbdac8e8ad6b08cd02cfb8672d518d611251) add a new event type `Axis` that carries an axis ID and a float value.

I tested the commit (that by now already has entered master for some time) with an adapted `usb_gamepad_input` driver where it works as advertised, thanks.
 
> To make it usable in practice, we will need to adjust nitpicker to forward the event to the focused client. @cnuke would that input-routing policy (applying the keyboard focus) be appropriate for this type of event?

I think that is fine as I imagine that in cases where the keyboard focus may be irrelevant the `Axis` event producer is probably colocated with the component that consumes the events.

> Is the form of predefining the ID values axis X and Y useful and sufficient?

Contemporary gamepads have normally 6 axes (2x analog sticks with 2 axes and 2x trigger with one that I have named X, Y, Z, RZ, LT and RT in the past) so from that perspective we could extend the list.

nfeske, 2024-01-03:
Thank you @cnuke for testing. Would commit 590d3e5 fit the bill regarding the axis IDs?

I'd defer the nitpicker topic for now because I would prefer focusing on such usability issues in tandem with other related features (like pointer grabbing, or relative mouse events). When getting to them, I plan to open dedicated issues.

chelmuth, 2024-01-03:
Wouldn't it make more sense to name axis enums according to their physical characteristics, e.g. LEFT_X/Y (RX, RY) and RIGHT_X/Y (LX, LY)? For example, SDL2 uses SDL_CONTROLLER_AXIS_LEFTX etc.

nfeske, 2024-01-03:
> Wouldn't it make more sense to name axis enums according to their physical characteristics, e.g. LEFT_X/Y (RX, RY) and RIGHT_X/Y (LX, LY)?

Sounds good to me. @cnuke do you agree?

cnuke, 2024-01-03:
> I'd defer the nitpicker topic for now because I would prefer focusing on such usability issues in tandem with other related features (like pointer grabbing, or relative mouse events). When getting to them, I plan to open dedicated issues.

I agree, there is no pressing matter to do that now.

> > Wouldn't it make more sense to name axis enums according to their physical characteristics, e.g. LEFT_X/Y (RX, RY) and RIGHT_X/Y (LX, LY)?
> 
> Sounds good to me. @cnuke do you agree?

That's fine with me as well.

I briefly thought about going into the opposite direction as the physical characteristics only match specific devices and depending on the use-case only the order of the axes is relevant (to do the mapping in a library, e.g. SDL2, or in the component itself). But than again it could be useful to address a common use-case for now since its possible to choose the ID freely for special cases anyway.

nfeske, 2024-01-03:
> That's fine with me as well.

Ok, I applied @chelmuth's suggestion in https://github.com/genodelabs/genode/commit/1edf2edfbd10c3b1ce464be562fd9d94296f7acf

> But than again it could be useful to address a common use-case for now since its possible to choose the ID freely for special cases anyway.

It's an unsigned value after all. If we find ourselves repeatedly casting axis IDs from/to custom values, we can always extend the enum.

chelmuth, 2024-01-03:
> I briefly thought about going into the opposite direction as the physical characteristics only match specific devices and depending on the use-case only the order of the axes is relevant (to do the [...]

First, I had the same impulse to remove the logical naming and just use *axis indices*. Fortunately, the current approach combines the *common case* of game controller input elements with the option to report *special axes* via `Axis::Id { <put some value here> }`.

commit 590d3e57 (2024-01-03): input/event.h: add axis ids X, Y, Z, RZ, LT and RT

The axis IDs correspond to 2x analog sticks with 2 axes and 2x triggers.
While being at it, the commit changes the Axis_id type to Axis::Id.

Fixes #3669
commit 87993a86 (2024-02-26): input/event.h: axis ids LX, LY, LT, RX, RY, and RT

The axis IDs correspond to 2x analog sticks with 2 axes and 2x triggers.
While being at it, the commit changes the Axis_id type to Axis::Id.

Fixes #3669
closed 2024-02-29 by chelmuth